mavlink/MAVSDK #717: warnings during mission upload and download

addriumruss, 2019-04-04:
2019-04-04 09:48:48.042 begin upload mission
[09:48:48|Info ] Mission accepted (mission_impl.cpp:163)
[09:48:48|Warn ] Error: not sure how to process Mission ack. (mission_impl.cpp:126)
[09:48:48|Warn ] Error: not sure how to process Mission ack. (mission_impl.cpp:126)
[09:48:48|Warn ] Error: not sure how to process Mission ack. (mission_impl.cpp:126)
[09:48:48|Warn ] Error: not sure how to process Mission ack. (mission_impl.cpp:126)
[09:48:48|Warn ] Error: not sure how to process Mission ack. (mission_impl.cpp:126)
[09:48:48|Warn ] Error: not sure how to process Mission ack. (mission_impl.cpp:126)
[09:48:48|Warn ] 2019-04-04 09:48:48.080 mission upload over. 
2019-04-04 09:48:48.080 begin download mission
Error: not sure how to process Mission ack. (mission_impl.cpp:126)
[09:48:48|Warn ] Error: not sure how to process Mission ack. (mission_impl.cpp:126)
[09:48:48|Warn ] Error: not sure how to process Mission ack. (mission_impl.cpp:126)
[09:48:48|Warn ] Error: not sure how to process Mission ack. (mission_impl.cpp:126)
[09:48:48|Warn ] Error: not sure how to process Mission ack. (mission_impl.cpp:126)
[09:48:48|Warn ] Error: not sure how to process Mission ack. (mission_impl.cpp:126)

julianoes, 2019-04-04:
Hi @addriumruss. Could you give me some more context? Is this happening every time? Are you connected every time? Which version are you using? Is the mission transferred completely? Are you running a certain integration test? How can I reproduce it?

addriumruss, 2019-04-10:
Hi @julianoes , I'm so sorry for seeing your responde until now. There is too much work to do recently.
My sdk version is 
DronecodeSDK version: 0.10.1 (dronecode_sdk_impl.cpp:24)
, and it indeed happens every time, because I set the action in each mission item to 5. 
Thanks to yours work, much times the mission is transferred successful, but sometimes it would fail. Right now the solution I could got is simply do it again.
I'm not sure if it could be reproduced with my code. The mission upload code is already pasted out within issue #720.
Maybe I should upgrade my sdk version first. 
I'll close this issue and try the newest sdk.
Thanks so much!

closed 2019-04-10 by addriumruss

julianoes, 2019-04-10:
>  Maybe I should upgrade my sdk version first.

Yes you should! Why would you not?

addriumruss, 2019-04-10:
The network is always that slow, it'll take much time to download the code. I could do nothing with it but waiting.....

julianoes, 2019-04-10:
Oh, so `git fetch` is slow for you?

addriumruss, 2019-04-10:
unfortunately it is.

julianoes, 2019-04-10:
That's odd. Whereabouts are you?

There are no binaries on the repository, it should all just be source.
If you look up "shallow clone" that might help you to download it quicker. Also, for some cases a VPN helps, you're probably aware of that.