mavlink/MAVSDK #719: Windows build is broken

julianoes, 2019-04-07:
Currently the Windows build seems broken, I tried in "x64 Native Tools Command Prompt for VS 2017". I'm not sure why this is not caught in appveyor.

```
"C:\DroneCore\build\ALL_BUILD.vcxproj" (default target) (1) ->
"C:\DroneCore\build\core\dronecode_sdk.vcxproj" (default target) (3) ->
(ClCompile target) ->
  c:\dronecore\core\locked_queue.h(27): error C2220: warning treated as error - no 'object' file generated [C:\DroneCor
e\build\core\dronecode_sdk.vcxproj]
  c:\dronecore\core\locked_queue.h(27): error C2061: syntax error: identifier 'iterator' [C:\DroneCore\build\core\drone
code_sdk.vcxproj]
  c:\dronecore\core\locked_queue.h(27): error C2334: unexpected token(s) preceding '{'; skipping apparent function body
 [C:\DroneCore\build\core\dronecode_sdk.vcxproj]
  c:\dronecore\core\locked_queue.h(29): error C2061: syntax error: identifier 'iterator' [C:\DroneCore\build\core\drone
code_sdk.vcxproj]
  c:\dronecore\core\locked_queue.h(29): error C2334: unexpected token(s) preceding '{'; skipping apparent function body
 [C:\DroneCore\build\core\dronecode_sdk.vcxproj]
  c:\dronecore\core\locked_queue.h(31): error C2061: syntax error: identifier 'iterator' [C:\DroneCore\build\core\drone
code_sdk.vcxproj]
  c:\dronecore\core\locked_queue.h(31): error C2334: unexpected token(s) preceding '{'; skipping apparent function body
 [C:\DroneCore\build\core\dronecode_sdk.vcxproj]
  c:\dronecore\core\locked_queue.h(27): error C2220: warning treated as error - no 'object' file generated [C:\DroneCor
e\build\core\dronecode_sdk.vcxproj]
  c:\dronecore\core\locked_queue.h(27): error C2061: syntax error: identifier 'iterator' [C:\DroneCore\build\core\drone
code_sdk.vcxproj]
  c:\dronecore\core\locked_queue.h(27): error C2334: unexpected token(s) preceding '{'; skipping apparent function body
 [C:\DroneCore\build\core\dronecode_sdk.vcxproj]
  c:\dronecore\core\locked_queue.h(29): error C2061: syntax error: identifier 'iterator' [C:\DroneCore\build\core\drone
code_sdk.vcxproj]
  c:\dronecore\core\locked_queue.h(29): error C2334: unexpected token(s) preceding '{'; skipping apparent function body
 [C:\DroneCore\build\core\dronecode_sdk.vcxproj]
  c:\dronecore\core\locked_queue.h(31): error C2061: syntax error: identifier 'iterator' [C:\DroneCore\build\core\drone
code_sdk.vcxproj]
  c:\dronecore\core\locked_queue.h(31): error C2334: unexpected token(s) preceding '{'; skipping apparent function body
 [C:\DroneCore\build\core\dronecode_sdk.vcxproj]
  c:\dronecore\core\locked_queue.h(27): error C2220: warning treated as error - no 'object' file generated [C:\DroneCor
e\build\core\dronecode_sdk.vcxproj]
  c:\dronecore\core\locked_queue.h(27): error C2061: syntax error: identifier 'iterator' [C:\DroneCore\build\core\drone
code_sdk.vcxproj]
  c:\dronecore\core\locked_queue.h(27): error C2334: unexpected token(s) preceding '{'; skipping apparent function body
 [C:\DroneCore\build\core\dronecode_sdk.vcxproj]
  c:\dronecore\core\locked_queue.h(29): error C2061: syntax error: identifier 'iterator' [C:\DroneCore\build\core\drone
code_sdk.vcxproj]
  c:\dronecore\core\locked_queue.h(29): error C2334: unexpected token(s) preceding '{'; skipping apparent function body
 [C:\DroneCore\build\core\dronecode_sdk.vcxproj]
  c:\dronecore\core\locked_queue.h(31): error C2061: syntax error: identifier 'iterator' [C:\DroneCore\build\core\drone
code_sdk.vcxproj]
  c:\dronecore\core\locked_queue.h(31): error C2334: unexpected token(s) preceding '{'; skipping apparent function body
 [C:\DroneCore\build\core\dronecode_sdk.vcxproj]
  c:\dronecore\core\locked_queue.h(27): error C2220: warning treated as error - no 'object' file generated [C:\DroneCor
e\build\core\dronecode_sdk.vcxproj]
  c:\dronecore\core\locked_queue.h(27): error C2061: syntax error: identifier 'iterator' [C:\DroneCore\build\core\drone
code_sdk.vcxproj]
  c:\dronecore\core\locked_queue.h(27): error C2334: unexpected token(s) preceding '{'; skipping apparent function body
 [C:\DroneCore\build\core\dronecode_sdk.vcxproj]
  c:\dronecore\core\locked_queue.h(29): error C2061: syntax error: identifier 'iterator' [C:\DroneCore\build\core\drone
code_sdk.vcxproj]
  c:\dronecore\core\locked_queue.h(29): error C2334: unexpected token(s) preceding '{'; skipping apparent function body
 [C:\DroneCore\build\core\dronecode_sdk.vcxproj]
  c:\dronecore\core\locked_queue.h(31): error C2061: syntax error: identifier 'iterator' [C:\DroneCore\build\core\drone
code_sdk.vcxproj]
  c:\dronecore\core\locked_queue.h(31): error C2334: unexpected token(s) preceding '{'; skipping apparent function body
 [C:\DroneCore\build\core\dronecode_sdk.vcxproj]
  c:\dronecore\core\locked_queue.h(27): error C2220: warning treated as error - no 'object' file generated [C:\DroneCor
e\build\core\dronecode_sdk.vcxproj]
  c:\dronecore\core\locked_queue.h(27): error C2061: syntax error: identifier 'iterator' [C:\DroneCore\build\core\drone
code_sdk.vcxproj]
  c:\dronecore\core\locked_queue.h(27): error C2334: unexpected token(s) preceding '{'; skipping apparent function body
 [C:\DroneCore\build\core\dronecode_sdk.vcxproj]
  c:\dronecore\core\locked_queue.h(29): error C2061: syntax error: identifier 'iterator' [C:\DroneCore\build\core\drone
code_sdk.vcxproj]
  c:\dronecore\core\locked_queue.h(29): error C2334: unexpected token(s) preceding '{'; skipping apparent function body
 [C:\DroneCore\build\core\dronecode_sdk.vcxproj]
  c:\dronecore\core\locked_queue.h(31): error C2061: syntax error: identifier 'iterator' [C:\DroneCore\build\core\drone
code_sdk.vcxproj]
  c:\dronecore\core\locked_queue.h(31): error C2334: unexpected token(s) preceding '{'; skipping apparent function body
 [C:\DroneCore\build\core\dronecode_sdk.vcxproj]
  c:\dronecore\core\locked_queue.h(27): error C2220: warning treated as error - no 'object' file generated [C:\DroneCor
e\build\core\dronecode_sdk.vcxproj]
  c:\dronecore\core\locked_queue.h(27): error C2061: syntax error: identifier 'iterator' [C:\DroneCore\build\core\drone
code_sdk.vcxproj]
  c:\dronecore\core\locked_queue.h(27): error C2334: unexpected token(s) preceding '{'; skipping apparent function body
 [C:\DroneCore\build\core\dronecode_sdk.vcxproj]
  c:\dronecore\core\locked_queue.h(29): error C2061: syntax error: identifier 'iterator' [C:\DroneCore\build\core\drone
code_sdk.vcxproj]
  c:\dronecore\core\locked_queue.h(29): error C2334: unexpected token(s) preceding '{'; skipping apparent function body
 [C:\DroneCore\build\core\dronecode_sdk.vcxproj]
  c:\dronecore\core\locked_queue.h(31): error C2061: syntax error: identifier 'iterator' [C:\DroneCore\build\core\drone
code_sdk.vcxproj]
  c:\dronecore\core\locked_queue.h(31): error C2334: unexpected token(s) preceding '{'; skipping apparent function body
 [C:\DroneCore\build\core\dronecode_sdk.vcxproj]

    1058 Warning(s)
    42 Error(s)
```

FYI @MaEtUgR

MaEtUgR, 2019-04-09:
@julianoes I wrote in https://github.com/Dronecode/DronecodeSDK/issues/718#issue-429710976 what I had to do to get a successful build with latest visual studio. Although just the build ran through and some tests failed but I'm not sure what should actually work since I'd probably need to have SITL running.

julianoes, 2019-04-09:
For the unit tests SITL doesn't need to be running.

julianoes, 2019-04-10:
@MaEtUgR but why am I seeing the errors above? And why does appveyor build just fine: https://ci.appveyor.com/project/Dronecode/dronecore/branch/develop/job/r6y86f94c340wqx5

MaEtUgR, 2019-04-11:
So the problem here was the dinamic linked libraries not being in the correct folder which get copied on appveyor?

julianoes, 2019-04-11:
I don't know about this one. It only happens on my Windows at Home.

julianoes, 2019-04-24:
Fixed by #729.

closed 2019-04-24 by julianoes