mavlink/MAVSDK #720: Aborted when download mission

addriumruss, 2019-04-09:
[06:03:21|Warn ] Error: not sure how to process Mission ack. (mission_impl.cpp:126)
[06:03:21|Warn ] Retrying requesting mission item... (mission_impl.cpp:1204)
[06:03:21|Warn ] Retrying requesting mission item... (mission_impl.cpp:1204)
[06:03:21|Warn ] Error: not sure how to process Mission ack. (mission_impl.cpp:126)
2019-04-09 18:03:21.774 [STATUS] lng:14.366780, lat:52.548896, hgt:0.049000, alt:49.945004, speed:1.703467, yaw:1.174842, pitch:-0.749213, armed:0, inair: 0 
[06:03:21|Warn ] Retrying requesting mission item... (mission_impl.cpp:1204)
[06:03:22|Warn ] Retrying requesting mission item... (mission_impl.cpp:1204)
[06:03:22|Warn ] Mission handling timed out while downloading mission. (mission_impl.cpp:1199)
terminate called after throwing an instance of 'std::future_error'
  what():  std::future_error: Promise already satisfied
Aborted (core dumped)

julianoes, 2019-04-09:
Can you share how to reproduce this?
And can you share the whole output so that I can see the version? Thanks.

addriumruss, 2019-04-09:
I upload one mission, then download it immediately.
this is the logs:
2019-04-09 18:03:01.773 [状态] lng:14.366780, lat:52.548896, hgt:0.049000, alt:49.945004, speed:1.703467, yaw:3.617299, pitch:-2.346539, armed:0, inair: 0 
2019-04-09 18:03:01.775 send cmd 50006, DEV_STATUS_UPDATE, target: usp.server 
2019-04-09 18:03:01.786 send success 

[06:03:08|Warn ] sending again, retries to do: 3  (528). (mavlink_commands.cpp:247)
2019-04-09 18:03:11.773 [状态] lng:14.366780, lat:52.548896, hgt:0.049000, alt:49.945004, speed:1.703467, yaw:3.807425, pitch:-3.225212, armed:0, inair: 0 
2019-04-09 18:03:11.777 send cmd 50006, DEV_STATUS_UPDATE, target: usp.server 
2019-04-09 18:03:11.788 send success 

2019-04-09 18:03:12.220 message arrived topicName=5be52a03f5ff60df786989dd, bodySize=855, body=<{"data":{"head":{"d":1,"id":55010,"n":244,"r":4,"t":1554804192245},"route":{"id":"5c9350d0978fdfd72fe48766","name":"设备SDK测试路线","pts":[{"action":0,"alt":0,"autoCon":1,"cmd":16,"flyThr":0,"head":0,"hov":0,"hspd":0,"ht":10,"lat":32.2525,"lng":118.5002,"radius":0,"spd":0},{"action":0,"alt":0,"autoCon":1,"cmd":16,"flyThr":0,"head":0,"hov":0,"hspd":0,"ht":15,"lat":32.2527,"lng":118.5004,"radius":0,"spd":0},{"action":0,"alt":0,"autoCon":1,"cmd":16,"flyThr":0,"head":0,"hov":0,"hspd":0,"ht":20,"lat":32.2521,"lng":118.5006,"radius":0,"spd":0},{"action":0,"alt":0,"autoCon":1,"cmd":16,"flyThr":0,"head":0,"hov":0,"hspd":0,"ht":15,"lat":32.2523,"lng":118.5007,"radius":0,"spd":0},{"action":0,"alt":0,"autoCon":1,"cmd":16,"flyThr":0,"head":0,"hov":0,"hspd":0,"ht":10,"lat":32.2525,"lng":118.5009,"radius":0,"spd":0}]}},"mid":55010,"src":"usp.server"}>
2019-04-09 18:03:12.220 FcowMsg::parseFromJson, mid=55010, name=DEV_NEWROUTE 
2019-04-09 18:03:12.220 打印刚生成的任务计划 
2019-04-09 18:03:12.220 [TESTING] 航点(lng, lat, hgt, act)={118.500200, 32.252500, 10.000000, 5} 
2019-04-09 18:03:12.220 [TESTING] 航点(lng, lat, hgt, act)={118.500400, 32.252700, 15.000000, 5} 
2019-04-09 18:03:12.220 [TESTING] 航点(lng, lat, hgt, act)={118.500600, 32.252100, 20.000000, 5} 
2019-04-09 18:03:12.220 [TESTING] 航点(lng, lat, hgt, act)={118.500700, 32.252300, 15.000000, 5} 
2019-04-09 18:03:12.220 [TESTING] 航点(lng, lat, hgt, act)={118.500900, 32.252500, 10.000000, 5} 
2019-04-09 18:03:12.220 上传航线到飞控系统. 
[06:03:20|Info ] Mission accepted (mission_impl.cpp:163)
2019-04-09 18:03:20.882 航线上传成功. 
2019-04-09 18:03:20.882 打印飞控系统中的飞行任务内容-开始 
[06:03:21|Warn ] Error: not sure how to process Mission ack. (mission_impl.cpp:126)
[06:03:21|Warn ] Retrying requesting mission item... (mission_impl.cpp:1204)
[06:03:21|Warn ] Retrying requesting mission item... (mission_impl.cpp:1204)
[06:03:21|Warn ] Error: not sure how to process Mission ack. (mission_impl.cpp:126)
2019-04-09 18:03:21.774 [状态] lng:14.366780, lat:52.548896, hgt:0.049000, alt:49.945004, speed:1.703467, yaw:1.174842, pitch:-0.749213, armed:0, inair: 0 
[06:03:21|Warn ] Retrying requesting mission item... (mission_impl.cpp:1204)
[06:03:22|Warn ] Retrying requesting mission item... (mission_impl.cpp:1204)
[06:03:22|Warn ] Mission handling timed out while downloading mission. (mission_impl.cpp:1199)
terminate called after throwing an instance of 'std::future_error'
  what():  std::future_error: Promise already satisfied
Aborted (core dumped)
root@ubt16:/home/russ/px4/DeviceSDK2# 

by the way, the device is a pure PIX2.4.8 with PX1.8.2, without any other sub-equipment. it's a HITL simulation, with PIX and jmavsim.  
I wonder if it's the configuration problem.

addriumruss, 2019-04-09:
@julianoes Thanks for attention!
all action in flypoint is set to 5, is this a potential problem?

JonasVautherin, 2019-04-09:
> terminate called after throwing an instance of 'std::future_error'
what(): std::future_error: Promise already satisfied

Can you show us the code? Ideally you could provide a minimal executable code reproducing that issue. But at least we will need to see some code, I believe.

addriumruss, 2019-04-09:
sure, this is the code:

int printMissionInPx4() {
#if DEBUG
	if(globalMissionPtr->total_mission_items() <= 0){
		printf("%s no fly point in px4 \n", FcowUtil::current_time());
		return -1;
	}
	//query missions in px4, ONLY FOR TESTING
	printf("%s 打印飞控系统中的飞行任务内容-开始 \n", FcowUtil::current_time());
	shared_ptr<promise<Mission::Result>> prom = make_shared<
			promise<Mission::Result>>();
	future<Mission::Result> future_result = prom->get_future();
	try{
		globalMissionPtr->download_mission_async(
			[prom](Mission::Result result, std::vector<std::shared_ptr<MissionItem>> mis) {
				//for testing, print the mission
				for (std::shared_ptr<MissionItem> mi : mis) {
					printf("%s 航点 (lng, lat, hgt, flythr, loiter, act)={%f, %f, %f, %s, %f, %d} \n", FcowUtil::current_time(), mi->get_longitude_deg(),
							mi->get_latitude_deg(),
							mi->get_relative_altitude_m(),
							mi->get_fly_through() ? "yes" : "no",
							mi->get_loiter_time_s(),
							mi->get_camera_action()
					);
				}
				prom->set_value(result);
			});
	}catch(...){
		printf("%s 航线下载异常. \n", FcowUtil::current_time());
		prom->set_value(Mission::Result::ERROR);
	}
	Mission::Result result = future_result.get(); // Wait on result
	if (result != Mission::Result::SUCCESS) {
		char info[1024] = { 0 };
		sprintf(info, "查询飞控当前航线失败: %s ", Mission::result_str(result));
		printf("%s %s \n", FcowUtil::current_time(), info);
		return -5;
	}
	printf("%s 打印飞控系统中的飞行任务内容-结束 \n", FcowUtil::current_time());
#endif
	return 0;
}
int droneNewRoute(FcowDroneNewRoutePara *para, const string& src) {
	if (!para || !para->route || !para->route->pts) {
		debug_printf("%s 新航线接收失败, 参数为空 \n", FcowUtil::current_time());
		mqttClient->feedback(para->head, 0, src.c_str(), "新航线接收失败, 参数为空");
		return -1;
	}
	if(para->route->id.length() == 0){
		debug_printf("%s 新航线接收失败, 航线编号缺失 \n", FcowUtil::current_time());
		mqttClient->feedback(para->head, 0, src.c_str(), "新航线接收失败, 航线编号缺失");
		return -2;
	}
	if(para->route->pts->size() == 0){
		debug_printf("%s 新航线接收失败, 航点缺失 \n", FcowUtil::current_time());
		mqttClient->feedback(para->head, 0, src.c_str(), "新航线接收失败, 航点缺失");
		return -3;
	}

	//生成px4格式的任务向量
	shared_ptr<vector<shared_ptr<MissionItem>>> missionPlan = flyRoute2MissionPlan(para->route);
	//for testing
	debug_printf("%s 打印刚生成的任务计划 \n", FcowUtil::current_time());
	printMission(missionPlan);

	//上传到px4
	debug_printf("%s 上传航线到飞控系统. \n", FcowUtil::current_time());
	shared_ptr<promise<Mission::Result>> prom = make_shared<promise<Mission::Result>>();
	future<Mission::Result> future_result = prom->get_future();
	globalMissionPtr->upload_mission_async(*missionPlan,
			[prom](Mission::Result result) {
				prom->set_value(result);
			});
	Mission::Result upload_result = future_result.get();
	if (upload_result != Mission::Result::SUCCESS) {
		char info[128] = { 0 };
		sprintf(info, "上传航线到飞控系统失败: %s", Mission::result_str(upload_result));
		debug_printf("%s %s \n", FcowUtil::current_time(), info);
		//反馈给后台
		mqttClient->feedback(para->head, 0, src.c_str(), info);
		return -7;
	}
	debug_printf("%s 航线上传成功. \n", FcowUtil::current_time());

	//print current mission in px4
	printMissionInPx4(); //for testing

	//删除旧缓存数据
	string routeId = para->route->id;
	removeRouteFromMap(routeId);

	//保存到缓存
	debug_printf("%s 添加到缓存 \n", FcowUtil::current_time());
	missionMap.insert(pair<string, FcowDroneRoute*> (routeId, para->route));
	para->route = NULL;

	debug_printf("%s 航线上传处理完成 \n", FcowUtil::current_time());
	return 0;
}

it seems the prolem occured in function printMissionInPx4,  can you see any problems with the code?

addriumruss, 2019-04-09:
it takes about 5 seconds to sucessfully upload the mission, then failed at downloading.

julianoes, 2019-04-09:
Can you paste the version output of the Dronecode SDK please? And when you paste code, please put ``` around it.

JonasVautherin, 2019-04-09:
First, why do you `try...catch`? `download_mission_async` doesn't throw exceptions, does it?

```c++
try{
		globalMissionPtr->download_mission_async(
...
```

Then, can you confirm it is happening when calling `prom->set_value(result);` in `globalMissionPtr->download_mission_async(...)`?

Like... is the callback of `download_mission_async` called twice? If yes, it may be a bug in the SDK.

addriumruss, 2019-04-10:
sure. I'll try this ``` . 
this sdk is listen on udp port 14550, because the 14540 port doesn't work.
I removed the try-catch yestoday, delete the code within the catch block,  and got the same result.

this is the version log:
[06:41:51|Info ] DronecodeSDK version: 0.10.1 (dronecode_sdk_impl.cpp:24)
[06:41:51|Info ] New device on: 127.0.0.1:41310 (udp_connection.cpp:200)
[06:41:52|Warn ] sending again, retries to do: 3  (520). (mavlink_commands.cpp:247)
....

this is the main business code:
``` 
int printMissionInPx4() {
#if DEBUG
	if(globalMissionPtr->total_mission_items() <= 0){
		printf("%s 飞控系统中没有航点信息 \n", FcowUtil::current_time());
		return -1;
	}
	//查询px4中的航线, ONLY FOR TESTING
	printf("%s 打印飞控系统中的飞行任务内容-开始 \n", FcowUtil::current_time());
	shared_ptr<promise<Mission::Result>> prom = make_shared<
			promise<Mission::Result>>();
	future<Mission::Result> future_result = prom->get_future();
	try{
		globalMissionPtr->download_mission_async(
			[prom](Mission::Result result, std::vector<std::shared_ptr<MissionItem>> mis) {
				//for testing, print the mission
				for (std::shared_ptr<MissionItem> mi : mis) {
					printf("%s 航点 (lng, lat, hgt, flythr, loiter, act)={%f, %f, %f, %s, %f, %d} \n", FcowUtil::current_time(), mi->get_longitude_deg(),
							mi->get_latitude_deg(),
							mi->get_relative_altitude_m(),
							mi->get_fly_through() ? "yes" : "no",
							mi->get_loiter_time_s(),
							mi->get_camera_action()
					);
				}
				prom->set_value(result);
			});
	}catch(...){
		printf("%s 航线下载异常. \n", FcowUtil::current_time());
		prom->set_value(Mission::Result::ERROR);
	}
	Mission::Result result = future_result.get(); // Wait on result
	if (result != Mission::Result::SUCCESS) {
		char info[1024] = { 0 };
		sprintf(info, "查询飞控当前航线失败: %s ", Mission::result_str(result));
		printf("%s %s \n", FcowUtil::current_time(), info);
		return -5;
	}
	printf("%s 打印飞控系统中的飞行任务内容-结束 \n", FcowUtil::current_time());
#endif
	return 0;
}

int droneNewRoute(FcowDroneNewRoutePara *para, const string& src) {
	if (!para || !para->route || !para->route->pts) {
		debug_printf("%s 新航线接收失败, 参数为空 \n", FcowUtil::current_time());
		mqttClient->feedback(para->head, 0, src.c_str(), "新航线接收失败, 参数为空");
		return -1;
	}
	if(para->route->id.length() == 0){
		debug_printf("%s 新航线接收失败, 航线编号缺失 \n", FcowUtil::current_time());
		mqttClient->feedback(para->head, 0, src.c_str(), "新航线接收失败, 航线编号缺失");
		return -2;
	}
	if(para->route->pts->size() == 0){
		debug_printf("%s 新航线接收失败, 航点缺失 \n", FcowUtil::current_time());
		mqttClient->feedback(para->head, 0, src.c_str(), "新航线接收失败, 航点缺失");
		return -3;
	}

	//生成px4格式的任务向量
	shared_ptr<vector<shared_ptr<MissionItem>>> missionPlan = flyRoute2MissionPlan(para->route);
	//for testing
	debug_printf("%s 打印刚生成的任务计划 \n", FcowUtil::current_time());
	printMission(missionPlan);

	//上传到px4
	debug_printf("%s 上传航线到飞控系统. \n", FcowUtil::current_time());
	shared_ptr<promise<Mission::Result>> prom = make_shared<promise<Mission::Result>>();
	future<Mission::Result> future_result = prom->get_future();
	globalMissionPtr->upload_mission_async(*missionPlan,
			[prom](Mission::Result result) {
				prom->set_value(result);
			});
	Mission::Result upload_result = future_result.get();
	if (upload_result != Mission::Result::SUCCESS) {
		char info[128] = { 0 };
		sprintf(info, "上传航线到飞控系统失败: %s", Mission::result_str(upload_result));
		debug_printf("%s %s \n", FcowUtil::current_time(), info);
		//反馈给后台
		mqttClient->feedback(para->head, 0, src.c_str(), info);
		return -7;
	}
	debug_printf("%s 航线上传成功. \n", FcowUtil::current_time());

	//print current mission in px4
	printMissionInPx4(); //for testing

	//删除旧缓存数据
	string routeId = para->route->id;
	removeRouteFromMap(routeId);

	//保存到缓存
	debug_printf("%s 添加到缓存 \n", FcowUtil::current_time());
	missionMap.insert(pair<string, FcowDroneRoute*> (routeId, para->route));
	para->route = NULL;

	debug_printf("%s 航线上传处理完成 \n", FcowUtil::current_time());
	return 0;
}

definitions of data structures:
class FcowDroneNewRoutePara {
public: 
    FcowDroneRoute *route = NULL;
public:
    FcowDroneNewRoutePara() {
    }
    virtual ~FcowDroneNewRoutePara() {
        if(route) {
            delete route;
        }
    }
};
class FcowDroneRoute {
public: 
    string id;
    string name;
    vector<FlyPoint*> *pts = NULL;
public:
    FcowDroneRoute() {
    }
    virtual ~FcowDroneRoute() {
        if(pts) {
            for(auto &it : *pts) {
                delete it;
            }
            delete pts;
        }
    }
};
class FlyPoint  {
public: 
    double lng;
    double lat;
    float ht = 0;
    float alt = 0;
    float spd = 0;
    float head = 0;
    char flyThr = 0;
    char action = 5;
};

```

addriumruss, 2019-04-10:
@JonasVautherin Thanks for attention!
The try-catch is added only for the business code. I have great confidence with the api function.
I can't be sure if it happened within download_mission_async, Is it possible that it haven't come to the callback plan, like these logs says "Mission handling timed out "?
Would function download_mission_async still be called when handling time out?

[06:03:21|Warn ] Error: not sure how to process Mission ack. (mission_impl.cpp:126)
[06:03:21|Warn ] Retrying requesting mission item... (mission_impl.cpp:1204)
[06:03:21|Warn ] Retrying requesting mission item... (mission_impl.cpp:1204)
[06:03:21|Warn ] Error: not sure how to process Mission ack. (mission_impl.cpp:126)
[06:03:21|Warn ] Retrying requesting mission item... (mission_impl.cpp:1204)
[06:03:22|Warn ] Retrying requesting mission item... (mission_impl.cpp:1204)
[06:03:22|Warn ] Mission handling timed out while downloading mission. (mission_impl.cpp:1199)
terminate called after throwing an instance of 'std::future_error'
what(): std::future_error: Promise already satisfied
Aborted (core dumped)

addriumruss, 2019-04-10:
This is the whole log info after removed the try-catch block:
[06:41:51|Info ] DronecodeSDK version: 0.10.1 (dronecode_sdk_impl.cpp:24)
2019-04-09 18:41:51.693 UDP连接 localhost:14550 成功 
2019-04-09 18:41:51.693 尝试连接飞控... 
2019-04-09 18:41:51.694 等待飞控... 
[06:41:51|Info ] New device on: 127.0.0.1:41310 (udp_connection.cpp:200)
[06:41:52|Warn ] sending again, retries to do: 3  (520). (mavlink_commands.cpp:247)

2019-04-09 18:41:52.401 发现飞控系统, UUID: 3690507558229980471 

2019-04-09 18:41:53.694 正在检测飞控系统, 请稍候...... 
2019-04-09 18:41:53.695 系统检查完成 

********************  DEVICE INFO *************************

2019-04-09 18:41:53.695 has_camera ? 0 
2019-04-09 18:41:53.695 has_autopilot ? 1 
2019-04-09 18:41:53.695 has_gimbal ? 0 
2019-04-09 18:41:53.695 uuid = 3690507558229980471 
[06:41:54|Error] Error: get param busy timeout: MIS_TAKEOFF_ALT (mavlink_parameters.cpp:458)
[06:41:54|Warn ] sending again, retries to do: 3  (511). (mavlink_commands.cpp:247)
[06:41:54|Warn ] Command ack 511 not matching our current command: 528 (mavlink_commands.cpp:141)
[06:41:54|Warn ] sending again, retries to do: 3  (528). (mavlink_commands.cpp:247)
[06:41:55|Error] Error: get param busy timeout: MIS_TAKEOFF_ALT (mavlink_parameters.cpp:458)
[06:41:55|Warn ] Command ack 528 not matching our current command: 521 (mavlink_commands.cpp:141)
[06:41:55|Warn ] sending again, retries to do: 3  (521). (mavlink_commands.cpp:247)
[06:41:55|Warn ] Command ack 521 not matching our current command: 520 (mavlink_commands.cpp:141)
[06:41:55|Warn ] sending again, retries to do: 3  (520). (mavlink_commands.cpp:247)
[06:41:56|Error] Error: get param busy timeout: CAL_ACC0_ID (mavlink_parameters.cpp:458)
[06:41:56|Error] Error: Param for accel cal failed. (telemetry_impl.cpp:545)
[06:41:57|Error] Error: get param busy timeout: SYS_HITL (mavlink_parameters.cpp:458)
[06:41:57|Error] Error: Param to determine hitl failed. (telemetry_impl.cpp:580)
2019-04-09 18:41:57.653 max speed: 5.000000 
2019-04-09 18:41:57.653 is armed: not armed 
2019-04-09 18:41:57.653 is in air: not in air 
2019-04-09 18:41:57.653 position: (lng,lat,alt,hgt) = (13.861582, 52.371299, 52.416004, 0.000000) 
2019-04-09 18:41:57.653 home position: (lng,lat,alt,hgt) = (13.986477, 52.415524, 51.459003, 0.000000) 
2019-04-09 18:41:57.654 euler angle: (yaw,pitch,roll) = (3.088418, 0.017888, -0.105752) 
2019-04-09 18:41:57.654 ground speed: (v.n,v.e,v.d) = (0.190000, 0.090000, -0.320000) 
2019-04-09 18:41:57.654 gps: num_satellites = 10, fix_type = 3 
2019-04-09 18:41:57.654 battery: remaining_percent = -1.000000%, voltage_v = 65.535004 
2019-04-09 18:41:57.654 flyMode = Hold 
2019-04-09 18:41:57.654 health: {gyrometer,accelerometer,magnetometer,level,local_position,global_position,home_position) = {0,0,0,1,1,1,1} 
2019-04-09 18:41:57.654 rcstatus: signal_strength_percent = 0.000000%, available = 0, available_once = 0 
2019-04-09 18:41:57.654 euler angle of camera: (yaw,pitch,roll) = (nan, nan, nan) 
2019-04-09 18:41:57.654 PositionNED: (north_m,east_m,down_m) = (566768.000000, 362618.625000, 435.553101) 
VelocityNED: (north_m_s, east_m_s, down_m_s) = (0.193655, 0.096237, -0.327722) 
2019-04-09 18:41:57.654 camera vendor: name = , model =  
2019-04-09 18:41:57.654 system.uuid = 3690507558229980471 
2019-04-09 18:41:57.654 flight.version = 1.8, system.version = 7.22 
2019-04-09 18:41:57.654 product(vendor_id,vendor_name,product_id,product_name) = (9900,Yuneec,17,undefined) 

2019-04-09 18:42:06.186 message arrived topicName=5be52a03f5ff60df786989dd, bodySize=855, body=<{"data":{"head":{"d":1,"id":55010,"n":246,"r":4,"t":1554806525134},"route":{"id":"5c9350d0978fdfd72fe48766","name":"设备SDK测试路线","pts":[{"action":0,"alt":0,"autoCon":1,"cmd":16,"flyThr":0,"head":0,"hov":0,"hspd":0,"ht":10,"lat":32.2525,"lng":118.5002,"radius":0,"spd":0},{"action":0,"alt":0,"autoCon":1,"cmd":16,"flyThr":0,"head":0,"hov":0,"hspd":0,"ht":15,"lat":32.2527,"lng":118.5004,"radius":0,"spd":0},{"action":0,"alt":0,"autoCon":1,"cmd":16,"flyThr":0,"head":0,"hov":0,"hspd":0,"ht":20,"lat":32.2521,"lng":118.5006,"radius":0,"spd":0},{"action":0,"alt":0,"autoCon":1,"cmd":16,"flyThr":0,"head":0,"hov":0,"hspd":0,"ht":15,"lat":32.2523,"lng":118.5007,"radius":0,"spd":0},{"action":0,"alt":0,"autoCon":1,"cmd":16,"flyThr":0,"head":0,"hov":0,"hspd":0,"ht":10,"lat":32.2525,"lng":118.5009,"radius":0,"spd":0}]}},"mid":55010,"src":"usp.server"}>
2019-04-09 18:42:06.186 FcowMsg::parseFromJson, mid=55010, name=DEV_NEWROUTE 
2019-04-09 18:42:06.186 打印刚生成的任务计划 
2019-04-09 18:42:06.186 [TESTING] 航点(lng, lat, hgt, act)={118.500200, 32.252500, 10.000000, 5} 
2019-04-09 18:42:06.186 [TESTING] 航点(lng, lat, hgt, act)={118.500400, 32.252700, 15.000000, 5} 
2019-04-09 18:42:06.186 [TESTING] 航点(lng, lat, hgt, act)={118.500600, 32.252100, 20.000000, 5} 
2019-04-09 18:42:06.186 [TESTING] 航点(lng, lat, hgt, act)={118.500700, 32.252300, 15.000000, 5} 
2019-04-09 18:42:06.186 [TESTING] 航点(lng, lat, hgt, act)={118.500900, 32.252500, 10.000000, 5} 
2019-04-09 18:42:06.186 上传航线到飞控系统. 
2019-04-09 18:42:07.655 [状态] lng:33.949695, lat:57.130945, hgt:0.000000, alt:48.179001, speed:0.712390, yaw:3.120135, pitch:0.758453, armed:0, inair: 0 
[06:42:14|Warn ] sending again, retries to do: 3  (528). (mavlink_commands.cpp:247)
[06:42:14|Info ] Mission accepted (mission_impl.cpp:163)
2019-04-09 18:42:14.830 航线上传成功. 
2019-04-09 18:42:14.830 打印飞控系统中的飞行任务内容-开始 
[06:42:15|Warn ] Error: not sure how to process Mission ack. (mission_impl.cpp:126)
[... cut ...]
[06:42:15|Warn ] Retrying requesting mission item... (mission_impl.cpp:1204)
[06:42:15|Warn ] Error: not sure how to process Mission ack. (mission_impl.cpp:126)
[06:42:15|Warn ] Retrying requesting mission item... (mission_impl.cpp:1204)
[06:42:16|Warn ] Retrying requesting mission item... (mission_impl.cpp:1204)
[06:42:16|Warn ] Mission handling timed out while downloading mission. (mission_impl.cpp:1199)
terminate called after throwing an instance of 'std::future_error'
  what():  std::future_error: Promise already satisfied
Aborted (core dumped)

JonasVautherin, 2019-04-10:
> DronecodeSDK version: 0.10.1 (dronecode_sdk_impl.cpp:24)

Would you mind updating the SDK to [the latest version (0.15.0)](https://github.com/Dronecode/DronecodeSDK/releases/tag/v0.15.0) and trying your code again?

julianoes, 2019-04-10:
Yes please update, we've recently fixed some issues around mission plugin crashes:
#714, #711, #701.

addriumruss, 2019-04-10:
Thanks so much! 
I'll spend some time to do the upgrade and try.

closed 2019-04-10 by julianoes