mavlink/MAVSDK #721: Get mavlink messages through the SDK

douglaswsilva, 2019-04-10:
We'd like to be able to read mavlink messages in the Swift SDK, similar to what QGC has. For instance, we were having problems with starting mission and didn't know what the problem was until we look at the debug logs in our console to see this message: 
```
```

douglaswsilva, 2019-04-10:
We can try to implement it ourselves, but want to have your guys input on what would be a good way to go about implementing it. @julianoes

JonasVautherin, 2019-04-10:
@julianoes: Would it make sense to add that to `Telemetry`, or another plugin?

julianoes, 2019-04-11:
Ok it's good that you're describing the actual problem which is that you need to know when and why things don't work. @bkueng is actually working on a proposal and implementation to address exactly this problem. This means we should get much more status information like this in the not so distant future :).

Now until then, we might want to just add the statustext messages to telemetry, it's just a `char[50]`, so nothing fancy.

JonasVautherin, 2019-04-11:
The next step for adding the statustext to the telemetry is to make a PR on the [API definition](https://github.com/Dronecode/DronecodeSDK-Proto/blob/master/protos/telemetry/telemetry.proto) :-).

douglaswsilva, 2019-05-07:
Done with: https://github.com/Dronecode/DronecodeSDK/pull/733

douglaswsilva, 2019-05-07:
Done in: https://github.com/Dronecode/DronecodeSDK/pull/733

closed 2019-05-07 by douglaswsilva