mavlink/MAVSDK #723: Including the SDK inside a Node C++ Addon

aokholm, 2019-04-17:
I’m trying to integrate the DronecodeSDK C++ lib into an electron (nodeJS) app.

I have had success running the offboard_velocity example. I have copied the almost all of the code from the offboard example into Nod  C++ addon.

I have the following std::out::

```before running[34m[11:47:48|Info ] [0mDronecodeSDK version: 0.15.0-14-g531a55b9 (dronecode_sdk_impl.cpp:25)
Wait for system to connect via heartbeat
[34m[11:47:48|Info ] [0mNew system on: 127.0.0.1:14580 (udp_connection.cpp:225)
[32m[11:47:48|Debug] [0mNew: System ID: 1 Comp ID: 1 (dronecode_sdk_impl.cpp:338)
[32m[11:47:48|Debug] [0mComponent Autopilot (1) added. (system_impl.cpp:385)
[32m[11:47:49|Debug] [0mDiscovered 1 component(s) (UUID: 5283920058631409231) (system_impl.cpp:548)```

The code fails when I  try to initialise the shared `auto action = std::make_shared<Action>(system);` or `auto offboard = std::make_shared<Offboard>(system);`

`EXC_BAD_ACCESS (code=EXC_I386_GPFLT)`

Is there something I should be aware of? Any hints for further  debugging? Do you know someone who managed to integrate the SDK into a NodeJS app?

I have uploaded a complete example to (github)[https://github.com/aokholm/nodeDronecodeSDK]

julianoes, 2019-04-17:
Thanks for reporting. Could you reproduce this in `gdb` to get a stacktrace?

aokholm, 2019-04-17:
Using lldb in command line:

```
lldb /usr/local/bin/node main.js 
(lldb) target create "/usr/local/bin/node"
Current executable set to '/usr/local/bin/node' (x86_64).
(lldb) settings set -- target.run-args  "main.js"
(lldb) run
Process 80249 launched: '/usr/local/bin/node' (x86_64)
before running[02:11:51|Info ] DronecodeSDK version: 0.15.0-18-gac0c1fde (dronecode_sdk_impl.cpp:25)
Wait for system to connect via heartbeat
[02:11:51|Info ] New system on: 127.0.0.1:14580 (udp_connection.cpp:225)
[02:11:51|Debug] New: System ID: 1 Comp ID: 1 (dronecode_sdk_impl.cpp:338)
[02:11:51|Debug] Component Autopilot (1) added. (system_impl.cpp:385)
[02:11:51|Debug] Discovered 1 component(s) (UUID: 5283920058631409231) (system_impl.cpp:548)
Process 80249 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=EXC_I386_GPFLT)
    frame #0: 0x0000000105027afb libdronecode_sdk_action.dylib`dronecode_sdk::SystemImpl::MAVLinkHandlerTableEntry::MAVLinkHandlerTableEntry(dronecode_sdk::SystemImpl::MAVLinkHandlerTableEntry const&) + 27
libdronecode_sdk_action.dylib`dronecode_sdk::SystemImpl::MAVLinkHandlerTableEntry::MAVLinkHandlerTableEntry:
->  0x105027afb <+27>: movw   %ax, (%rsi)
    0x105027afe <+30>: movq   %rsi, %rdi
    0x105027b01 <+33>: addq   $0x10, %rdi
    0x105027b05 <+37>: movq   -0x10(%rbp), %rcx
Target 0: (node) stopped.
```

From Xcode

```
Thread 1 Queue : com.apple.main-thread (serial)
#0	0x0000000105becafb in dronecode_sdk::SystemImpl::MAVLinkHandlerTableEntry::MAVLinkHandlerTableEntry(dronecode_sdk::SystemImpl::MAVLinkHandlerTableEntry const&) ()
#1	0x0000000105becacd in dronecode_sdk::SystemImpl::MAVLinkHandlerTableEntry::MAVLinkHandlerTableEntry(dronecode_sdk::SystemImpl::MAVLinkHandlerTableEntry const&) ()
#2	0x0000000105bdaf7f in dronecode_sdk::SystemImpl::register_mavlink_message_handler(unsigned short, std::__1::function<void (__mavlink_message const&)>, void const*) ()
#3	0x0000000105bc83f3 in dronecode_sdk::ActionImpl::init() ()
#4	0x0000000105be4585 in dronecode_sdk::SystemImpl::register_plugin(dronecode_sdk::PluginImplBase*) ()
#5	0x0000000105bc8006 in dronecode_sdk::ActionImpl::ActionImpl(dronecode_sdk::System&) ()
#6	0x0000000105bc802d in dronecode_sdk::ActionImpl::ActionImpl(dronecode_sdk::System&) ()
#7	0x0000000105bc755d in dronecode_sdk::Action::Action(dronecode_sdk::System&) ()
#8	0x0000000105bc761d in dronecode_sdk::Action::Action(dronecode_sdk::System&) ()
#9	0x0000000105807a18 in std::__1::__compressed_pair_elem<dronecode_sdk::Action, 1, false>::__compressed_pair_elem<dronecode_sdk::System&, 0ul>(std::__1::piecewise_construct_t, std::__1::tuple<dronecode_sdk::System&>, std::__1::__tuple_indices<0ul>) at /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/memory:2153
#10	0x0000000105807985 in std::__1::__compressed_pair<std::__1::allocator<dronecode_sdk::Action>, dronecode_sdk::Action>::__compressed_pair<std::__1::allocator<dronecode_sdk::Action>&, dronecode_sdk::System&>(std::__1::piecewise_construct_t, std::__1::tuple<std::__1::allocator<dronecode_sdk::Action>&>, std::__1::tuple<dronecode_sdk::System&>) at /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/memory:2255
#11	0x00000001058075e5 in std::__1::__compressed_pair<std::__1::allocator<dronecode_sdk::Action>, dronecode_sdk::Action>::__compressed_pair<std::__1::allocator<dronecode_sdk::Action>&, dronecode_sdk::System&>(std::__1::piecewise_construct_t, std::__1::tuple<std::__1::allocator<dronecode_sdk::Action>&>, std::__1::tuple<dronecode_sdk::System&>) at /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/memory:2256
#12	0x00000001058074d8 in std::__1::__shared_ptr_emplace<dronecode_sdk::Action, std::__1::allocator<dronecode_sdk::Action> >::__shared_ptr_emplace<dronecode_sdk::System&>(std::__1::allocator<dronecode_sdk::Action>, dronecode_sdk::System&) at /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/memory:3668
#13	0x000000010580705d in std::__1::__shared_ptr_emplace<dronecode_sdk::Action, std::__1::allocator<dronecode_sdk::Action> >::__shared_ptr_emplace<dronecode_sdk::System&>(std::__1::allocator<dronecode_sdk::Action>, dronecode_sdk::System&) at /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/memory:3669
#14	0x0000000105806e57 in std::__1::shared_ptr<dronecode_sdk::Action> std::__1::shared_ptr<dronecode_sdk::Action>::make_shared<dronecode_sdk::System&>(dronecode_sdk::System&) at /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/memory:4327
#15	0x00000001058030ef in std::__1::enable_if<!(is_array<dronecode_sdk::Action>::value), std::__1::shared_ptr<dronecode_sdk::Action> >::type std::__1::make_shared<dronecode_sdk::Action, dronecode_sdk::System&>(dronecode_sdk::System&) at /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/memory:4706
#16	0x00000001058028ef in runProgram(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >) at /Users/aokholm/src/kitex/_3_GroundStation/nodeDronecodeSDK/bridge.cc:209
#17	0x000000010580332b in Method(Napi::CallbackInfo const&) at /Users/aokholm/src/kitex/_3_GroundStation/nodeDronecodeSDK/bridge.cc:261
[... cut ...]
#43	0x0000000100041616 in node::Start(uv_loop_s*, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > const&, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > const&) ()
#44	0x00000001000410f6 in node::Start(int, char**) ()
#45	0x0000000100001034 in start ()
````

julianoes, 2019-04-17:
Hm, not sure why that would happen yet. Can you do `thread apply all backtrace` in gdb?

aokholm, 2019-04-17:
Sure. Thanks

```
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x0)
  * frame #0: 0x0000000105027afb libdronecode_sdk_action.dylib`dronecode_sdk::SystemImpl::MAVLinkHandlerTableEntry::MAVLinkHandlerTableEntry(dronecode_sdk::SystemImpl::MAVLinkHandlerTableEntry const&) + 27
    frame #1: 0x0000000105027acd libdronecode_sdk_action.dylib`dronecode_sdk::SystemImpl::MAVLinkHandlerTableEntry::MAVLinkHandlerTableEntry(dronecode_sdk::SystemImpl::MAVLinkHandlerTableEntry const&) + 29
    frame #2: 0x0000000105015f7f libdronecode_sdk_action.dylib`dronecode_sdk::SystemImpl::register_mavlink_message_handler(unsigned short, std::__1::function<void (__mavlink_message const&)>, void const*) + 607
    frame #3: 0x00000001050033f3 libdronecode_sdk_action.dylib`dronecode_sdk::ActionImpl::init() + 723
    frame #4: 0x000000010501f585 libdronecode_sdk_action.dylib`dronecode_sdk::SystemImpl::register_plugin(dronecode_sdk::PluginImplBase*) + 117
    frame #5: 0x0000000105003006 libdronecode_sdk_action.dylib`dronecode_sdk::ActionImpl::ActionImpl(dronecode_sdk::System&) + 422
    frame #6: 0x000000010500302d libdronecode_sdk_action.dylib`dronecode_sdk::ActionImpl::ActionImpl(dronecode_sdk::System&) + 29
    frame #7: 0x000000010500255d libdronecode_sdk_action.dylib`dronecode_sdk::Action::Action(dronecode_sdk::System&) + 125
    frame #8: 0x000000010500261d libdronecode_sdk_action.dylib`dronecode_sdk::Action::Action(dronecode_sdk::System&) + 29
    frame #9: 0x0000000102c04022 bridge.node`std::__1::shared_ptr<dronecode_sdk::Action> std::__1::shared_ptr<dronecode_sdk::Action>::make_shared<dronecode_sdk::System&>(dronecode_sdk::System&) [inlined] std::__1::__compressed_pair_elem<dronecode_sdk::Action, 1, false>::__compressed_pair_elem<dronecode_sdk::System&, 0ul>(this=<unavailable>, __args=<unavailable>) at memory:2153:9 [opt]
    frame #10: 0x0000000102c04010 bridge.node`std::__1::shared_ptr<dronecode_sdk::Action> std::__1::shared_ptr<dronecode_sdk::Action>::make_shared<dronecode_sdk::System&>(dronecode_sdk::System&) [inlined] std::__1::__compressed_pair<std::__1::allocator<dronecode_sdk::Action>, dronecode_sdk::Action>::__compressed_pair<std::__1::allocator<dronecode_sdk::Action>&, dronecode_sdk::System&>(this=<unavailable>, __first_args=<unavailable>, __second_args=<unavailable>) at memory:2255 [opt]
    frame #11: 0x0000000102c04010 bridge.node`std::__1::shared_ptr<dronecode_sdk::Action> std::__1::shared_ptr<dronecode_sdk::Action>::make_shared<dronecode_sdk::System&>(dronecode_sdk::System&) [inlined] std::__1::__compressed_pair<std::__1::allocator<dronecode_sdk::Action>, dronecode_sdk::Action>::__compressed_pair<std::__1::allocator<dronecode_sdk::Action>&, dronecode_sdk::System&>(this=<unavailable>, __second_args=<unavailable>) at memory:2256 [opt]
    frame #12: 0x0000000102c04010 bridge.node`std::__1::shared_ptr<dronecode_sdk::Action> std::__1::shared_ptr<dronecode_sdk::Action>::make_shared<dronecode_sdk::System&>(dronecode_sdk::System&) [inlined] std::__1::__shared_ptr_emplace<dronecode_sdk::Action, std::__1::allocator<dronecode_sdk::Action> >::__shared_ptr_emplace<dronecode_sdk::System&>(this=<unavailable>, __args=<unavailable>) at memory:3668 [opt]
    frame #13: 0x0000000102c03ffe bridge.node`std::__1::shared_ptr<dronecode_sdk::Action> std::__1::shared_ptr<dronecode_sdk::Action>::make_shared<dronecode_sdk::System&>(dronecode_sdk::System&) [inlined] std::__1::__shared_ptr_emplace<dronecode_sdk::Action, std::__1::allocator<dronecode_sdk::Action> >::__shared_ptr_emplace<dronecode_sdk::System&>(this=<unavailable>, __args=<unavailable>) at memory:3669 [opt]
    frame #14: 0x0000000102c03ffe bridge.node`std::__1::shared_ptr<dronecode_sdk::Action> std::__1::shared_ptr<dronecode_sdk::Action>::make_shared<dronecode_sdk::System&>(__args=<unavailable>) at memory:4327 [opt]
    frame #15: 0x0000000102c02d6b bridge.node`runProgram(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >) [inlined] std::__1::enable_if<!(is_array<dronecode_sdk::Action>::value), std::__1::shared_ptr<dronecode_sdk::Action> >::type std::__1::make_shared<dronecode_sdk::Action, dronecode_sdk::System&>(__args=<unavailable>) at memory:4706:12 [opt]
    frame #16: 0x0000000102c02d63 bridge.node`runProgram(connection_url=<unavailable>) at bridge.cc:209 [opt]
    frame #17: 0x0000000102c0356a bridge.node`Method(info=<unavailable>) at bridge.cc:261:16 [opt]
    frame #18: 0x0000000102c04307 bridge.node`napi_value__* Napi::details::WrapCallback<Napi::details::CallbackData<Napi::String (*)(Napi::CallbackInfo const&), Napi::String>::Wrapper(napi_env__*, napi_callback_info__*)::'lambda'()>(Napi::details::CallbackData<Napi::String (*)(Napi::CallbackInfo const&), Napi::String>::Wrapper(napi_env__*, napi_callback_info__*)::'lambda'()) [inlined] Napi::details::CallbackData<Napi::String (*)(Napi::CallbackInfo const&), Napi::String>::Wrapper(this=<unavailable>)::'lambda'()::operator()() const at napi-inl.h:82:14 [opt]
    frame #19: 0x0000000102c042dc bridge.node`napi_value__* Napi::details::WrapCallback<Napi::details::CallbackData<Napi::String (*)(Napi::CallbackInfo const&), Napi::String>::Wrapper(napi_env__*, napi_callback_info__*)::'lambda'()>(callback=<unavailable>)(Napi::CallbackInfo const&), Napi::String>::Wrapper(napi_env__*, napi_callback_info__*)::'lambda'()) at napi-inl.h:61 [opt]
[... cut ...]
    frame #7: 0x0000000102c8b923 bridge.node`void* std::__1::__thread_proxy<std::__1::tuple<std::__1::unique_ptr<std::__1::__thread_struct, std::__1::default_delete<std::__1::__thread_struct> >, void (dronecode_sdk::ThreadPool::*)(), dronecode_sdk::ThreadPool*> >(__vp=0x00000001029123f0) at thread:352:5
    frame #8: 0x00007fff641c42eb libsystem_pthread.dylib`_pthread_body + 126
    frame #9: 0x00007fff641c7249 libsystem_pthread.dylib`_pthread_start + 66
    frame #10: 0x00007fff641c340d libsystem_pthread.dylib`thread_start + 13
```

julianoes, 2019-04-17:
Thanks for the repo with the source to reproduce this, that's really nice!

Unfortunately this does not crash on my Arch Linux. I can try reproduce it next week on the Mac at work.

julianoes, 2019-04-23:
Ok, I tested this and it doesn't crash on macOS either for me.

[image]

One thing I noticed is that in order to run it I needed to set the path where the libs are to link to:
```
export LD_LIBRARY_PATH=./lib/DronecodeSDK/install/lib
```
Otherwise it did not find `libdronecode_sdk_*.so/dylib`.

Could it be that you are loading libraries that you have previously installed in your system, e.g. in `/usr/lib/` or `/usr/local/lib`? This could explain the crash that you're seeing.

aokholm, 2019-04-23:
Thanks, that's been a huge help. 

It seems your diagnostics could be correct. Using `export LD_LIBRARY_PATH=./lib/DronecodeSDK/install/lib` does solve the problem and I did do a system install of the SDK some months ago to `/usr/local/lib/` previously.

I will try to setup the binding-gyp file correctly so the dylibs can be found at runtime.

julianoes, 2019-04-23:
> I will try to setup the binding-gyp file correctly so the dylibs can be found at runtime.

I found this:
```
'libraries': ['-L/path/to/libraries', '-lA', '-lB'],
```

Here: https://github.com/nodejs/node-gyp/issues/892#issuecomment-197459165

aokholm, 2019-04-26:
The libraries was already included explicit, but the link_setting was missing.  

```
      "link_settings": {
        	"libraries": [
        		"-Wl,-rpath,@loader_path/../../lib/DronecodeSDK/install/lib"
        	],
        },
```

Now the example run as expected!

closed 2019-04-26 by aokholm

julianoes, 2019-04-26:
Nice, thanks for sharing!