mavlink/MAVSDK #728: Adding support for APM

JonasVautherin, 2019-04-23:
## Context

A while ago, @peterbarker worked on adding support for APM in the SDK ([here](https://github.com/peterbarker/DroneCore/tree/ardupilot)). More recently, @jinchengde (and others) started looking into that, again. The goal of this issue is to discuss the high-level requirements and architecture.

## Benefits for APM

The project is architectured in such a way that everything relies on a C++ MAVLink implementation. Making this implementation compatible with APM means that APM will also get the language bindings ([Python](https://github.com/dronecode/dronecodesdk-python), [Swift](https://github.com/dronecode/dronecodesdk-swift), [Java](https://github.com/dronecore/dronecore-java) and future ones like [Javascript](https://github.com/mrpollo/DronecodeSDK-JavaScript)).

The interface common to all our bindings is defined [here](https://github.com/Dronecode/DronecodeSDK-Proto/tree/master/protos) and is used to auto-generate language-specific APIs (e.g. using RxSwift in Swift, RxJava in Java, and asyncio in Python).

## SDK architecture (overview)

The SDK can been seen as composed from 4 parts:

1. The [interface definition](https://github.com/Dronecode/DronecodeSDK-Proto/tree/master/protos) (protobuf files), common to all the languages (C++, Swift, Java, Python). Each proto file is considered a "plugin", which is a group of (MAVLink) features.
2. The C++ [MAVLink implementation](https://github.com/Dronecode/DronecodeSDK), which actually maintains a connection to the drone and does the MAVLink logic.
3. The [backend](https://github.com/Dronecode/DronecodeSDK/tree/develop/backend), which exposes the C++ implementation of (2) as a messaging server through the interface of (1).
4. The language bindings ([Python](https://github.com/dronecode/dronecodesdk-python), [Swift](https://github.com/dronecode/dronecodesdk-swift), [Java](https://github.com/dronecore/dronecore-java)), which implement (1) to communicate with the backend (3).

At the lowest level, each plugin (defined by one proto file) is a C++ library. The backend uses those libraries, as can been seen [here](https://github.com/Dronecode/DronecodeSDK/blob/develop/backend/src/CMakeLists.txt#L38-L46).

JonasVautherin, 2019-04-23:
I can see different ways to achieve that (and I'm looking for more thoughts). I believe that for each feature, there can be 3 situations:

1. The feature has (can have) the same interface between APM and PX4, but not the same MAVLink implementation
2. The feature does not have the same interface between APM and PX4, and therefore not the same MAVLink implementation
3. The feature is compatible between APM and PX4

Because features are grouped into plugins, that applies to plugins as well. The following scenarios could then occur:
1. We could keep the same interface (public headers) and build a different library that could be "swapped": for instance `libdronecode_sdk_apm_mission` and `libdronecode_sdk_px4_mission`, in case mission shares the interface but not the implementation. If the differences are really small, we could also imagine building the same library, but having an interface to decide whether it should start as PX4 or as APM (seems to me that it is what @peterbarker started [here](https://github.com/peterbarker/DroneCore/tree/ardupilot)).
2. In this case, we could possibly have a separate plugin definition (e.g. `mission_px4.proto` and `mission_apm.proto`), and therefore a separate implementation.
3. There is nothing to do here, it should just work

The questions I have would therefore be:
1. Do we have plugins that are completely compatible between APM and PX4 (e.g. [Action](https://github.com/Dronecode/DronecodeSDK-Proto/blob/master/protos/action/action.proto))?
2. What are the features that are not compatible?
3. For the features that are not compatible, could we share the higher-level interface (at the proto file level) and hide the difference as an implementation detail, or do we have to expose a different interface?

Great contributions would be to test the current SDK against an APM SITL and list the features/plugins that are working or not, and possibly why.

jinchengde, 2019-04-25:
I suggest use the same interface but different implementation what peter had done before. because the different between APM and PX4 is not too much bust just some detail in MAVLINK,  therefor we could implement takeoff and land example in APM sitl first.

JonasVautherin, 2019-04-25:
Could we start a list of differences we know? I've heard that there are "not so many" differences quite a few times, but I still have no clue what that means and what they are :-).

hamishwillee, 2019-04-29:
@auturgy and @peterbarker probably the best people for this. Generally speaking the systems use the same messages to achieve the same things, and will often use the same "main" message sequence. They may however use different sequences for edge cases, or use things that are not clearly defined in the spec.

The ones with highest impact are parameter protocol, mission protocol and command protocol. 
- Param protocol *high level* differences here: https://mavlink.io/en/services/parameter.html#implementations
- Mission protocol high level differences: https://mavlink.io/en/services/mission.html#implementations
- Command protocol differences are not documented yet (so I don't know what they are). The main difference is that I believe ArduPilot does not use the LONG form of commands, so suffers rounding errors for positional messages. 

In addition, ArduPilot uses different modes - ie Guided vs Offboard and may require different sending rates when in these modes. 

I expect you'll discover lots more. Let me know - then I can document them :-)

auturgy, 2019-04-29:
Just so they’re listed: Peter and Sid (@bugobliterator) have both done relevant work, although quite some time ago. Someone might be able to build on one of them.

https://github.com/peterbarker/DroneCore/tree/ardupilot

https://github.com/bugobliterator/DroneCore/tree/pr-apm-dronecore

Regards,

James

JonasVautherin, 2019-05-24:
Also, for testing: is there a SITL that can run headless (e.g. on a CI server) for APM? We will need something like that to run integration tests.

auturgy, 2019-05-24:
http://ardupilot.org/dev/docs/the-ardupilot-autotest-framework.html

fnoop, 2019-08-10:
Hi, has there been any movement/development on this?

JonasVautherin, 2019-08-11:
Unfortunaly, not that I am aware of. On my end, I would still like to have an idea of the differences between PX4 and APM, in order to open a discussion also at the MAVLink level. IMO MAVSDK should implement the "standard" set of messages (presumably `common.xml`), and allow for "unofficial" plugins supporting dialects.

APM does implement parts of `common.xml`, and therefore parts of MAVSDK should already work with it. It would be awesome to try to make a proof of concept of that before thinking about how to deal with the differences.

What do you think @fnoop? Could you help with that?

hamishwillee, 2019-08-11:
@JonasVautherin Just FYI, ArduPilot differs in (at least) Mission and Parameter protocols. Probably also in the command protocol, but I haven't documented that properly, so no detail.

Anyway, you can see quite a lot of detail about the differences in the service docs [mission](https://mavlink.io/en/services/mission.html#ardupilot) and [parameter](https://mavlink.io/en/services/parameter.html#ardupilot). @DonLakeFlyer and @WickedShell are both very good people to talk to about these. 

Also of interest is that @tridge and @dagar  been talking about updating the parameter protocol for some time, and so this might provide us a road forward for a common implementation there. Don't hold your breath though.

@peterbarker has done proof of concept plugins for an older version of the SDK which may work as a starting point.

SamuelDudley, 2019-10-08:
Hi All,
I'm looking to help out with this issue. It seems that a compatibility check between ArduPilot and MAVSDK (in its current form) needs to be undertaken. Ideally we can come up with a set of functionality tests to track development...

These are my current thoughts / action plan:
1. Setup a CI server with MAVSDK and ArduPilot SITL.
2. Write tests to determine current MAVSDK functionality against Ardupilot.
3. Write plugins (or modify ArduPilot, or another approach TBD) for MAVSDK to bridge the gap where items in 2 are failing.

SamuelDudley, 2019-10-17:
Initial testing of MAVSDK examples and integration tests result in a segfault occurring in the ardupilot SITL application. 

Looking into why this might occur...

SamuelDudley, 2019-10-17:
[New Thread 0x7ffffab80700 (LWP 8519)]
[New Thread 0x7ffffa370700 (LWP 8520)]
[New Thread 0x7ffff9b60700 (LWP 8521)]
[05:50:50|Debug] Ignoring msg 30 (system_impl.cpp:179)
[05:50:50|Debug] Ignoring msg 33 (system_impl.cpp:179)                                                                  [05:50:50|Debug] Ignoring msg 1 (system_impl.cpp:179)                                                                   [05:50:50|Debug] Ignoring msg 125 (system_impl.cpp:179)                                                                 [05:50:50|Debug] Ignoring msg 62 (system_impl.cpp:179)                                                                  [05:50:50|Debug] Ignoring msg 42 (system_impl.cpp:179)                                                                  [05:50:50|Debug] Ignoring msg 74 (system_impl.cpp:179)                                                                  [05:50:50|Debug] Ignoring msg 36 (system_impl.cpp:179)                                                                  [05:50:50|Debug] Ignoring msg 65 (system_impl.cpp:179)                                                                  [05:50:50|Debug] Ignoring msg 27 (system_impl.cpp:179)                                                                  [05:50:50|Debug] Ignoring msg 116 (system_impl.cpp:179)                                                                 [05:50:50|Debug] Ignoring msg 129 (system_impl.cpp:179)                                                                 [05:50:50|Debug] Ignoring msg 29 (system_impl.cpp:179)                                                                  [05:50:50|Debug] Ignoring msg 24 (system_impl.cpp:179)                                                                  [05:50:50|Debug] Ignoring msg 2 (system_impl.cpp:179)
[05:50:50|Debug] Ignoring msg 136 (system_impl.cpp:179)                                                                 [05:50:50|Debug] Ignoring msg 32 (system_impl.cpp:179)                                                                  [05:50:50|Debug] Ignoring msg 241 (system_impl.cpp:179)                                                                 [05:50:50|Debug] Ignoring msg 147 (system_impl.cpp:179)                                                                 [05:50:50|Debug] Forwarding msg 0 to 140737287032224 (system_impl.cpp:157)

gdb is not being kind to me but it looks like ardupilot is bombing out when the forwarded message is received....

julianoes, 2019-10-18:
> I will have to add some debugging on the ardupilot side to see why the heartbeat (msgid = 0) is causing this issue.

Heartbeat from MAVSDK?

SamuelDudley, 2019-10-19:
Yeah, looks that way.

JonasVautherin, 2020-06-12:
It seems that one aspect on which MAVSDK relies and where PX4 and APM differ is the flight modes. Flight modes are not part of the MAVLink specification, and MAVSDK currently uses the PX4 definitions (because it is developed and tested against PX4). However, there has been work towards specifying common flight modes (see e.g. [here](https://docs.google.com/document/d/1LIcfOL3JrX-EznvXArna1h-sZ7va7LRTteIUzISuD8c/edit)).

The current state, as far as I can tell, is that the PX4 community is open to working on a flight modes specification and updating PX4 to comply with it, but this cannot be done unilaterally. Anybody wanting to work on that part on the ArduPilot side would be more than welcome to the discussion :blush:.

auturgy, 2020-06-13:
I’m part of the discussion.
ArduPilot already implements modes via mavlink, and has done for many years. This might be a useful reference for you:
https://ardupilot.org/dev/docs/apmcopter-adding-a-new-flight-mode.html

Regards,

James

bazfp, 2021-05-03:
I identified a bug where Ardupilot throws an error when using MAVSDK NaN blanked `command_long`

https://github.com/ArduPilot/ardupilot/issues/17288

I have testied this with a naive patch to resolve it and and MAVSDK seems to identify Ardupilot autopilot. However actions (arm, takeoff + telemetry) are pretty broken as they wait for specific state Ardupilot never delivers (PX4 params?)

Following from this I would like to see an abstraction API put into MAVSDK so autopilot implementation quirks be handled, alternatively I would like to see MAVLink standard define a lot more of the implementation so these implementation quirks don't exist in the first place :)

ykhedar, 2022-01-11:
> Hi All, I'm looking to help out with this issue. It seems that a compatibility check between ArduPilot and MAVSDK (in its current form) needs to be undertaken. Ideally we can come up with a set of [...]

Hi,
  is Pt. 1 already done? If yes where I can see it? If not then I would like to contribute on this. I have created a docker image containing the ardupilot sitl (Only for the Rover Firmware at the moment but easily expandable.). 

How would you prefer to integrate it into the MAVSDK CI tests?  Should I provide the docker file like the PX4 sitl at [MAVSDK Github Actions PX4-SITL](https://github.com/mavlink/MAVSDK/blob/main/docker/Dockerfile-Ubuntu-20.04-PX4-SITL-v1.11) or simply the docker hub tags are sufficient.

The github repo to the current development is at: [Ardupilot SITL Headless](https://github.com/ykhedar/ardupilot-sitl-headless). The project was inspired by another such development from [radarku](https://github.com/radarku/ardupilot-sitl-docker). The main difference is the multi-stage docker build approach to bring the image size from 4GB to 700MB.

bazfp, 2022-01-11:
An Ardupilot MAVSDK integration testing suite does not exist at the moment for MAVSDK and would be much appreciated.

I have used the radarku SITL image for my personal testing and it has worked well. If you are happy to contribute your docker file that would be useful.

Then we can set up some integration tests to work with GitHub actions and the new Ardupilot dockerfile

ykhedar, 2022-01-11:
Happy to contribute. Need some guidance on next steps though. Should I create a pull request with the addition of docker files for Ardupilot SITL and changes in the github action file to include the Ardu-SITL jobs as well?

prokas-nikos, 2022-03-09:
Hi,

Is there any progress to this topic? I would like to know the status and how can I contribute.

julianoes, 2022-03-10:
Hi @prokas-nikos thanks for writing. So, in general, we want to try to minimize specific hacks and move towards a spec that both flight stacks are compatible with. These discussions and work is generally done in the MAVLink call every two week.

One thing that is being standardized is [common flight modes](https://github.com/mavlink/rfcs/pull/17), so one goal would be to implement that for ArduPilot, PX4, MAVSDK, etc..

Apart from that, in the meantime, you can always test specific functionality that you need/use with MAVSDK and check whether it works, and if not, fix it ;).

prokas-nikos, 2022-04-11:
Hello all,

I am noticing some actions about this topic. I can see that in Action Class there is a mapping between Ardupilot Modes and PX4 Modes. I am trying to send mission commands to ArduPilot but I fail. I just want to ask if there is a release notes document where we can track the changes regarding the topic of Ardupilot functionalities in MAVSDK and know if I have to implement something on my own or I am misusing the MAVSDK api.

JonasVautherin, 2022-04-11:
@prokas-nikos I would suggest you write a small example showing what you are trying to do, open a new issue with it, and explain what you expect and what happens instead :blush:. That will help people understand your problem better, and possibly help you :+1:.

bazfp, 2022-04-11:
I will tell you that I run Ardupilot SITL missions using MAVSDK

JonasVautherin, 2022-06-14:
There is some support for Ardupilot now and we even run some integration tests against Ardupilot in the CI.

I'd say we can close this one. PRs are welcome when needed or if contributors want to enable more integration tests in the CI.

closed 2022-06-14 by JonasVautherin

LiangHuangBC, 2022-11-08:
What is the "some support" any doc about this?  I tried gotoLocation and mission, both are not working for Ardupilot. But same code are working well on px4.

prokas-nikos, 2022-11-08:
@LiangHuangBC  I found out that for ardupilot the working lib was missionraw

kripper, 2023-02-18:
Can you please comment about this specific situation in https://github.com/mavlink/MAVSDK/issues/1990? Thanks.

ykhedar, 2023-05-30:
Hi @OliverHeilmann, 
target system and component is something which is added by mavlink sender automatically before sending actual command and we don't need to add that in the mission item. 

Looking closer at the warning, it is caused by this for loop:

https://github.com/mavlink/MAVSDK/blob/5eaf2cc929ba25903407943a4a90e2fa4157496c/src/mavsdk/core/mavlink_mission_transfer.cpp#L247

Now lot of ifs: This seems to be added few months before or at least after September 2022 which is the last time I checked the mission raw worked fine for Ardupilot and use the server built at that commit version which continues to work for me. If you are using the newest mavsdk server it should contain this check which basically checks if the index of the mission item list is always one greater than the value of seq in that mission item. It seems reasonable and should not cause problem as long as you are using the exact same code to create the mission plan. So I would ask you to check the following:

1. Are you using the exact same example code to upload the mission?
2. Are you able to disable this check, build the server yourself and upload the mission successfully? If yes I would request you to print the mission items in the abovementioned for loop to see what is going on. 

My only guess is somehow while marshal/unmarshal of the mission item, the seq value at mission item 1 is dropped since protobuf likes to not send over default values and we are not handling it correctly in the check mentioned above. I might able to have a look at it in a few days or sooner, until then if you can identify the problem would be nice ;)

Best,
khedar

ykhedar, 2023-05-30:
Alright so i did have a look and to me the check for ardupilot mission seems not to be correct. ArduPilot Mission Plan sequence starts at 0 just like PX4, its just that the item 0 should be home position and item 1 should be takeoff and then follows other mission commands. 

@julianoes since you implemented this check way back, just wanted to ask if I am missing something here? If not then I would simply remove the if/else blocks and check if seq is starting at 0 and increments monotonically.

julianoes, 2023-05-30:
Ok, so that's not matching what I was told when I first asked about this. And it did sound very odd to me.

I'm happy to change it if it's really wrong. It would be nice to get some sort of confirmation or proof though.

OliverHeilmann, 2023-05-31:
I followed your suggestions @ykhedar and commented out the check you linked, built the project again and then ran the test that you have [posted above](https://github.com/mavlink/MAVSDK/issues/728#issuecomment-1307368833).

Looks like the mission/ flight plan is uploaded correctly with the fix (I added a comment with lots of ! to clearly mark the point). Arm then fails but this is a gyro inconsistency issue which does not seem code related - will investigate further and post anything relevant.

```
04:53:19|Info ] MAVSDK version: v1.4.12-dirty (mavsdk_impl.cpp:20)
[04:53:19|Info ] New system on: 127.0.0.1:40465 (with sysid: 1) (udp_connection.cpp:192)
[04:53:19|Debug] New: System ID: 1 Comp ID: 1 (mavsdk_impl.cpp:496)
[04:53:19|Debug] Component Autopilot (1) added. (system_impl.cpp:377)
[04:53:20|Warn ] Vehicle type changed (new type: 2, old type: 0) (system_impl.cpp:225)
[04:53:20|Debug] Discovered 1 component(s) (system_impl.cpp:578)
Waiting for system to be ready 
Waiting for system to be ready 

 !!!!!!DISABLED MISSION CHECK!!!!!! 

[04:53:21|Debug] MAVLink: info: Flight plan received (system_impl.cpp:250)
/home/oliver/.../Tests/px4sitl.cpp:144: Failure
Expected equality of these values:
  action.arm()
    Which is: Command Denied
  Action::Result::Success
    Which is: Success
[04:53:21|Debug] MAVLink: critical: Arm: Gyros inconsistent (system_impl.cpp:250)
```

@julianoes  thanks for looking into this as well. I see you merged and [posted](https://github.com/mavlink/MAVSDK/issues/2023#issuecomment-1569309210) says the below. What does this mean in reference to a fix going forward? Thanks!

> I was wrong. The sequence just increases normally from 0, even if 0 is the home position.

ykhedar, 2023-05-31:
Hi, @OliverHeilmann, the PR https://github.com/mavlink/MAVSDK/pull/2064 was the fix to the problem and it was merged, so you should be able to compile the latest master branch and should not see the problem. 

Regarding the set_takeoff_issue, i guess in rover that command is not supported i guess :D Since rover cannot change takeoff altitude. So its fine to comment that out.