mavlink/MAVSDK #737: Health not clearing cache

AuterionWrikeBot, 2019-04-30:
Telemetry Health Subscription does not clear cache when aircraft is disconnected, so if you turn aircraft off and on again, it will say it has home position and global position even though it has not locked GPS yet. This will prevent vehicle form taking off.

JonasVautherin, 2019-04-30:
Duplicate of #736.

closed 2019-04-30 by JonasVautherin

AuterionWrikeBot, 2019-04-30:
➤ Jonas Vautherin commented:

@julianoes: how does a plugin know the vehicle is disconnected?

AuterionWrikeBot, 2019-04-30:
➤ Julian Oes commented:

Good find @douglaswsilva.

@JonasVautherin we should reset in here:
https://github.com/Dronecode/DronecodeSDK/blob/0e6a1aeb6ce3132e04ee922720dfc875ea138bf1/plugins/telemetry/telemetry_impl.cpp#L126

julianoes, 2019-04-30:
What's wrong with this bot!