mavlink/MAVSDK #745: Rename to MAVLink SDK

JonasVautherin, 2019-05-10:
There are some concerns with the "Dronecode" branding of the project, such as:

* Confusion between the Dronecode Foundation and the SDK
* Confusion on slack (#sdk vs #dronecode)
* Makes it sound like it is limited to PX4 and may be offensive to other communities (even though there is some work regarding adding support for APM in #728)

The proposition here is to rename it to "MAVLink SDK", as it is a MAVLink implementation and a high-level API in different languages. As a consequence, the project would move to the MAVLink organization.

And RFC has been open on the [MAVLink repo](https://github.com/mavlink/rfcs/pull/15) to see if that would be acceptable from the MAVLink side.

Please feel free to share your opinion and discuss it here :blush:!

laurenwinter, 2019-05-10:
This is a good clarification and moving it to MAVLink sounds like the best solution. I'd like to add that as a MAVLink SDK it will need MAVLink command passthrough functions to provide complete MAVLink functionality.

julianoes, 2019-05-13:
> I'd like to add that as a MAVLink SDK it will need MAVLink command passthrough functions to provide complete MAVLink functionality.

That's not the intent and not easily possible. The intent is to provide an SDK with certain functionality which uses MAVLink under the hood for the communcation to the drone. However, it does not necessarily expose the MAVLink API. If the raw MAVLink API is required, an implementation of MAVLink in a specific programming language should be used instead.

For the C++ library it is possible to export the C API as a passthrough / backdoor. However, if this API would be passed on to the language bindings, either all of the MAVLink messages would need to be added to the protobuf messaging or everything would need to be assembled at runtime which would require the SDK to parse the whole xml MAVLink definitions.

JonasVautherin, 2019-05-13:
Thanks @laurenwinter for the comment!

I agree with Julian, though, and I would like to add that IMO, the passthrough is not a very nice way to provide "complete MAVLink functionality" (as in: not super elegant, not community-friendly, and soon not faster). Let's compare what it (currently) takes to add a feature to the SDK and what it takes to use the passthrough (again, that's my opinion and maybe I am missing something there):

### SDK

1. Define the API as a proto definition
2. Implement the mavlink logic in a C++ plugin
3. Implement the gRPC server boilerplate to bring that to the language bindings (Swift, Python, Java, ...)
4. Auto-generate the language bindings

### Passthrough

A. Define the interface for the new features (not necessary, but any readable piece of code would likely do it: that is about putting one feature in one function)
B. Implement the mavlink logic through the passthrough

Note that:

* 1 is trivial and corresponds to A.
* 2 corresponds to B.
* 3 is currently "boilerplate" to write, but we are working on having that auto-generated (that's one of the top priorities). Once finished, it will be merged with 4.
* 4 comes almost for free ("just run the auto-generation script").

As a conclusion, I understand that the passthrough may seem faster to implement now (because of 3), but we are putting a lot of effort into removing the costs related to 3.
Also, what is implemented into the SDK is "contributed" to the community in all the supported languages, whereas whatever is written with the passthrough will never come back upstream, and is technically limited to C++. And in that sense, using the passthrough goes "against" the community.

Not sure if that makes sense, but under those conditions, I don't see a use-case that actually wants the mavlink passthrough. What you want is the full auto-generation, and we are working hard on that! :blush:.

JonasVautherin, 2019-05-13:
Back to the topic of this issue, it seems to have more activity on the [MAVLink RFC](https://github.com/mavlink/rfcs/pull/15), so let's close this one in favor of the RFC.

closed 2019-05-13 by JonasVautherin