mavlink/MAVSDK #749: Migration to MAVLink organisation

JonasVautherin, 2019-05-16:
Copying @hamishwillee's message on the PX4 Slack, to keep a record of the discussion.

The MAVLink dev team accepted the "MAVLink RFC Proposal" so we'll be taking the Dronecode SDK into the MAVLink organisation and renaming it to the "MAVSDK": https://github.com/mavlink/rfcs/blob/master/text/0011-mavlink-sdk.md

### Code

- [ ] Agree new names for SDK repos
- [ ] Move SDK repos into github mavlink org
- [ ] Agree new names for namespaces and other code items
- [ ] Update code in line with new names
- [ ] Update READMEs with appropriate new links to repos and docs
- [ ] Make an official first release. Would this be 1.0?
- [ ] Update all the proto/language repos
- [ ] Cut release for those too - e.g. Python

### Docs

- [ ] Confirm name of docs repo in mavlink org [I think `sdk_docs` to `mavsdk_docs`]
- [ ] Move docs repo into mavlink org
- [ ] Update head revision docs API reference
- [ ] Update docs guide materials
- [ ] Review/update docs on marketing/positioning. Ie where it is hosted and the goals.
- [ ] Decide what to do about previous releases/docs. These all will refer to the Dronecode SDK.
 - [ ] Propose that we delete all the old ones and start from scratch with a master release and initial release
- [ ] Re-sync via gitbook
- [ ] Get new domain/URL. 
        Current: `https://sdk.dronecode.org/en/`. Proposed: `https://mavsdk.mavlink.io/en/`. This is probably easier than try to do `mavlink.io/mavlink`, `mavlink.io/mavsdk` etc.
- [ ] Themes updates
        The theme for the SDK becomes more like the mavlink.io one: https://mavlink.io/en/
- [ ] Other dronecode sites will need to have theme updated to point to SDK and any notes about SDK updated too.

### DroneCode Website

- [ ] Update top bar
        This is likely to be SLOW process.
- [ ] Update SDK page and other mentions.

JonasVautherin, 2019-05-16:
- SDK repos: I propose `mavsdk-proto`, `mavsdk`, `mavsdk-swift`, `mavsdk-python`, `mavsdk-java`, `mavsdk-js`.
- Namespaces, code items: I would use "mavsdk" (lowercase) as much as possible.
- First release: I wouldn't synchronize the github releases between the different repos. Also, I wouldn't make 1.0 just because of a name change. I would move to 1.0 once all the API is autogenerated (C++ is the next and last one coming for that). However, we should release packages when possible (I'm thinking about CocoaPods and python-pip here).

julianoes, 2019-05-16:
For the SDK repos I'd prefer `MAVSDK-Proto`. Although it depends on the font and logo as well there.

hamishwillee, 2019-05-16:
@julianoes and @JonasVautherin Why don't you sit down together and agree what you want to do - and update the post/s above appropriately :-)

FWIW
- SDK repos. I'm ambivalent. I slightly lean towards MAVSDK  _in text_ when referring to the project and product. However I don't extend that desire to package or repo names. Ie. We could use mavsdk-proto for the repo URL and still refer to it in the readme as MAVSDK-Proto.

hamishwillee, 2019-05-16:
We need a new URL. Anyone got objection to https://mavsdk.mavlink.io/en/ (or better suggestions)?

We need a new Logo, presumably with a new tagline - maybe something "like":
   [image]

@LorenzMeier @mrpollo Are either of you able to help with the above - ie resourcing/making it happen?

bys1123, 2019-05-18:
Maybe the slack channel #sdk also need rename to #MAVSDK

mrpollo, 2019-05-18:
Just so we are clear @hamishwillee, you are asking for help creating a logo for MAVSDK?

hamishwillee, 2019-05-20:
> Just so we are clear @hamishwillee, you are asking for help creating a logo for MAVSDK?

@mrpollo Yes. 

That logo might include a tagline like "MAVLink APP and Integration Toolkit". If it includes a tagline that also need to be discussed/agreed.

julianoes, 2019-05-20:
> "MAVLink APP and Integration Toolkit"

APP -> App, it's not an acronym. I'm not super happy with the tagline but I can't think of something better yet.

bys1123, 2019-05-20:
[image]
I made a logo for you to choose. Hope you like it.

bys1123, 2019-05-20:
[image]
I made a logo for you to choose, hope you like it.

hamishwillee, 2019-05-21:
@julianoes I'm quite happy with @bys1123 "MAVLink Integration Toolkit" (though to me "APP" is fine in the previous case, because it is well understood.

I'd love a designer in on the logo to confirm colours etc. Stone, I don't hate the icon, but "what are you trying to say" - ie this is not like the plane with radio comms logo.

bys1123, 2019-05-21:
@hamishwillee Yes, the color not well adjustment yet.
The reason why it's looks like this:
0. When people talk about drone, This first thing come out in their mind is Quadcopter. So, the main logo is quad arrow. So, it makes people feels right on the outline. (We don't need to say this public)
1. This sdk is about control most of time, makes your MAV go some where .
2. Out of box.
3. Colors means it support different programming languages, also have many good features. 
4. User-friendly. Lots of programmer in other industry start learning control drones through this SDK.

bys1123, 2019-05-21:
Consistent:
1. MAVLink logo is made about ten years ago. The draft feeling is not fit the power of this brand and project.
2. Adobe, Autodest, STM32 changed lots of logo in past years, "industrial feeling" is changed.
3. So, this MAVSDK logo is for fit the new trend of the industrial logo design. I tried my best make it will fit the new version MAVLink logo that will come out at some point by someone.

mrpollo, 2019-05-21:
This thread is already discussing a lot of different topics, and the conversations are tough to follow. Thanks, @JonasVautherin, and @hamishwillee for putting this together, there are lots of areas of concern here, I recommend we create a Project Board and move the focused discussions elsewhere, per-issue if possible, creating the least overhead to everyone involved. 

---

Thanks for the high energy and fast responses @bys1123, at this point, we are not ready for contributions, for the logo, tagline, or any branding collateral.

hamishwillee, 2019-05-21:
> , I recommend we create a Project Board and move the focused discussions elsewhere, per-issue if possible, creating the least overhead to everyone involved.

Thanks @mrpollo 

On the technical aspects @julianoes and @JonasVautherin should just make some decisions and let us know. That is enough for all the technical and docs side to proceed and does not require discussion with anyone. Though if they want to rubber stamp those, we can discuss in next weeks MAVLink meeting. 

Julian, Jonas, can you confirm the names of repos, and name changes you want? Essentially summarise and provide a combined response for https://github.com/Dronecode/DronecodeSDK/issues/749#issuecomment-492962221 and https://github.com/Dronecode/DronecodeSDK/issues/749#issuecomment-492974967

For non technical aspects, yes, makes sense to at least have issues for separate tracking (because these require a lot of chit-chat).

I have created https://github.com/Dronecode/DronecodeSDK/issues/750 for the logo. It has one question that should be answered before it gets discussed - who is going to do the final design work.

JonasVautherin, 2019-05-23:
I edited the issue after discussions with @julianoes. I believe we are aligned on the "code" points, and will start renaming after #724 is merged (should be next week). I believe we could start moving before that, as the redirection should make it transparent.

> Julian, Jonas, can you confirm the names of repos, and name changes you want?

That's edited in the issue:
* The official name for communications/docs is `MAVSDK`.
* Repos will be named `MAVSDK`, `MAVSDK-Proto`, `MAVSDK-Swift`, `MAVSDK-Python`, etc.
* In the code, we will prefer lowercase for readability (e.g. `mavsdk::` as a namespace, and probably `class MavSdk` for a class.

JonasVautherin, 2019-05-23:
@mrpollo: Would you mind renaming #sdk to #mavsdk on Slack (last point on the list above)?

JonasVautherin, 2019-05-23:
@hamishwillee: I mostly agree with the Documentation points above, I'd like clarification on:

> * Decide what to do about previous releases/docs. These all will refer to the Dronecode SDK.
> * Propose that we delete all the old ones and start from scratch with a master release and initial release

So you suggest that the mavsdk docs start with master on the mavsdk repo, and we remove all the previous releases because the documentation says "DronecodeSDK" everywhere and we don't want to create confusion or go edit all occurrences there. Correct? I would be fine with that as well.

hamishwillee, 2019-05-23:
> So you suggest that the mavsdk docs start with master on the mavsdk repo, and we remove all the previous releases because the documentation says "DronecodeSDK" everywhere and we don't want to create confusion or go edit all occurrences there. Correct? I would be fine with that as well.

@JonasVautherin Yes! I'd also like to do an official release "as an SDK" as soon as possible once we move to MAVSDK.

Re the docs, are you OK with repo move from sdk_docs to mavsdk_docs or would you prefer consistency MAVSDK-Docs? 
FYI I like the consistency. The reason I tend to use underscores because when you select a bunch of text for copying it tends to break on "-" but not _ (may seem trivial, but I do a lot of text selection/copy/paste).

JonasVautherin, 2019-06-06:
I moved the Java SDK for a first test, seems alright: https://github.com/mavlink/MAVSDK-Java

hamishwillee, 2019-06-07:
I moved the docs repo https://github.com/mavlink/MAVSDK-docs but now have lost admin rights and have broken auto builds from legacy gitbook.com. Chasing @dagar to build locally.

julianoes, 2019-06-07:
@hamishwillee I made you admin for MAVSDK-docs.

JonasVautherin, 2019-06-07:
@hamishwillee: FYI, all the repos are now in the mavlink org :+1:.

hamishwillee, 2019-06-11:
Awesome. @JonasVautherin FYI I can't tick off my items above now because I don't have needed rights in this repo. I have moved docs to MAVSDK-docs. Note that we can now no longer build them. Waiting on Dagar for support.

JonasVautherin, 2019-06-11:
@hamishwillee: Just gave you the rights!

julianoes, 2019-07-01:
@hamishwillee this page now 404s https://sdk.dronecode.org/ Is it possible to have it forward to the new docs page, best case including the path so that each link goes to its new page?

hamishwillee, 2019-07-01:
@julianoes Not by me (limitations of legacy gitbook). See if you can get @LorenzMeier to do the redirect at the domain level.

hamishwillee, 2019-07-03:
@JonasVautherin This can be closed now I think!

JonasVautherin, 2019-07-03:
Agreed. There is still some renaming going on in the code but that will come eventually :-).

closed 2019-07-03 by JonasVautherin