mavlink/MAVSDK #750: Create MAVSDK Logo

hamishwillee, 2019-05-21:
We need to have a new logo (and tagline) for the MAVSDK.

This should share "common design language" with MAVLink logo, and possibly both should be updated.

If this logo has a "tagline" included (like the current MAVLink logo, then this will have to be chosen

There are some discussions of this in #749 - e.g.:
https://github.com/Dronecode/DronecodeSDK/issues/749#issuecomment-494126762
https://github.com/Dronecode/DronecodeSDK/issues/749#issuecomment-493264694

hamishwillee, 2019-05-21:
@mrpollo @LorenzMeier Do we have access to professional design resources to support this task?

mrpollo, 2019-05-22:
Dronecode will provide any resources necessary to MAVSDK, this includes but is not limited to a graphic designer.

hamishwillee, 2019-05-22:
Kevin Sartori proposed that we use the same MAVLink logo with appropriate/slight modifications. His argument being that MAVLink has a strong existing brand. FWIW I agree - I don't love the iconography, but it is not a priority to revisit it right now.

I therefore revert proposal to my original one, but using @bys1123's proposed tagline. We could get @mrpollo's designer to produce a "proper" version using correct fonts, colours, whatever.

[image]

Any strong disagreement with "something like this"? Assuming not - are we OK with the tagline?

JonasVautherin, 2019-05-22:
I don't have a super strong opinion there, honestly. @julianoes any comment?

hamishwillee, 2019-05-23:
@LorenzMeier @auturgy @mrpollo OK with above logo/tagline (tidied of course)?

auturgy, 2019-05-23:
I have no strong feelings on the logo/branding but marketing it as a mavlink toolkit, it needs to be tested against more than just px4: ie at least another common.xml user.

sartorikevin, 2019-05-23:
I think it's interesting what Red Hat did with their new logo. You clearly recognize the company, however, the logo looks much more modern. This could be the opportunity to do this with the MAVLink logo.

[image]

On the tagline: Toolkit seems to me a loose term to refer to any collection of “tools” (another loose term) that have a common goal. Does that conflict with the term SKD that is also in the logo?

hamishwillee, 2019-05-23:
> I think it's interesting what Red Hat did with their new logo. You clearly recognize the company, however, the logo looks much more modern. This could be the opportunity to do this with the MAVLink logo.

@sartorikevin We can certainly ask the designer to modernise. I just hope not to derail the effort to migrate the SDK. 

> On the tagline: Toolkit seems to me a loose term to refer to any collection of “tools” (another loose term) that have a common goal. Does that conflict with the term SKD that is also in the logo?

Not "conflict", but it is a bit of a tautology. It may be that we don't need a tagline with a modernised logo. If we do, what I was hoping to capture was that it can be used for MAVLink integration - of components into a system and to control a system.

Happy to take suggestions????

JonasVautherin, 2019-05-23:
> Not "conflict", but it is a bit of a tautology.

@hamishwillee I think it would work without the tagline. Just `MAVSDK`. And then more details about what it is are on the webpage.

> I have no strong feelings on the logo/branding but marketing it as a mavlink toolkit, it needs to be tested against more than just px4: ie at least another common.xml user.

@auturgy: I do agree on that, though I don't think it is necessarily a prerequisite. Just like mavsdk does not support all the mavlink messages (yet), it does not support all the mavlink platforms. That will come, but it still requires some contributions. My fear is that if we say "for now it is a PX4 toolkit because it is only tested against PX4", then why would people from another platform contribute to a PX4 toolkit? If we say "it is a mavlink toolkit, but right now the contributions have been tested against PX4 only", then it is open to contributions testing against other platforms.

hamishwillee, 2019-05-23:
I'd be happy without the tagline, but that depends on what the designer says re consistency with MAVLink logo. Best to prepare as if it will be needed.

Another idea: Cross-Platfom MAVLink Microservice Library

JonasVautherin, 2019-05-23:
> Another idea: Cross-Platfom MAVLink Microservice Library

I like "MAVLink Integration Toolkit" better :smile:. I find "microservice" confusing in this context, and exposing the implementation.

hamishwillee, 2019-05-23:
@JonasVautherin @julianoes @mrpollo @sartorikevin @bys1123 

How about some suggestions from someone else?
Here is a compiled list of ideas (some old, some new):
- MAVLink App and Integration Toolkit
- MAVLink Integration ToolKit
- Cross-Platfom MAVLink Microservice Library
- MAVLink Integration Library for UAS    (or Drones)
- MAVLink Application and Integration Library
- Cross Platform Development Framework for MAVLink  (though not really a framework)
- High Level APIs for MAVLink Developers

The main ideas we might want to include being: MAVLink, Library/SDK/Tools/Toolkit/API, Integration (cameras, Higher level command and control - companion computer), MAVLink Apps, Cross platform, Higher level API than MAVLink libs, Developer Friendly, Communications.

I'll give this until Monday then ask @mrpollo to see if our designer can come up with a modernised version of current branding - both omitting tagline and using one of the current suggestions as a placeholder.

bys1123, 2019-05-24:
Wow, looks like a mind storm is happening here.

LorenzMeier, 2019-05-24:
I think all of these are just confusing and not to the point. If we update it, it should say what it is doing. It also doesn't make sense to me that the TLA is SDK and then the tagline says something else - if there is any tagline, then it needs to be software development kit.

JonasVautherin, 2019-05-24:
I would really vote for "no tagline".

I believe that "drone SDKs" are now quite known. People will refer to the "DJI SDK" and the "Parrot SDK". So I don't feel a need to explain further what a "drone SDK" is. The fact that it looks similar to the MAVLink logo is probably enough for people who know MAVLink, and people who don't know MAVLink would not understand the tagline anyway, right? :sweat_smile:

hamishwillee, 2019-05-27:
@JonasVautherin I tend to agree - in particular that the term "MAV" has a life of its own outside of "Micro Air Vehicle". I'll ask for both for comparison.

bys1123, 2019-05-27:
[image]
VS code logo upgrade is also a good reference.

hamishwillee, 2019-07-03:
All done and live.

closed 2019-07-03 by hamishwillee