mavlink/MAVSDK #754: Expose angular velocities and acceleration in telemetry

julianoes, 2019-05-30:
It would probably be nice if angular velocities and also acceleration would be exposed in telemetry. I have heard people asking for this quite a bit.

This data could be received by the [HIGHRES_IMU](https://mavlink.io/en/messages/common.html#HIGHRES_IMU) message and added to the telemetry plugin in the same way as other data is exposed there.

krady21, 2019-06-27:
I am working on solving it. Should both angular speed and acceleration be a part of the same struct in telemetry.h or do i have to make one struct for each of them?

JonasVautherin, 2019-06-27:
> Should both angular speed and acceleration be a part of the same struct in telemetry.h or do i have to make one struct for each of them?

I believe I would put them in the same struct, as they make sense together. It's easier to ignore unwanted info than to merge separate messages later. Do you have an opinion there?

krady21, 2019-06-27:
I agree with you. I guess I will emulate what has already been done [here](https://github.com/mavlink/MAVSDK/blob/e6b98aa1910b1f5cbf84449f72b4bbc50d7ef023/src/plugins/telemetry/include/plugins/telemetry/telemetry.h#L53-L79) with PositionNED, VelocityNED and PositionVelocityNED

PR #779 (2019-07-09): Telemetry: add angular velocity and acceleration
1. adds struct IMUReadingNED encompassing acceleration ,angular speed, magnetic field and temperature
2. adds setters and getters(both synchronous and asynchronous for interacting with acceleration and angular speed) and a process function for getting the desired info from the mavlink message.

Resolves #754 and #651
closed 2019-07-09 by julianoes