mavlink/MAVSDK #780: Support indoor VIO/flow setups

julianoes, 2019-07-01:
It would be nice if MAVSDK was compatible with setups without GPS (so no global position) for indoor use based on Flow or VIO localization.

hamishwillee, 2019-07-02:
That would be awesome. I assume we can simulate this already in Gazebo and test where it fails?

julianoes, 2019-07-02:
> I assume we can simulate this already in Gazebo and test where it fails?

In theory yes. I've never actually been able to use Flow in Gazebo.

shrit, 2019-07-02:
Yes, I totally agree, since it is not possible to fly it with the actual flying mode.
You have once pointed to me that we need to comments some of these lines to fly without GPS in Gazebo.
It would be also nice if the user can be able to recover the video from the optical flow in real time. In case if we are using a pico quad copter that can not support another camera

https://github.com/mavlink/MAVSDK/blob/4ea885ac0a3e8a6ff533d5c58be9c24ecb15a913/src/plugins/action/action_impl.cpp#L53-L73

hamishwillee, 2019-07-02:
> It would be also nice if the user can be able to recover the video from the optical flow in real time. In case if we are using a pico quad copter that can not support another camera

I would address that completely separately. But in any case this is only a "MAVLink/MAVSDK" question if the optical flow camera supports the MAVLink Camera API and exposes the video feed via an RTSP stream. Otherwise there is no way for the SDK to know about the output of optical flow.

hamishwillee, 2019-07-02:
>> I assume we can simulate this already in Gazebo and test where it fails?

> In theory yes. I've never actually been able to use Flow in Gazebo.

Fair enough. I'd say getting this working and in CI would be a precondition for doing anything else.