mavlink/MAVSDK #783: Add brew install for lib

julianoes, 2019-07-03:
It would be nice to be able to this on macOS:
```
brew install mavsdk
```

krady21, 2019-07-17:
I am looking to solve this! Should we add the future homebrew formula to homebrew-core or should we make and maintain our own tap? I prefer the first alternative after reading [this](https://discourse.brew.sh/t/pushing-to-homebrew-core-vs-creating-a-new-tap/1121).

JonasVautherin, 2019-07-17:
> I prefer the first alternative after reading this.

Same for me :+1:. Let us know how it goes!

@unipheas: FYI

unipheas, 2019-07-18:
> A good rule of thumb is: is your software included in another package manager like Debian? If so, we’ll also package it, if not please consider creating your own tap. Thanks!

I think for now maintaining our own tap is ideal. Maybe we can take a vote on this in slack or something? That said, I'd like to believe MAVSDK is really popular and essential to the community 😁

JonasVautherin, 2019-07-18:
> I think for now maintaining our own tap is ideal.

What's the advantage of that? I was naively thinking that adding the formula to homebrew-core would mean less maintenance on our side :laughing:.

unipheas, 2019-07-18:
>  I was naively thinking that adding the formula to homebrew-core would mean less maintenance on our side 😆.

That's actually a good point. It won't hurt to try to get it on core. Worst that happens is they deny it.

daniel-bryant, 2019-10-17:
### Progress
I just created a formula that I tested and used to install mavsdk on my machine. A few things:
- I named the formula "mavsdk", but "mavsdk-core" _could_ make sense
- Installation takes quite a bit longer than most homebrew packages ("built in 20 minutes 20 seconds")
- I had to fork the sdk and create my own release bc the source code under the releases here don't include submodules

here's the tap/formula: https://github.com/daniel-bryant/homebrew-mavsdk
here's my forked release: https://github.com/daniel-bryant/MAVSDK/releases/tag/v0.20.0

### Next steps
More developers should test the formula. You need to **not** have mavsdk currently installed system-wide. We should update the documentation to use this tap (or another) first so that the general public starts using it as well. Once it's been proven stable, we can make the PR for homebrew core.

julianoes, 2019-10-17:
@daniel-bryant nice! 

> - I named the formula "mavsdk", but "mavsdk-core" could make sense

What about mavsdk-cpp?

> Installation takes quite a bit longer than most homebrew packages ("built in 20 minutes 20 seconds")

The thing here is to provide the bottle (precompiled package), I think we would want to create these automatically in CI and upload them to s3.

> I had to fork the sdk and create my own release bc the source code under the releases here don't include submodules

Interesting, how did you make sure that the zip "contains" the submodules?

JonasVautherin, 2019-10-17:
> What about mavsdk-cpp?

Does brew install libraries other than C++? And why wouldn't `brew install mavsdk` install `mavsdk_server`, as well?

In other languages, we call that it "mavsdk" (e.g. `pip install mavsdk`). Swift is an exception, but I believe we should rename `MAVSDK_Swift` in `MAVSDK`.

> The thing here is to provide the bottle (precompiled package), I think we would want to create these automatically in CI and upload them to s3.

Or can we release an artifact on github? I think people have issues accessing s3 in China.

unipheas, 2019-10-17:
@daniel-bryant If you wouldn't mind sending me a message in the slack group I can work with you and get this tested from China. 

> In other languages, we call that it "mavsdk" (e.g. `pip install mavsdk`). Swift is an exception, but I believe we should rename `MAVSDK_Swift` in `MAVSDK`.

Agreed

julianoes, 2019-10-17:
> And why wouldn't brew install mavsdk install mavsdk_server, as well?

@JonasVautherin you're right. `mavsdk` is fine. And including `mavsdk_server` would be awesome.

> Or can we release an artifact on github? I think people have issues accessing s3 in China.

Right, that would be better.

julianoes, 2019-12-10:
@daniel-bryant thanks for your work. I've taken your commit and I'm now working on this tap:

https://github.com/mavlink/homebrew-mavsdk

```
brew tap mavlink/mavsdk
brew install --verbose mavsdk
```

This seems to work for now. Next step is to check everything that homebrew needs in order to make a PR to homebrew-core.

julianoes, 2019-12-12:
Wish me luck: https://github.com/Homebrew/homebrew-core/pull/47770

julianoes, 2020-11-20:
This has been working for a while now.

closed 2020-11-20 by julianoes