mavlink/MAVSDK #784: How to set home point for indoor control with uwb ?

wuxianshen, 2019-07-04:
Hi, everyone; we are trying to build a C++ application indoor with px4 and uwb; I can push mocap position as vision pose to EKF2 ( through the MavSDK Passthrough ), but when I try to do some offboard control with mavsdk, the telemetry::health() prompts "Home position to be set"; I try to set home position with mavlink_set_home_position_t message, but it doesn't work; Has anyone met this problem before? What should I do to use mavsdk and uwb for auto control indoor? Thanks...

wuxianshen, 2019-07-04:
The uwb postion is sent as the mavlink_vision_position_estimate_t message through mavsdk::PassThrough

wuxianshen, 2019-07-04:
I follow the instructions from Support indoor VIO/flow setups #780 and comments mode set in arm();
then the vehicle can arm...

julianoes, 2019-07-04:
You can ignore this health flag and still try to arm.

However, then you need to comment out this to takeoff:

https://github.com/mavlink/MAVSDK/blob/develop/src/plugins/action/action_impl.cpp#L124-L125

wuxianshen, 2019-07-04:
Thanks Julianoe for your reply!
I comment the mode setting in action arm and takeoff, and for now the vehicle in jmavsim simulation can takeoff; And I still want to test some offboard apis before I apply the mission on px4, so I follow the example in "offboard" and call the offb_ctrl_ned function after takeoff.
During the offboard control, I continuously push fake x/y/z position as mavlink_vision_position_estimate_t message, and the x/y/z is got from mavsdk::telemetery  position_velocity_ned_async. 
But during the offboard demo, the offboard control update would failed after a while, prompting "[mc_pos_control] Position-Control Setpoint-Update failed" and "Descend with land-speed", and the vehicle landed by itself.
Is this problem caused by offboard control under no GPS mode? What should I do to fix if I want to use indoor uwb position lock and offboard control?
I'm really a green hand in px4... Looking forward to your help, thanks very very much...

wuxianshen, 2019-07-04:
`float current_pos_x = 0.0f;
float current_pos_y = 0.0f;
float current_pos_z = 0.0f;

int main(int argc, char **argv)
{
    //.....Skip some connect and init

    auto telemetry = std::make_shared<Telemetry>(system);
    auto action = std::make_shared<Action>(system);

    auto mavlink_passthrough = std::make_shared<MavlinkPassthrough>(system);

    // subscribe position info @ 50Hz
    const Telemetry::Result set_rate_result = telemetry->set_rate_position_velocity_ned(50.0);
    if (set_rate_result != Telemetry::Result::SUCCESS)
    {
        std::cout << ERROR_CONSOLE_TEXT
                  << "Setting rate failed:" << Telemetry::result_str(set_rate_result)
                  << NORMAL_CONSOLE_TEXT << std::endl;
        return 1;
    }

    telemetry->position_velocity_ned_async(local_pos_update);

    std::thread pos_pub_thread = std::thread(pos_publish_loop, &dc);

    std::cout << "Start position publish thread..." << std::endl;

    std::this_thread::sleep_for(std::chrono::seconds(2));

    std::cout << "Arming..." << std::endl;
    const Action::Result arm_result = action->arm();
    if (arm_result != Action::Result::SUCCESS)
    {
        std::cout << ERROR_CONSOLE_TEXT << "Arming failed:" << Action::result_str(arm_result)
                  << NORMAL_CONSOLE_TEXT << std::endl;
        //return 1;
    }
    else
    {
      std::cout << "Armed successfully!" << std::endl;
    }

    Action::Result takeoff_result = action->takeoff();
    if (takeoff_result != Action::Result::SUCCESS)
    {
        std::cout << ERROR_CONSOLE_TEXT << "Takeoff failed:" << Action::result_str(arm_result)
                  << NORMAL_CONSOLE_TEXT << std::endl;
        //return 1;
    }
    else
    {
      std::cout << "Takeoff successfully!" << std::endl;
    }

    std::cout << "In Air..." << std::endl;
    sleep_for(seconds(5));

    auto offboard = std::make_shared<Offboard>(system);
    offb_ctrl_ned(offboard);

    const Action::Result land_result = action->land();
    if (land_result != Action::Result::SUCCESS)
    {
        std::cout << ERROR_CONSOLE_TEXT << "Landing failed:" << Action::result_str(arm_result)
                  << NORMAL_CONSOLE_TEXT << std::endl;
    }

    // Check if vehicle is still in air
    while (telemetry->in_air()) {
        std::cout << "Vehicle is landing..." << std::endl;
        sleep_for(seconds(1));
    }
    std::cout << "Landed!" << std::endl;

    // We are relying on auto-disarming but let's keep watching the telemetry for a bit longer.
    sleep_for(seconds(3));
    std::cout << "Finished..." << std::endl;
    return 0;
}`

wuxianshen, 2019-07-04:
`
uint64_t get_cur_time_stamp_us()
{
  std::chrono::time_point<std::chrono::system_clock, std::chrono::microseconds> tp =
      std::chrono::time_point_cast<std::chrono::microseconds>(std::chrono::system_clock::now());

  return static_cast<unsigned long long>(tp.time_since_epoch().count());
}

void pos_publish_loop(Mavsdk* p_dc)
{
  System &system = p_dc->system();
  auto mavlink_passthrough = std::make_shared<MavlinkPassthrough>(system);

  unsigned cnt = 0;
  mavlink_att_pos_mocap_t att_pos_mocap;
  memset(&att_pos_mocap, 0, sizeof(att_pos_mocap));

  mavlink_vision_position_estimate_t vision_pos;
  memset(&vision_pos, 0, sizeof(vision_pos));

  mavlink_message_t msg;
  while( cnt < 1000 * 10 )
  {
    cnt ++;
    vision_pos.usec = get_cur_time_stamp_us();
    vision_pos.x = current_pos_x;
    vision_pos.y = current_pos_y;
    vision_pos.z = current_pos_z;
    mavlink_msg_vision_position_estimate_encode(0, 0, &msg, &vision_pos);

    const MavlinkPassthrough::Result pass_result = mavlink_passthrough->send_message(msg);
    if (pass_result != MavlinkPassthrough::Result::SUCCESS)
    {
      std::cout << ERROR_CONSOLE_TEXT << "Passthrough failed:" << mavlink_passthrough->result_str(pass_result)
                << NORMAL_CONSOLE_TEXT << std::endl;
      continue;
    }
    else
    {
      if ( cnt % 100 == 0 )
      {
        std::cout << "Passthrough send successfully " << cnt / 100 << std::endl;
      }
    }

    sleep_for(std::chrono::milliseconds(20));
  }
}

bool offb_ctrl_ned(std::shared_ptr<mavsdk::Offboard> offboard)
{
    const std::string offb_mode = "NED";
    // Send it once before starting offboard, otherwise it will be rejected.
    offboard->set_velocity_ned({0.0f, 0.0f, 0.0f, 0.0f});

    std::cout << "Offboard start..." << std::endl;
    Offboard::Result offboard_result = offboard->start();
    if (offboard_result != Offboard::Result::SUCCESS)
    {
      std::cout << ERROR_CONSOLE_TEXT << "Offboard start failed:" << Offboard::result_str(offboard_result)
                << NORMAL_CONSOLE_TEXT << std::endl;
      return false;
    }

    std::cout << "Turn to face East" << std::endl;
    offboard->set_velocity_ned({0.0f, 0.0f, 0.0f, 90.0f});
    sleep_for(seconds(1)); // Let yaw settle.

    {
        const float step_size = 0.01f;
        const float one_cycle = 2.0f * (float)M_PI;
        const unsigned steps = 2 * unsigned(one_cycle / step_size);

        std::cout << "Go North and back South" << std::endl;
        for (unsigned i = 0; i < steps; ++i) {
            float vx = 5.0f * sinf(i * step_size);
            offboard->set_velocity_ned({vx, 0.0f, 0.0f, 90.0f});
            sleep_for(milliseconds(10));
        }
    }

    std::cout << "Turn to face West" << std::endl;
    offboard->set_velocity_ned({0.0f, 0.0f, 0.0f, 270.0f});
    sleep_for(seconds(2));

    std::cout << "Turn to face West" << std::endl;
    offboard->set_velocity_ned({0.0f, 0.0f, -2.0f, 180.0f});
    sleep_for(seconds(4));

    std::cout << "Turn to face West" << std::endl;
    offboard->set_velocity_ned({0.0f, 0.0f, 1.0f, 0.0f});
    sleep_for(seconds(4));

    // Now, stop offboard mode.
    offboard_result = offboard->stop();
    std::cout << "Offboard stopped" << std::endl;

    return true;
}

void local_pos_update(Telemetry::PositionVelocityNED pos_vel)
{
  current_pos_x = pos_vel.position.north_m;
  current_pos_y = pos_vel.position.east_m;
  current_pos_z = pos_vel.position.down_m;
}`

wuxianshen, 2019-07-04:
The testing code is posted above if needed...
Thank you!@!

julianoes, 2019-07-05:
So the error that you're seeing is sent here in PX4:
https://github.com/PX4/Firmware/blob/dce1dda8717090bcf7d55210a1628f2d3ad91e90/src/modules/mc_pos_control/mc_pos_control_main.cpp#L611-L613

Now I would assume this happens because offboard is no longer sending setpoints, however, it should automatically do that.

I suggest that you add something like `LogInfo() << "Sent setpoint";` here:
https://github.com/mavlink/MAVSDK/blob/develop/src/plugins/offboard/offboard_impl.cpp#L347

And then watch if this is still sent when the problem happens.

Also, do you have a simulation setup to test this, e.g. in SITL Gazebo? If so you can give me the steps to reproduce it.

Another thing to check is if the link to the drone. Maybe you're sending too much over the air and it gets congested.

wuxianshen, 2019-07-05:
@julianoes
Thanks for your reply!
I met this problem in jmavsim simulation; I added a log output in the sending setpoints function and test again, the sdk is continuously sending setpoints, even when prompted "Position-Control Setpoint-Update failed" warnning.
[image]

Both the position setpoints and velocity setpoints will have this problem.

wuxianshen, 2019-07-05:
The sitl environment:
px4 firmware master branch, sudo make px4_sitl_default jmavsim
[image]
mavsdk master branch:
[image]

wuxianshen, 2019-07-05:
The only parameter configuration is the EKF2_AID_MASK, disable GPS and enable vision position fusion:
[image]

wuxianshen, 2019-07-05:
Test Source File  :
[publish_position.txt](https://github.com/mavlink/MAVSDK/files/3362287/publish_position.txt)

julianoes, 2019-07-05:
Interesting, this could be a bug. I'll try to reproduce this, thanks.

wuxianshen, 2019-07-05:
Thanks for your patience~!~

wuxianshen, 2019-07-08:
@julianoes  This problem is also occured in gazebo simulation environment. Have you reproduced it yet?