mavlink/MAVSDK #787: unit tests fail (curl)

Ahmad-Drak, 2019-07-10:
Performing unit tests after a new build and system-wide install of MAVSDK, 3 tests fail:

[----------] Global test environment tear-down
[==========] 61 tests from 12 test cases ran. (42358 ms total)
[  PASSED  ] 58 tests.
[  FAILED  ] 3 tests, listed below:
[  FAILED  ] CurlTest.Curl_DownloadFile_WithoutProgressFeedback_Success
[  FAILED  ] CurlTest.Curl_DownloadFile_ProgressFeedback_Success
[  FAILED  ] CurlTest.Curl_DownloadFile_ProgressFeedback_COULDNT_RESOLVE_HOST

any ideas?

[log.txt](https://github.com/mavlink/MAVSDK/files/3376671/log.txt)

julianoes, 2019-07-10:
Thanks for reporting, let me have a look if I can reproduce.

julianoes, 2019-07-10:
It works for me which would mean either your internet connection was flaky when you ran it or the servers were not responding for some reason.

This reminds me that we should actually move the curl unit tests to integration tests because unit tests are not supposed to need anything to run, so they should not depend on an internet connection.

closed 2019-07-10 by julianoes