mavlink/MAVSDK #800: Windows Local Install

SelimOzel, 2019-07-17:
Getting the following error:
```
json11 failed to build!
Call Stack (most recent call first):
  third_party/CMakeLists.txt:10 (build_target)
```

Following steps for Windows in [1]. Went more-or-less smooth until Local Install option. The error is from the following call:
`cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_SHARED_LIBS=ON -DCMAKE_INSTALL_PREFIX=install -Bbuild/default -H.`

[1] https://mavsdk.mavlink.io/develop/en/contributing/build.html#sdk_local_install

JonasVautherin, 2019-07-17:
There must be more logs explaining why json11 failed to build. Can you copy that here? Or worst case, link your full build log?

SelimOzel, 2019-07-17:
It was terribly long! Didn't want to copy paste a wall of logs! 

I don't have the logs open in front of me but I'll keep this updated when I try it again later on. Just wanted to update you and thanks for fast response!

hamishwillee, 2019-07-18:
@SelimOzel 

> It was terribly long! Didn't want to copy paste a wall of logs!

Fair enough. Sometimes I just post these into gists and then link.

imfatant, 2019-07-19:
I'll carry on where @SelimOzel left off since I have exactly the same problem.

Let me mention that I have to use the following in order to build in the first place:
```
cmake -G "Visual Studio 16 2019" -A x64 -DBUILD_SHARED_LIBS=ON -Bbuild/default -H.
cmake --build build/default --config Debug

```
Then, I try the following for local install with -DCMAKE_INSTALL_PREFIX=install, but notice that it ignores this, and also errors. In Command Prompt:
```
C:\Users\Anthony\git\MAVSDK>cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_SHARED_LIBS=ON -DCMAKE_INSTALL_PREFIX=install -Bbuild/default -H.
-- Selecting Windows SDK version 10.0.17763.0 to target Windows 10.0.17134.
-- Selecting Windows SDK version 10.0.17763.0 to target Windows 10.0.17134.
-- Preparing external project "json11" with args:
-- -- -DCMAKE_INSTALL_PREFIX:PATH=C:/Users/Anthony/git/MAVSDK/build/default/third_party/install
-- -- -DCMAKE_TOOLCHAIN_FILE:PATH=
-- -- -DCMAKE_POSITION_INDEPENDENT_CODE=ON
-- -- -DBUILD_SHARED_LIBS=OFF
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/Anthony/git/MAVSDK/build/default/third_party/json11
Microsoft (R) Build Engine version 16.1.76+g14b0a930a7 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

  Performing update step for 'json11'
  Performing patch step for 'json11'
CUSTOMBUILD : error : patch failed: CMakeLists.txt:23 [C:\Users\Anthony\git\MAVSDK\build\default\third_party\json11\json11.vcxproj]
CUSTOMBUILD : error : CMakeLists.txt: patch does not apply [C:\Users\Anthony\git\MAVSDK\build\default\third_party\json11\json11.vcxproj]
CMake Error at third_party/cmake/build_target.cmake:54 (message):
  C:/Users/Anthony/git/MAVSDK/build/default/third_party/json11 failed to
  build!
Call Stack (most recent call first):
  third_party/CMakeLists.txt:10 (build_target)

-- Configuring incomplete, errors occurred!
See also "C:/Users/Anthony/git/MAVSDK/build/default/CMakeFiles/CMakeOutput.log".
See also "C:/Users/Anthony/git/MAVSDK/build/default/CMakeFiles/CMakeError.log". 
```
I have attached the two CMake error logs.

[CMakeError.zip](https://github.com/mavlink/MAVSDK/files/3408976/CMakeError.zip)

JonasVautherin, 2019-07-19:
> error : CMakeLists.txt: patch does not apply

So what it does is:

* Download json11 from https://github.com/dropbox/json11, at tag `v1.0.0`.
* `git apply` our patch, which is [this file](https://github.com/mavlink/MAVSDK/blob/develop/third_party/json11/export_config.patch).

Apparently, it fails at this point. Can you try to understand why? I would look for:

* Is `git` installed in the first place? After all, it calls `git apply`
* Download [json11 `v1.0.0`](https://github.com/dropbox/json11/tree/v1.0.0) and [`export_config.patch`](https://raw.githubusercontent.com/mavlink/MAVSDK/develop/third_party/json11/export_config.patch) manually, and try to apply the patch with `git apply export_config.patch`.

imfatant, 2019-07-19:
Git is installed.

I shall try to do the manual git apply.

It is nothing to do with not being able to find pthread.h on a Windows system?

imfatant, 2019-07-19:
> Download json11 v1.0.0 and export_config.patch manually

I don't know how to do this.

SelimOzel, 2019-07-20:
Jonas, I'll only have the chance to try this out maybe mid-week next week? Just wanted to update you. And thanks for instructions.

JonasVautherin, 2019-07-20:
@imfatant: for json, you can go to the link I provided and download it through the "download" button of GitHub. For the patch, you can just open the link I provided and save that file.

Does that make sense?

imfatant, 2019-07-20:
@JonasVautherin: Yes, I've done that. Now, where do I place the json11-1.0.0 directory after I've applied the patch?

JonasVautherin, 2019-07-20:
> Now, where do I place the json11-1.0.0 directory after I've applied the patch?

Well, the question is actually about that: "does `git apply` work on your Windows, or not?

So did it work? You could patch json11 with `export_config.patch`? If yes, can you try a clean build again (by removing `build/default`, e.g. `rm -rf build/default`)?

imfatant, 2019-07-20:
```git apply``` did work. I'll try a clean build, but I think we're going to have a problem with zlib.

JonasVautherin, 2019-07-20:
> but I think we're going to have a problem with zlib.

Why is that?

imfatant, 2019-07-20:
Let's go back a bit. I can download json11-1.0.0 and patch it. But can you confirm where I'm going to put the patched directory back into the MAVSDK directory structure?

imfatant, 2019-07-20:
I presume one simply copies the manually patched json11-1.0.0 directory (having renamed it to json11) over MAVSDK\build\default\third_party\json11. It would tres helpful if you could confirm this.

I have tried this, and it seems to work, but then one runs into the same problem for zlib, and then, having patched that, the same problem for curl.. which appears not to have a patch. After which one gives up.

JonasVautherin, 2019-07-20:
No you won't do anything with that anymore. The message says "CMakeLists.txt: patch does not apply". I wanted to check if `git apply` was working, and it seems like `git apply` does work. So maybe it's because it is not a clean build.

> I have tried this

Did you try the clean build?

imfatant, 2019-07-22:
Yes, I did try a clean build, and no it doesn't work.

Now it's failing before I even get to trying the local install:

```
C:\Users\Anthony\git\MAVSDK>cmake -G "Visual Studio 16 2019" -A x64 -DBUILD_SHARED_LIBS=ON -Bbuild/default -H.
-- Selecting Windows SDK version 10.0.17763.0 to target Windows 10.0.17134.
-- The C compiler identification is MSVC 19.21.27702.2
-- The CXX compiler identification is MSVC 19.21.27702.2
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.21.27702/bin/Hostx64/x64/cl.exe
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.21.27702/bin/Hostx64/x64/cl.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.21.27702/bin/Hostx64/x64/cl.exe
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.21.27702/bin/Hostx64/x64/cl.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
...
   -- Installing: C:/Users/Anthony/git/MAVSDK/build/default/third_party/install/lib/cmake/CURL/CURLConfig.cmake
    -- Installing: C:/Users/Anthony/git/MAVSDK/build/default/third_party/install/lib/libcurl-d.lib
    -- Installing: C:/Users/Anthony/git/MAVSDK/build/default/third_party/install/bin/curl.exe
  Completed 'curl'
  Building Custom Rule C:/Users/Anthony/git/MAVSDK/third_party/curl/CMakeLists.txt
fatal: not a git repository (or any of the parent directories): .git
-- Version:
-- Looking for pthread.h
-- Looking for pthread.h - not found
-- Found Threads: TRUE
CMake Error at src/CMakeLists.txt:85 (add_subdirectory):
  The source directory

    C:/Users/Anthony/git/MAVSDK/src/third_party/gtest

  does not contain a CMakeLists.txt file.

-- BUILD_BACKEND not set: not building grpc backend
-- Configuring incomplete, errors occurred!
See also "C:/Users/Anthony/git/MAVSDK/build/default/CMakeFiles/CMakeOutput.log".
See also "C:/Users/Anthony/git/MAVSDK/build/default/CMakeFiles/CMakeError.log".
```

JonasVautherin, 2019-07-22:
> CMake Error at src/CMakeLists.txt:85 (add_subdirectory):

Can you clean build with `-DBUILD_TESTS=OFF`?

julianoes, 2019-07-23:
@SelimOzel in what command line thing are you running this on Windows? I tried in the MSVC console and could not reproduce it.

SelimOzel, 2019-07-23:
Hey! I tried it in default command line and not the console that pops-up from MSVC. Is that the preferred one?

julianoes, 2019-07-24:
So you tried in `cmd`?

imfatant, 2019-07-28:
I did everything in `cmd`. How does one get a console in MSVC? I thought it was for output only?

julianoes, 2019-07-29:
I have tested this in `cmd` and it works fine for me.

JonasVautherin, 2019-07-29:
Did you guys try a clean build with `-DBUILD_TESTS=OFF`?

imfatant, 2019-07-29:
OK, this works for a local install:
```
git clone https://github.com/mavlink/MAVSDK.git
cd MAVSDK
git checkout develop
git submodule update --init --recursive
cmake -G "Visual Studio 16 2019" -A x64 -DBUILD_SHARED_LIBS=ON -DCMAKE_INSTALL_PREFIX=install -Bbuild/default -H.)
cmake --build build/default --target install
```

JonasVautherin, 2019-07-29:
So what was the issue before? What did you change to get it to work?

imfatant, 2019-07-29:
I don't have the dll's though. Only the .lib static library files and the headers. Where are the dll's?

The way I was doing it before was:
```
cd MAVSDK
git checkout develop
git submodule update --init --recursive
cmake -G "Visual Studio 16 2019" -A x64 -DBUILD_SHARED_LIBS=ON -Bbuild/default -H.
cmake --build build/default --config Debug
cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_INSTALL_PREFIX=install -Bbuild/default -H.)
cmake --build build/default --target install
```
The last two commands were the problem.

JonasVautherin, 2019-07-29:
> Only the .lib static library files and the headers.

So you get `libmavsdk_telemetry.lib` and not `libmavsdk_telemetry.dll`, for instance? You set `-DBUILD_SHARED_LIBS=ON`, so you should get shared libraries... :thinking:

julianoes, 2019-08-20:
Has this been resolved @SelimOzel or @imfatant ?

imfatant, 2019-08-24:
Yes, I think so.

SelimOzel, 2019-08-24:
Haven't had the chance to take a look. I'll try to squeeze it in and test sometime next week.

bvandonselaar, 2019-12-04:
> > Only the .lib static library files and the headers.
> 
> So you get `libmavsdk_telemetry.lib` and not `libmavsdk_telemetry.dll`, for instance? You set `-DBUILD_SHARED_LIBS=ON`, so you should get shared libraries... 🤔

I have exactly this problem, in master branch and develop branch. On Windows 10 64-bit to be specific.

julianoes, 2019-12-04:
@bvandonselaar can you share the steps that you did and the error that you see? That way I can try to reproduce.

bvandonselaar, 2019-12-04:
I do not have any errors, but I only receive static libraries after building. When I link them in a project, I get the error that the .dll version is missing on runtime.
Here are the commands I used to build:

```
cmake -G "Visual Studio 16 2019" -DCMAKE_BUILD_TYPE=Debug -DBUILD_SHARED_LIBS=ON -DCMAKE_INSTALL_PREFIX=install -Bbuild/default -S.

cmake --build build/default --target install
```
Yes I have used the `-S.` flag instead of `-H.` because H wasn't working, I believe that the cmake version I use is too new for `-H.`?

JonasVautherin, 2019-12-04:
Can you try the following?

1. Check in `build/default/CMakeCache.txt` the value of `BUILD_SHARED_LIBS` and let us know what it is set to (if set at all).
2. Clean build: `rm -rf build/default && <your cmake commands above>`.

bvandonselaar, 2019-12-04:
CMakeCache.txt partly copied:
```
########################
# EXTERNAL cache entries
########################

//Build gRPC backend server
BUILD_BACKEND:BOOL=OFF

//Builds the googlemock subproject
BUILD_GMOCK:BOOL=ON

//Build core as shared libraries instead of static ones
BUILD_SHARED_LIBS:BOOL=ON

//Build tests
BUILD_TESTS:BOOL=ON

//Use ccache if available
CCACHE:BOOL=ON
```

I also get this warning when running my first cmake command:
```
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    BUILD_SHARED_LIBS
```

JonasVautherin, 2019-12-04:
> I also get this warning when running my first cmake command

Did you try the clean build, i.e. `rm -rf build/default` and re-run cmake? Does it still say that?

bvandonselaar, 2019-12-05:
Yes I did, but what I said about the "Manually-specified variables were not used by the project" is wrong. It said that for curl, zlib etc. I'm sorry.

But when building it just says:
```
Building Custom Rule C:/<my-path>/Downloads/MAVSDK/src/plugins/shell/CMakeLists.txt
  shell.cpp
  shell_impl.cpp
  Generating Code...
  Auto build dll exports
     Creating library C:/<my-path>/Downloads/MAVSDK/build/default/src/plugins/shell/Debug/mavsdk_shell.lib and object C:/<my-path>/Downloads/MAVSD
  K/build/default/src/plugins/shell/Debug/mavsdk_shell.exp
  mavsdk_shell.vcxproj -> C:\<my-path>\Downloads\MAVSDK\build\default\src\plugins\shell\Debug\mavsdk_shell.dll
```
In the directory of `C:\<my-path>\Downloads\MAVSDK\build\default\src\plugins\shell\Debug\` I can see both a .dll version and a .lib version of the library. So maybe the `install` phase is wrong?

JonasVautherin, 2019-12-05:
> In the directory of C:\<my-path>\Downloads\MAVSDK\build\default\src\plugins\shell\Debug\ I can see both a .dll version and a .lib version of the library.

Oh, that's interesting. I'm a bit confused, though, as I would not expect you to get both shared and static libs. Could you try to write a small toy project that just builds a simple library, and check if it behaves similarly there? I can't see what could cause that in the MAVSDK CMakeLists files :confused:.

reve1, 2020-02-29:
Successfully got install/lib and install/include

```
c:\Qt\Tools\CMake_64\bin\cmake.exe -G "Visual Studio 16 2019" -DCMAKE_BUILD_TYPE=Debug -DBUILD_SHARED_LIBS=OFF -DCMAKE_INSTALL_PREFIX=install -Bbuild/default -S.
c:\Qt\Tools\CMake_64\bin\cmake.exe --build build/default --target install
```
I'm trying to create a simple application in Qt Creator on Windows with MAVSDK installed locally 
```
#include <QCoreApplication>
#include <QDebug>
#include <mavsdk/mavsdk.h>
#include <mavsdk/plugins/action/action.h>
#include <mavsdk/plugins/telemetry/telemetry.h>

using namespace mavsdk;

int main(int argc, char *argv[])
{
    QCoreApplication a(argc, argv);

        Mavsdk dc;

    return a.exec();
}

```

I Get

> 
> 14:33:23: Running steps for project test_mavsdk...
> 14:33:23: Starting: "C:\Qt\Tools\CMake_64\bin\cmake.exe" --build . --target clean
> 14:33:23: The process "C:\Qt\Tools\CMake_64\bin\cmake.exe" exited normally.
> 14:33:23: Starting: "C:\Qt\Tools\CMake_64\bin\cmake.exe" --build . --target all
> [ 25%] Automatic MOC and UIC for target test_mavsdk
> [ 25%] Built target test_mavsdk_autogen
> Scanning dependencies of target test_mavsdk
> [ 50%] Building CXX object CMakeFiles/test_mavsdk.dir/test_mavsdk_autogen/mocs_compilation.cpp.obj
> [ 75%] Building CXX object CMakeFiles/test_mavsdk.dir/main.cpp.obj
> [100%] Linking CXX executable test_mavsdk.exe
> CMakeFiles\test_mavsdk.dir/objects.a(main.cpp.obj): In function `main':
> C:/Qt/proj/test_mavsdk/main.cpp:13: undefined reference to `mavsdk::Mavsdk::Mavsdk()'
> C:/Qt/proj/test_mavsdk/main.cpp:13: undefined reference to `mavsdk::Mavsdk::~Mavsdk()'
> C:/Qt/proj/test_mavsdk/main.cpp:13: undefined reference to `mavsdk::Mavsdk::~Mavsdk()'
> collect2.exe: error: ld returned 1 exit status
> mingw32-make.exe[2]: *** [CMakeFiles\test_mavsdk.dir\build.make:103: test_mavsdk.exe] Error 1
> mingw32-make.exe[1]: *** [CMakeFiles\Makefile2:76: CMakeFiles/test_mavsdk.dir/all] Error 2
> mingw32-make.exe: *** [Makefile:83: all] Error 2
> 14:33:26: The process "C:\Qt\Tools\CMake_64\bin\cmake.exe" exited with code 2.
> Error while building/deploying project test_mavsdk (kit: Desktop Qt 5.14.1 MinGW 64-bit)
> When executing step "CMake Build"
> 14:33:26: Elapsed time: 00:04.
> 

my CMakeLists.txt
```
cmake_minimum_required(VERSION 3.5)

project(test_mavsdk LANGUAGES CXX)

set(CMAKE_INCLUDE_CURRENT_DIR ON)

set(CMAKE_AUTOUIC ON)
set(CMAKE_AUTOMOC ON)
set(CMAKE_AUTORCC ON)

set(CMAKE_CXX_STANDARD 11)
set(CMAKE_CXX_STANDARD_REQUIRED ON)

find_package(Qt5Core)

include_directories(
    mavsdk/include
)

link_directories(
    mavsdk/lib
)

add_executable(test_mavsdk
    main.cpp
)
target_link_libraries(test_mavsdk Qt5::Core)

target_link_libraries(test_mavsdk
    mavsdk
    mavsdk_action
    mavsdk_telemetry
)
```
MAVSDK is located in the same directory as the project

julianoes, 2020-02-29:
I'm not sure if this matters but I would put all library dependencies in one:

```
target_link_libraries(test_mavsdk
    Qt5::Core
    mavsdk
    mavsdk_action
    mavsdk_telemetry
)
```

And you should also be able to do a `find_package(MAVSDK REQUIRED)` and then do:

```
target_link_libraries(test_mavsdk
    Qt5::Core
    MAVSDK::mavsdk
    MAVSDK::mavsdk_action
    MAVSDK::mavsdk_telemetry
)
```

svatambeti, 2020-04-27:
Have same issue while I am trying to build MAVSDK libs on Windows using CMake and MinGW:
Using MAVSDK master branch.

Command I used to start build:
cmake -G "MinGW Makefiles" -DBUILD_SHARED_LIBS=OFF -DBUILD_TESTS=OFF -Bbuild/default -H.

Error log below..

Scanning dependencies of target jsoncpp
[ 12%] Creating directories for 'jsoncpp'
[ 25%] Performing download step (git clone) for 'jsoncpp'
Cloning into 'jsoncpp'...
Note: checking out '1.8.4'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at ddabf50... 1.8.4; soversion=20
[ 37%] Performing patch step for 'jsoncpp'
[ 50%] Performing update step for 'jsoncpp'
F:/Projects/MAVSDK/third_party/jsoncpp/fixlibname.patch:9: trailing whitespace.
                NAMESPACE JsonCpp::
F:/Projects/MAVSDK/third_party/jsoncpp/fixlibname.patch:24: trailing whitespace.
    ADD_LIBRARY(jsoncpp SHARED ${PUBLIC_HEADERS} ${jsoncpp_sources})
F:/Projects/MAVSDK/third_party/jsoncpp/fixlibname.patch:25: trailing whitespace.
    SET_TARGET_PROPERTIES( jsoncpp PROPERTIES VERSION ${JSONCPP_VERSION} SOVERSION ${JSONCPP_SOVERSION})
F:/Projects/MAVSDK/third_party/jsoncpp/fixlibname.patch:26: trailing whitespace.
    SET_TARGET_PROPERTIES( jsoncpp PROPERTIES OUTPUT_NAME jsoncpp
F:/Projects/MAVSDK/third_party/jsoncpp/fixlibname.patch:32: trailing whitespace.
        SET_TARGET_PROPERTIES( jsoncpp PROPERTIES INSTALL_RPATH "@loader_path/." )
error: patch failed: CMakeLists.txt:148
error: CMakeLists.txt: patch does not apply
error: patch failed: src/lib_json/CMakeLists.txt:69
error: src/lib_json/CMakeLists.txt: patch does not apply
CMakeFiles\jsoncpp.dir\build.make:118: recipe for target 'jsoncpp/src/jsoncpp-stamp/jsoncpp-patch' failed
mingw32-make.exe[2]: *** [jsoncpp/src/jsoncpp-stamp/jsoncpp-patch] Error 1
mingw32-make.exe[2]: *** Waiting for unfinished jobs....
CMakeFiles\Makefile2:94: recipe for target 'CMakeFiles/jsoncpp.dir/all' failed
mingw32-make.exe[1]: *** [CMakeFiles/jsoncpp.dir/all] Error 2
Makefile:102: recipe for target 'all' failed
mingw32-make.exe: *** [all] Error 2
CMake Error at third_party/cmake/build_target.cmake:48 (message):
  F:/Projects/MAVSDK/build/default/third_party/jsoncpp failed to
  build!
Call Stack (most recent call first):
  third_party/CMakeLists.txt:10 (build_target)

-- Configuring incomplete, errors occurred!

Can someone help me through this?

julianoes, 2020-04-27:
@svatambeti ok, thanks for the info. Why are you using Mingw rather than MSVC?

JonasVautherin, 2020-04-27:
> error: patch failed: CMakeLists.txt:148
error: CMakeLists.txt: patch does not apply
error: patch failed: src/lib_json/CMakeLists.txt:69
error: src/lib_json/CMakeLists.txt: patch does not apply
CMakeFiles\jsoncpp.dir\build.make:118: recipe for target 'jsoncpp/src/jsoncpp-stamp/jsoncpp-patch' failed

I think that you need a clean build. When building the dependencies (in this case jsoncpp), we apply a patch. It turns out that it's not so easy to do in Windows, because there doesn't seem to be a default system `patch` executable. Hence I believe that it sometimes ends up in situations where the patch was already applied, and it fails to apply it again.

Can you try a clean build and let us know if that works?

svatambeti, 2020-04-27:
@JonasVautherin  I have already tried all of the above suggestions that were suggested in this post, did not help. I am stuck.
If this does not resolve, then I may have to consider cross-compiling option

svatambeti, 2020-04-27:
> 
> 
> @svatambeti ok, thanks for the info. Why are you using Mingw rather than MSVC?

My end application I am working is using MingGW toolchain to build. MAVSDK will be one of the libs this end application uses. Hence I need to build MAVSDK libs using Mingw so that all dependent libs are built using same toolchain.

JonasVautherin, 2020-04-27:
Are you sure it is a clean build? Could you try for instance to make a clean checkout somewhere else and run the build? I mean, it really says that the patch does not apply...

svatambeti, 2020-04-30:
@JonasVautherin after a few failed attempts of deleting build/default folder and then rebuild, I tried it on a completely fresh setup on a different computer. The only difference between these two machines is first one uses MinGW64, later used Ming32.
On second machine, jsoncpp did not complain, went well but then got stuck with same patch error on zlib.
So I removed zlib from CMakeLists since MinGW will have a zlib already installed. So with this change, I was able to successfully build static libs.

Though my work around is not a solution, thought I will share what worked for me. It would be nice to have a clean working build setup for Windows as well.

JonasVautherin, 2020-04-30:
@svatambeti: got it, thanks. This is how we patch the two dependencies: 
* jsoncpp: https://github.com/mavlink/MAVSDK/blob/develop/third_party/jsoncpp/CMakeLists.txt#L25
* zlib: https://github.com/mavlink/MAVSDK/blob/develop/third_party/zlib/CMakeLists.txt

Essentially by running `PATCH_COMMAND git apply ${PROJECT_SOURCE_DIR}/<patch>`.

Now I'm not a Windows user, and I am not sure what the best way is. Is there a generic way for Windows (on Linux, you can assume that the `patch` function exists, and here we assume that `git` exists because `patch` definitely doesn't exist on Windows :sweat_smile:).

What's your take on that? How would you do it?

clementgoiz, 2020-06-29:
Hi, I am on Windows too and have the same problem :
error: patch failed: CMakeLists.txt:148
error: CMakeLists.txt: patch does not apply
error: patch failed: src/lib_json/CMakeLists.txt:69
error: src/lib_json/CMakeLists.txt: patch does not apply
CMakeFiles\jsoncpp.dir\build.make:118: recipe for target 'jsoncpp/src/jsoncpp-stamp/jsoncpp-patch' failed

JonasVautherin, 2020-06-29:
"patch does not apply" suggests that the command works (`git apply ...`) but that the patch itself doesn't. E.g. because it was already patched before.

@clementgoiz: Is this a clean build?

julianoes, 2020-11-23:
I'm closing this because there was no response. If someone runs into this again and can provide more data, please re-open and comment below.

closed 2020-11-23 by julianoes

sub131, 2022-12-08:
In some configurations, windows/unix line feeds can cause a patch failed symptom.
I saw it when building in Git-Bash on a Windows machine. Using `git apply patch -v path/to/patch/file` helped see why the patch failed.
I converted the patch files to unix LF and the patch worked.

julianoes, 2022-12-13:
That's odd. So you're saying the patches are in Windows format? I would have thought they come with Unix line endings.

sub131, 2022-12-15:
> That's odd. So you're saying the patches are in Windows format? I would have thought they come with Unix line endings.

Yes in my case, but it depends on how everything's setup. I was using Windows for Git, which converts the files to Windows format when it downloads. I was using a Docker image to cross-compile for arm64, so it needed Linux. I should have made sure to configure GIT to download with Unix line endings.