mavlink/MAVSDK #802: Has the directory structure changed?

imfatant, 2019-07-17:
I've built and installed the MAVLINK library as described in the documentation (Linux).

In takeoff_and_land.cpp, we have:

#include <mavsdk/mavsdk.h>
#include <mavsdk/plugins/action/action.h>
#include <mavsdk/plugins/telemetry/telemetry.h>

This directory structure seems completely differenty to the one extant.

JonasVautherin, 2019-07-17:
I'm not sure I understand the question. `takeoff_and_land.cpp` seems fine to me. Did you install using a package (e.g. `*.deb`) from the [releases](https://github.com/mavlink/MAVSDK/releases)?

imfatant, 2019-07-18:
I'm using vscode and would rather compile without using cmake. Now, I've got it working with cmake by setting up two build tasks in vscode, which means that the install went fine, but when using g++ directly, I get:
```
> Executing task: g++ -Wno-unused-parameter -Wno-unused-variable -Wall -Weffc++ -Wextra -Wsign-conversion -g takeoff_and_land.cpp -o takeoff_and_land -I/usr/local/include -L/usr/local/lib -lmavsdk -lmavsdk_action -lmavsdk_telemetry <

In file included from takeoff_and_land.cpp:8:0:
/usr/local/include/mavsdk/plugins/action/action.h:6:10: fatal error: plugin_base.h: No such file or directory
 #include "plugin_base.h"
          ^~~~~~~~~~~~~~~
compilation terminated.
The terminal process terminated with exit code: 1
```
Would be most grateful for any ideas.

JonasVautherin, 2019-07-18:
What about adding something like `-I/usr/local/include/mavsdk`? Seems to me like `MAVSDKTargets.cmake` does something like that.

imfatant, 2019-07-18:
That works!

Thank you:)

closed 2019-07-18 by imfatant