mavlink/MAVSDK #804: don't support VTOL TAKEOFF and LAND in fly_qgc_mission

dangviethieu, 2019-07-19:
Hello guys
I get problems while test fly_qgc_mission function with vtol-simulator(https://docs.deltaquad.com/simulator.html)
[image]
Anyone can help me with it? I check in mission_impl.cpp and it seems no support for MAV_CMD_NAV_VTOL_TAKEOFF and MAV_CMD_NAV_VTOL_LAND command.
I want know why? please help me? Thanks 
[image]

julianoes, 2019-07-19:
Correct, the `mission` plugin does not support VTOL items yet.

If you need support for that you would have to use the `mission_raw` plugin for now.

dangviethieu, 2019-07-19:
@julianoes thank you for reply.
We does haven't example for mission_raw yet?
In mission plugin, we have MissionItem class for setup waypoint, but mission_raw is don't have.
So how to we can create mission use mission_raw plugin?

julianoes, 2019-07-19:
Yes, with mission_raw you basically get a default MAVLink style mission and there is no easy types or examples to use.

julianoes, 2019-07-19:
Alternatively, you could add support for VTOL actions to the `mission` plugin and that way enable it.

closed 2019-11-15 by dangviethieu

itrigitech, 2019-12-22:
Any progress on that? I also need to add support for VTOL missions.

Checking the code i see how the mission plugin is implemented and for example it checks if the first mission item is a waypoint. I thought that this item could be a takoff. In fact it stops downloading the mission if is not a waypoint. 

Do you know why is that? I don't understand why this check exists.

hamishwillee, 2020-01-06:
@itrigitech Adding to old items is frowned upon. The right way to handle this would be to raise in discussion board or slack and link here.

That said, I have responded to your query in https://github.com/mavlink/MAVSDK/issues/938

The current check is waypoint because the plugin doesn't support takeoff items. It should IMO.

JonasVautherin, 2020-01-06:
> The current check is waypoint because the plugin doesn't support takeoff items. It should IMO.

Agreed. And @julianoes does, as well:

> Alternatively, you could add support for VTOL actions to the mission plugin and that way enable it.

We are open for contributions :blush: