mavlink/MAVSDK #806: Vision Support?

Doodle1106, 2019-07-21:
Hi guys, 

    Is there any support for vision related messages such as VISION_POSITION_ESTIMATE or will there be in the near future?
    
    Thanks for the impressive work!

Joey

julianoes, 2019-07-22:
Hi Joey,
Thanks :).

Yes it would indeed be nice to send vision position estimates to a vehicle using MAVSDK. I have never actually used that feature with PX4, so I'd have to play with it myself first.

If you need it, I'd suggest to write a new plugin for it and make a pull request!

julianoes, 2021-10-20:
I would have to investigate that deeper on the PX4 and EKF2 side. Have you tried this in SITL where you can add printfs to see what is going on?

ArtBor86, 2021-10-21:
Oh, it's my bad, i'm an idiot. I fogot about the VPE.pose_covariance.covariance_matrix. 

Big thanks for your answer!