mavlink/MAVSDK #811: How to connect Px4 on SITL

friendbaby, 2019-07-23:
The Firmware version is 1.8.0dev.
PX4 SITL listen  on UDP port 14560, MAVSDK in UDP Mode  listen one port too.
So, How to connect Px4 on SITL ?

julianoes, 2019-07-23:
Does this diagram help?
https://dev.px4.io/v1.8.2/en/simulation/#sitl-simulation-environment

QGC usually listens on UDP port 14550, MAVSDK or MAVROS listen on UDP port 14540.

friendbaby, 2019-07-23:
> Does this diagram help?
> https://dev.px4.io/v1.8.2/en/simulation/#sitl-simulation-environment
> QGC usually listens on UDP port 14550, MAVSDK or MAVROS listen on UDP port 14540.

I have seen this diagram, and I test result is different it.

the pix4  is not connect to QGC(udp port 14550), it is :

pix4(udp 14560) <---->Simulator<---->QGC(udp 14550).

So if remove Simulator , the pix4 can not connect to QGC.

In actually, my target is write a new Simulator with MAVSDK.  I need  connect pix4 udp active,just like jMAVSim.

julianoes, 2019-07-23:
Ah that's an idea. You can do that. I think you just need to connect to UDP port 14560 instead of 14550.

DiegoGLagash, 2019-10-01:
Is there any support for v1.9.2 and setting a TCP port?

julianoes, 2019-10-02:
Please re-open if this is not working as intended.

closed 2019-10-02 by julianoes