mavlink/MAVSDK #814: Error building mavsdk_server (LNK1149)

uccsoft, 2019-07-24:
Got problem trying to compile mavsdk with BUILD_BACKEND = ON option. I am using VS 2017 Build Tools on Windows 10 Enterprise. I am getting the error below. Any help would be appriciated.

[build]        "D:\Work\MAVSDK\build\ALL_BUILD.vcxproj" (default target) (1) ->
[build]        "D:\Work\MAVSDK\build\src\backend\src\mavsdk_server_bin.vcxproj" (default target) (32) ->
[build]        (Link target) -> 
[build]          LINK : fatal error LNK1149: output filename matches input filename 'D:\Work\MAVSDK\build\src\backend\src\Debug\mavsdk_server.lib' [D:\Work\MAVSDK\build\src\backend\src\mavsdk_server_bin.vcxproj]
[build] 
[build]     0 Warning(s)
[build]     1 Error(s)

JonasVautherin, 2019-07-24:
I'm completely speculating, but seeing `mavsdk_server.lib` and `mavsdk_server_bin` in the error output make me think it could be related to the way we name our binaries.

Indeed, we have two targets ending up being called `mavsdk_server`:

* The target `mavsdk_server` becomes a shared library (`mavsdk_server.so` on Linux, I presume `mavsdk_server.lib` on Windows)
* The target `mavsdk_server_bin` becomes an executable: `mavsdk_server`. By default it should be `mavsdk_server_bin`, but we rename it with [this property](https://github.com/mavlink/MAVSDK/blob/develop/src/backend/src/CMakeLists.txt#L87).

That's obviously working in our CI, but for Windows we build with MinGW and not with Visual Studio, so maybe that's failing there.

Could you try to comment out [these lines](https://github.com/mavlink/MAVSDK/blob/develop/src/backend/src/CMakeLists.txt#L86-L88) and build again?

julianoes, 2019-07-24:
> The target mavsdk_server_bin becomes an executable: mavsdk_server. By default it should be mavsdk_server_bin, but we rename it with this property.

Oh I missed this! What should we rename? :smile:

uccsoft, 2019-07-25:
> I'm completely speculating, but seeing `mavsdk_server.lib` and `mavsdk_server_bin` in the error output make me think it could be related to the way we name our binaries.
>
> Indeed, we have two targets [...]

Thanks a lot. It has worked. mavsdk_server_bin.exe is generated under debug folder.

julianoes, 2019-07-25:
@uccsoft thanks for trying that out.

@JonasVautherin I vote we rename the library to `mavsdk_server_lib`. I know that will create the ugly file `libmavsdk_server_lib.lib/so` but that's better than renaming the bin, right?

JonasVautherin, 2019-07-25:
> @JonasVautherin I vote we rename the library to mavsdk_server_lib. I know that will create the ugly file libmavsdk_server_lib.lib/so but that's better than renaming the bin, right?

I would first try to see if there is a way to make that work for Visual Studio (it works everywhere else). If there isn't, then I think I would rather rename the executable only for Visual Studio. So that everybody has `mavsdk_server`, and people building with Visual Studio have `mavdsk_server_bin.exe`.

If you rename the library, it has consequences on the CMakeLists linking it. Whereas the executable is an "end product" :sweat_smile:.

julianoes, 2019-07-25:
@JonasVautherin alright, convinced. I vote we change it unless someone knows how to avoid the problem in VS Studio but I doubt it.

JonasVautherin, 2019-07-25:
Fair enough.

PR #834 (2019-08-12): generate mavsdk_server_bin.exe instead of mavsdk_server.exe with MSVC
For some reason, MSVC is the only one not allowing us to generate `mavsdk_server` as both an executable and a library.

Resolves #814.
closed 2019-08-12 by julianoes