mavlink/MAVSDK #823: Missing plugins when Ubuntu install the SDK from *.deb

bozkurthan, 2019-07-30:
Hello,
I was trying to build all examples from MAVSDK on Ubuntu 18.04. I installed MAVSDK from releases page https://github.com/mavlink/MAVSDK/releases

When I use `cmake ..` to build battery example, I'm giving following error :
```

-- Configuring done
CMake Warning (dev) at CMakeLists.txt:15 (add_executable):
  Policy CMP0028 is not set: Double colon in target name means ALIAS or
  IMPORTED target.  Run "cmake --help-policy CMP0028" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  Target "battery" links to target "MAVSDK::mavsdk_mavlink_passthrough" but
  the target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done
-- Build files have been written to: /home/hanco/MAVSDK-develop/example/battery/build
```

After that I navigated to installation folder /usr/include/mavsdk and compared cloned github folder. I found that there is a some missing plugins on installation from release page. 

[image]

Am I missing some steps?

Thanks.

julianoes, 2019-07-30:
Ok, the MAVLinkPassthrough plugin is not included yet. The reason I did not add it was because it essentially brings all the generated mavlink c headers with it as well, and because in general usage of MAVLink passthrough is discouraged unless really needed.

bozkurthan, 2019-07-30:
I see it's normal to encourage people to contribute while SDK is still under development. For me, I  just want to combine some examples like https://github.com/bozkurthan/PX4-MAVSDK-OFFBOARD-EXAMPLES and share these examples with community. Is there any option to workaround this issue?

julianoes, 2019-07-30:
The offboard examples should work fine. I think it's only the battery one which does not.

JonasVautherin, 2019-07-30:
> The offboard examples should work fine. I think it's only the battery one which does not.

Yes! @bozkurthan just remove [this line](https://github.com/mavlink/MAVSDK/blob/develop/example/CMakeLists.txt#L5) in the examples.

> Ok, the MAVLinkPassthrough plugin is not included yet.

And I'm still not sure if we should include it in the packages we distribute. Or maybe we should add it, but add a big `WARNING` where people instantiate it, saying that it is not supported. My fear is that the C++ people use that because it seems faster than properly adding features, and then the only ones who will actually need to extend the C++ implementation will be those who use language wrappers :thinking:.

hamishwillee, 2019-07-30:
> And I'm still not sure if we should include it in the packages we distribute. Or maybe we should add it, but add a big WARNING where people instantiate it, saying that it is not supported. 

IMO end users who are inclined to choose MAVLinkPassThrough are probably never going to write plugins - their choice is really use it or go find another library. I think we should support it, and properly.

A warning is a good idea though to encourage users who just don't understand that this is not the best way to do things.

bozkurthan, 2019-07-31:
> The offboard examples should work fine. I think it's only the battery one which does not.

Thanks. It's enough for me :) @julianoes 

> Or maybe we should add it, but add a big WARNING where people instantiate it, saying that it is not supported. 

@JonasVautherin , @hamishwillee  IMO you shouldn't. I think everyone wants to choose easy way to implement their application faster. MAVSDK is still growing. So, if you add this feature, MAVSDK stands on this library even you put warning message. It can be good for end-users but we should wait for bigger community supports like other SDKs. 

Thank you all :)

JonasVautherin, 2019-07-31:
> Thanks. It's enough for me :)

Good! I'll close the issue then. Feel free to re-open if needed.

Regarding MAVLinkPassthrough, I guess the discussion will come back anyway, and this issue is maybe not the right place to discuss it. Thanks for the inputs!

closed 2019-07-31 by JonasVautherin