mavlink/MAVSDK #827: Missing dll on windows

Benjdu76, 2019-07-31:
When I run an example, I get errors which said that :
- libgcc_s_seh-1.dll is missing
- libwinpthread-1.dll is missing
- libstdc++-6.dll is missing

I also have the following error in the python shell :
`grpc._channel._Rendezvous: <_Rendezvous of RPC that terminated with:
        status = StatusCode.UNAVAILABLE
        details = "failed to connect to all addresses"
        debug_error_string = "{"created":"@1564499126.880000000","description":"Failed to pick subchannel","file":"src/core/ext/filters/client_channel/client_channel.cc","file_line":3528,"referenced_errors":[{"created":"@1564499126.879000000","description":"failed to connect to all addresses","file":"src/core/ext/filters/client_channel/lb_policy/pick_first/pick_first.cc","file_line":399,"grpc_status":14}]}"`

I run the script on python 3.7 with mavsdk 0.1.0 installed with pip3 install.
I tried on windows 10 and windows 7 with the same issue.

JonasVautherin, 2019-07-31:
@Benjdu76: would you mind checking if you can build MAVSDK from sources on your system?

More generally, we need to fix our build system such that those libraries are either not linked (@julianoes was mentioning that we should probably not need `libgcc_s_seh-1.dll`) or linked statically. @Benjdu76 can you help with that?

raul7, 2019-08-09:
@JonasVautherin I was having the same isssue as @Benjdu76, so I built MAVSDK from sources on Windows 10 with Visual Studio 16 2019 tools (using the comands given  in "appveyor.yml" for Visual Studio), copied "mavsdk_server_bin.exe" to my python mavsdk installation, renamed to "mavsdk_server" and it looks like now mavsdl-server launches OK. I also installed mavsdk python using pip (but I'm using anaconda)

JonasVautherin, 2019-08-10:
Thanks @raul7 for the report. So this means that it works when you build from sources, but the binary we distribute over pip is missing DLLs. We should probably make it such that `mavsdk_server` links those libraries statically...

julianoes, 2020-11-23:
I believe we no longer link to these dlls because we switched back from Cygwin to MSVC build.

However, as I can see, we are linking to the VCRUNTIME140.DLL instead. So far there have not been any complaints regarding this, so I'm assuming it's fine and closing this issue.

closed 2020-11-23 by julianoes