mavlink/MAVSDK #830: Add option to build without curl

julianoes, 2019-08-07:
For slow compilation (e.g. on embedded boards) it would be nice to be able to build without curl.

JonasVautherin, 2019-08-07:
Isn't it good enough to cross-compile for those boards? If we make it optional in the code, we will also need to make it optional in the superbuild (because I guess we don't expect people to figure out that by themselves), and that adds complexity and corner cases everywhere...

julianoes, 2019-08-08:
Sometimes you're in a debugging situation where you just need to rebuild on the device.

JonasVautherin, 2019-08-09:
Right. But I don't think that adding options in CMakeLists and ifdefs in the code is the right solution for debugging on the device.

If you are debugging curl on the device, then you need to compile it. If you are debugging MAVSDK, then you don't have to recompile curl. Yes, that implies that you may need to not use our superbuild and to learn how to do that (e.g. cross-compile the dependencies separately, copy that on your device and build mavsdk on the device without SUPERBUILD=ON). But that's "basic" dependency management in cmake. I'm happy to write a superbuild to make that easier for most use-cases, but for specific use-cases, I believe that developers should learn how to deal with that.

To me it would be the same as saying "we should make it such that cmake checks if the submodules are correctly synced, because we don't want our developers to have to know about git submodules", and then "cmake should show a warning when people push to their fork without having an 'upstream' branch, because not everyone knows how to work with a fork".

We can do all that, but it results in more code that we need to maintain, and it forces devs to learn about our custom wrappers instead of learning to use the actual tools.

Am I too much of a purist? :laugh:

julianoes, 2019-08-12:
I agree with your examples but this is more a request that the MAVSDK should be able to build and run with minimal dependencies if required.

closed 2019-09-30 by JonasVautherin

julianoes, 2019-10-01:
For the record: it still requires curl but you can now also use libcurl from the system instead of getting it via git and building it. The way to do so is to use the configure option `cmake -DSUPERBUILD=OFF` for the configure step. Note this is supported as of Ubuntu 18.04 but not 16.04.

PR #865 (2019-09-30): Move to jsoncpp and make non-superbuild work with jsoncpp and tinyxml2
This enables building without superbuild (SUPERBUILD=OFF).

Currently contains #864.

Fixes  #830.

FYI @caleridas
PR #1955 (2023-12-17): Add option to build without cURL
This adds the cmake configure option BUILD_WITHOUT_CURL which allows building MAVSDK without the cURL dependency.

This has been requested numerous times, so it's probably worthwhile adding as an option.

For now it means that camera definition files can't be downloaded via http.

In the future we will also add the option to download the definition files via MAVLink FTP.

Closes #830.