mavlink/MAVSDK #835: Deprecate embedded camera definition files

JonasVautherin, 2019-08-11:
I am wondering if we should consider deprecating the embedded camera definition files, because that creates a dependency from MAVDSK on those cameras (as in: updating the camera firmware will break apps that don't have the latest version of MAVSDK).

Looking at the [embedded camera definition files](https://github.com/mavlink/MAVSDK/tree/develop/src/plugins/camera/camera_definition_files), it seems like they all correspond to Yuneec cameras. If I remember correctly, this was implemented as a temporary workaround more than a year ago, but the original intent was always to download this file from the camera.

@julianoes: what's your opinion on that? Or should that wait until we have MAVLink FTP, so that the camera plugin can benefit from that?

hamishwillee, 2019-08-11:
> I remember correctly, this was implemented as a temporary workaround more than a year ago, but the original intent was always to download this file from the camera.

To be very pedantic, the intent was to download _and cache_ the file from the camera using the revision number.

There is a bit of a discussion around the caching [here](https://github.com/mavlink/mavlink/issues/1198), which I haven't read/followed up yet.

Anyway, personally I wouldn't remove those files, but I would start checking the revision number and work out what you're going to do if there is a mismatch.

JonasVautherin, 2019-08-12:
The thing is that right now we don't cache those files, we hardcode them. A cache should be transparent, meaning that disabling the cache should not break the functionality.

We should first have the implementation without cache, and then we could add the cache to improve performance. 

This means I agree with you, @hamishwillee. But what we have now is not a cache, it is prone to errors and it makes our build system more complex (by adding `cpp_rsc`), and that makes cross-compiling harder. So I don't think we want to build on top of that...

hamishwillee, 2019-08-13:
@JonasVautherin I'm just being pedantic on where we need to go ... eventually. Up to you what we do now.

FWIW though, it is hard to say whether removing the files completely is OK without knowing what you will do if the files cannot be downloaded or take some time to download.

closed 2022-05-24 by JonasVautherin