mavlink/MAVSDK #836: Build errors on gcc8

fnoop, 2019-08-12:
Various places in the code fail with references to __atomic_xxx calls.  

```
[ 98%] Built target libprotoc
Scanning dependencies of target protoc
[ 99%] Building CXX object CMakeFiles/protoc.dir/srv/maverick/var/build/dronecore/build/default/third_party/protobuf/protobuf/src/protobuf/src/google/protobuf/compiler/main.cc.o
[100%] Linking CXX executable protoc
/usr/bin/ld: libprotobuf.a(arena.cc.o): in function `google::protobuf::internal::ArenaImpl::Init()':
arena.cc:(.text+0x44): undefined reference to `__atomic_fetch_add_8'
```

This can be worked around by adding the following to CMakeLists.txt:
`set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -latomic")`

```
gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/8/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../src/configure -v --with-pkgversion='Raspbian 8.3.0-6+rpi1' --with-bugurl=file:///usr/share/doc/gcc-8/README.Bugs --enable-languages=c,ada,c++,go,d,fortran,objc,obj-c++ --prefix=/usr --with-gcc-major-version-only --program-suffix=-8 --program-prefix=arm-linux-gnueabihf- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-libitm --disable-libquadmath --disable-libquadmath-support --enable-plugin --with-system-zlib --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-sjlj-exceptions --with-arch=armv6 --with-fpu=vfp --with-float=hard --disable-werror --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf
Thread model: posix
gcc version 8.3.0 (Raspbian 8.3.0-6+rpi1)
```

julianoes, 2019-08-12:
Thanks for the issue. I wouldn't say this is a GCC 8 issue but rather an issue of that toolchain on arm.

I don't think that the linker flag should be added to the `CXX` build flags though but rather we need to link to `atomic` as well.

julianoes, 2019-08-12:
@fnoop could you test if this works?
https://github.com/mavlink/MAVSDK/pull/837

fnoop, 2019-08-12:
@julianoes No, still the same problem with that PR.  Note the problem above is in third_party/protobuf, even though I have libprotobuf-dev installed - does it not use system library if already present?

julianoes, 2019-08-12:
Oh of course, I put it in the wrong place.

> does it not use system library if already present?

Not if you build with the super build.

fnoop, 2019-08-16:
So two options are:
 - Add `-DCMAKE_CXX_FLAGS=-latomic` to cmake, eg:
`/usr/bin/cmake ${atomic_flag} -DCMAKE_BUILD_TYPE=Release -DBUILD_SHARED_LIBS=ON -DBUILD_BACKEND=ON -DCMAKE_INSTALL_PREFIX=/srv/maverick/software/mavsdk`
It doesn't look like this gets passed through to the thirdparty builds however, although the thirdparty CMakeLists.txt could be altered to pull it in, possibly.

Or, set the environment variable `CXXFLAGS=-latomic` - this is a bit more 'sticky' as it gets pulled in by cmake to set CMAKE_CXX_FLAGS if found, and this does fix the thirdparty builds.

So it could be added as a build note/step for Raspberry (or Debian Buster in general, if it affects other architectures), but ideally it would be auto detected and set within mavsdk.

Ideally it would use system provided protobuf if available, unless there's a particular reason for using a separate compile, to save compile time and other problems like this.

JonasVautherin, 2019-08-16:
First, I would like to make sure I understand what problem you are trying to solve :blush:.

1. What is wrong with cross-compiling with dockcross?
2. Why do you want to build on the RPi instead of cross-compiling for it?

Assuming that there really is a need for compiling `mavsdk_server` on the RPi, I believe that the solution is to fix the protobuf compilation for this specific platform (i.e. in `third_party/protobuf/CMakeLists.txt`) and not to propagate `CXX_FLAGS` everywhere :sweat_smile:. In other words, go into `third_party/protobuf` and try to compile it from there.

> Ideally it would use system provided protobuf if available, unless there's a particular reason for using a separate compile, to save compile time and other problems like this.

The way it works is that the `MAVSDK` project finds its dependencies using [`find_package`](https://cmake.org/cmake/help/latest/command/find_package.html). So if you want to deal with the dependencies manually (which is perfectly fine), you should run with [`-DSUPERBUILD=OFF`](https://github.com/mavlink/MAVSDK/blob/develop/CMakeLists.txt#L5). The reason why we default to `-DSUPERBUILD=ON` is that most of our users don't know or don't want to deal with the dependencies themselves. Other reasons for having the superbuild at all is that we sometimes need to build our dependencies for some platforms (e.g. manylinux), and we sometimes need to patch them for other platforms.

What the superbuild does is conceptually quite easy:

a. It builds the dependencies separately and installs them ([by default](https://github.com/mavlink/MAVSDK/blob/develop/CMakeLists.txt#L11)) in `${PROJECT_BINARY_DIR}/third_party/install`
b. It sets `CMAKE_PREFIX_PATH` to that install directory for mavsdk
c. It builds MAVSDK

Which means that if MAVSDK doesn't find the dependency in `${PROJECT_BINARY_DIR}/third_party/install`, it will look for it on your system. But we don't have options to enable/disable each dependency separately: either you enable the superbuild, or you don't. If you want to do something in-between (i.e. have some dependencies on your system, and some built by our scripts), please feel free to do that! It is not super complicated, but you have to do it manually.

fnoop, 2019-08-16:
> First, I would like to make sure I understand what problem you are trying to solve 😊.
mavsdk failing to build as per instructions on latest Raspbian :)

The build is broken on Raspbian, which probably means it's broken on other non-x86 Debian buster builds, and possibly recent/future Ubuntu.  A workaround by cross compiling is not a solution, it's a workaround.  There's still a problem and it should be fixed.

My particular use-case is a uav companion computer project which implements a multi-platform automated build process.  This is my mavsdk build manifest, with the current CXXFLAGS workaround:
https://github.com/goodrobots/maverick/blob/master/manifests/maverick-modules/maverick_dev/manifests/mavsdk.pp#L9

See here for similar protobuf issue:
https://github.com/protocolbuffers/protobuf/issues/5219
In that issue is a PR from llvm with a bit more background:
https://github.com/protocolbuffers/protobuf/pull/5227

Thanks for background on the superbuild, would be very useful to document that in the build docs.

JonasVautherin, 2019-08-16:
> A workaround by cross compiling is not a solution, it's a workaround. There's still a problem and it should be fixed.

I don't deny the fact that building on the RPi fails, I was just wondering if that was really what you wanted. For instance, building on a RPi zero is probably super slow, which is why we worked hard on getting the cross-compilation to work.

But I get it, you want to build from the RPi.

> See here for similar protobuf issue

So we agree that this is a protobuf compilation issue, which means that there is no need to change the CXX_FLAGS of MAVSDK.

In my opinion, there are two ways to solve that. The "workaround", and the "solution", as per your definition above:

* Workaround: Go in MAVSDK's `third_party/protobuf/CMakeLists.txt`, and change the CXX_FLAGS there for armv7 (or something along those lines).
* Solution: Fix the protobuf build for armv7, getting inspiration from the PR you linked above that fixed armv6. NOTE: Maybe the fix for armv6 is even valid for armv7? Did you check that?

I totally agree with you that the "solution" would be much nicer, and I'm hoping you'll have a look at that :+1:! Otherwise, I'm obviously open to an armv7-specific workaround in our protobuf CMakeLists.

fnoop, 2019-08-16:
> I totally agree with you that the "solution" would be much nicer, and I'm hoping you'll have a look at that 👍

Honestly if I had the confidence to produce a proper solution I would have tried and submitted a PR :)

It looks like this problem will occur on platforms where gcc doesn't provide atomic operations but instead relies on libatomic.  I don't think it matters whether it's armv6 or armv7.  A quick grep through mavsdk for std::atomic indicates this will affect:
 - third_party/protobuf
 - third_party/grpc
 - mavsdk itself

JonasVautherin, 2019-08-16:
Thanks for the clarifications!

One thing I don't get, though, is that the build works for armv7 with dockcross, i.e. when not compiling `protoc`. So I don't think we need to link `libatomic` in mavsdk itself.

The second thing is that this was supposedly fixed in October 2018 in [protobuf](https://github.com/protocolbuffers/protobuf/pull/5227), and we use version 3.7.1 which is from March 2019. So I'm not sure why this happens :thinking:.

Did you try adding your CXX_FLAGS in `third_party/protobuf/CMakeLists.txt` to see where that goes?

fnoop, 2019-08-17:
> One thing I don't get, though, is that the build works for armv7 with dockcross, i.e. when not compiling protoc. So I don't think we need to link libatomic in mavsdk itself.

dockcross is useful to build generic binaries for a different architecture, but it's not a replacement for an optimised or current complete environment for a specific platform.  

I've now manually added in `-DCMAKE_CXX_FLAGS=-latomic` to protobuf, got past that successfully and then hit the same problem in grpc.   So I've made the same fix to grpc and got past that, and now hitting the same problem with mavsdk itself:
```
[ 86%] Linking CXX executable mavsdk_server
/usr/bin/ld: libmavsdk_server.so: undefined reference to `__atomic_fetch_add_8'
collect2: error: ld returned 1 exit status
make[2]: *** [src/backend/src/CMakeFiles/mavsdk_server_bin.dir/build.make:86: src/backend/src/mavsdk_server] Error 1
make[1]: *** [CMakeFiles/Makefile2:1580: src/backend/src/CMakeFiles/mavsdk_server_bin.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

The fix in protobuf was for autoconf, not cmake.

JonasVautherin, 2019-08-17:
Hmm. Of course protobuf will not export libatomic in its cmake targets when we add it through the CXX_FLAGS :thinking:. And therefore the CXX_FLAGS workaround has to be done at the mavsdk level.

If that's fine, give me a few days to check if there is an easy fix at the protobuf level, which would be much more elegant. If I don't manage to do that, I'll accept a PR fixing it at the mavsdk level.

But then, instead of CXX_FLAGS, wouldn't it make sense to try @julianoes approach of correctly linking it through target_link_libraries? Because setting CXX_FLAGS will propagate that issue to whoever uses mavsdk, and force them to link libatomic themselves, right?

JonasVautherin, 2019-08-18:
I managed to get an RPi to try to reproduce it myself and confirm that this is a protobuf issue. Turns out it is, and it [was fixed last month](https://github.com/protocolbuffers/protobuf/commit/55ed1d427ccc0d200927746329ac9b811dee77b9). So using latest `master` seems to just work.

@fnoop: Please review the PR

fnoop, 2019-08-18:
Works, thanks :)

PR #844 (2019-08-19): Fail to build protobuf on RPi
Protobuf fails to build on armv6/armv7 because of some `libatomic` linking. This got fixed recently, so let's update protobuf to latest upstream until their next release.

Resolves #836.
closed 2019-08-19 by julianoes