mavlink/MAVSDK #845: Quadcopter does not apply set_takeoff_altitude

shrit, 2019-08-19:
Hello, 

I am asking several quadcopters to takeoff to 10 meters, (I have already set takeoff altitude to 10 meters). In the following simulations, the quadcopters are situated on a ground that is 5 meters up sea levels. 
Sometimes, the quadcopters have tendency to increase their takeoff altitude during the taking off process. While in another cases, They fully respect the 10 meters altitude.
 
In following logs, we can see that the quadcopter take off altitude is about 10 meters + 5 meters elevated ground which equal to 15 meters (This is odd since we have told the quadcopter to go up only 10 meters):

https://logs.px4.io/plot_app?log=1c038323-8294-4aff-9799-07d2cca494be
https://logs.px4.io/plot_app?log=e07f68dc-330b-4d3d-a37a-11fad330ea2b

In this case every thing is fine, the quadcopters takeoff only 5 meters + 5 meters elevated ground which equal to 10 meters (Here everything is normal):

https://logs.px4.io/plot_app?log=f04e65c9-b145-456c-bfb5-4145577c3396
https://logs.px4.io/plot_app?log=e174cb28-47fe-4676-9ff9-c75f73c48a09

I have also noticed that the speed are different between going up and down, given the same speed as a parameters to body_velocity_function. I have noticed that the quadcotpers go up with higher speed than going down. Is it related to a fixed configuration in the firmware, quadcopters type, or some configurations?
Looking at the logs, it is noted that max_speed_up is different from max_speed_down

I do not know if it is the SDK or the PX4. However, I am afraid that this problem is related to the issue that I have posted before in PX4 
https://github.com/PX4/Firmware/issues/12206

Thanks for you help,

julianoes, 2019-08-20:
Regarding the takeoff speed, there are two params:
https://dev.px4.io/v1.9.0/en/advanced/parameter_reference.html#MPC_TKO_SPEED
https://dev.px4.io/v1.9.0/en/advanced/parameter_reference.html#MPC_LAND_SPEED

For the takeoff altitude, as far as I understood it, it should always end up 10 meters above the takeoff location no matter the ground altitude.

julianoes, 2019-08-20:
I think what happens in the case where it only goes up 6 meters instead of 10 meters, you're activating offboard too early when it is still climbing.

[image]

shrit, 2019-08-20:
@julianoes, I totally agree with you,
Now, I have better understanding about what is happening.
In fact, it is true I am switching too fast to offboard mode. This is why we need to have the landed state PR.
The only thing that confuse me, is the fact that the altitude measured by the quadcopters in SITL gazebo is the absolute altitude from the sea level. I think this is measured by the GPS.
But if this is the case, how does the quadcopter measure the altitude in Z during the taking off?
Is it possible to recover the true altitude value through the SDK? rather than GPS estimation.
Thanks

julianoes, 2019-08-20:
This should give you the altitude relative to ground:
https://github.com/mavlink/MAVSDK/blob/2f48667a80d01aa2f58b207f71d76957ab38c635/src/plugins/telemetry/include/plugins/telemetry/telemetry.h#L45

shrit, 2019-08-27:
Finally, after testing the value of relative altitude, it does not seems close to the one provided by simulator, I think it is initialized in px4 like GPS altitude and latitude for SITL. 
I think I need to use IRIS with optical flow to recover the true relative value correctly. To do so, I think we need to add automatic support in the SDK for optical flow quadcopters,

julianoes, 2019-08-28:
@shrit I agree: #780

julianoes, 2020-11-20:
This is outdated and I see no action to take, so closing it.

closed 2020-11-20 by julianoes