mavlink/MAVSDK #852: Support for incoming connections (ex from GCS)

pklapperich, 2019-09-10:
It would be nice if the SDK supported inbound connections as well as outbound connections. It would them be possible use the SDK to make a proxy application that sits between a drone and a groundstation. This could be useful for things like logging, injecting packets into the stream, etc.

JonasVautherin, 2019-09-10:
I wonder if that would not be more something for [mavlink-router](https://github.com/intel/mavlink-router/). What's your opinion on that?

pklapperich, 2019-09-10:
After a glance, I think that'll suite my needs out of box. And it's reassuring to see from [https://github.com/intel/mavlink-router/issues/164](issue 164) that it'll be around long term.

JonasVautherin, 2019-09-10:
Awesome!

closed 2019-09-10 by JonasVautherin