mavlink/MAVSDK #853: x86 vs ARM

tpennypacker, 2019-09-11:
Is this only usable on x86 processors? I just spent some time trying to get it to run on ARM, and was unsuccessful. If it's only x86 that should probably be stated somewhere in the readme.

JonasVautherin, 2019-09-12:
It does compile for iOS, Android, and arm generally. We have people compiling for RPis. What exactly is your issue?

AlexKlimaj, 2019-09-18:
If you follow the instructions in the link below, you should be able to build the library on ARM. I'm currently using it on a RPi. 

https://mavsdk.mavlink.io/develop/en/contributing/build.html

julianoes, 2019-09-19:
You can also cross compile using dockcross? @JonasVautherin are the instructions for that?

JonasVautherin, 2020-01-08:
Closing due to inactivity. Feel free to reopen if necessary.

closed 2020-01-08 by JonasVautherin