mavlink/MAVSDK #854: Unable to access IMU data on terminal using MAVSDK

deepak-1530, 2019-09-14:
I am not able to access IMU data of pixhawk over USB. I am able to access the data using MAVROS but not using the telemetry class provided in mavsdk.

JonasVautherin, 2019-09-14:
Can you elaborate a bit? What did you try exactly? How did you setup mavsdk, and what functions did you call?

deepak-1530, 2019-09-15:
I tried one of the sample codes (offboard_control) and in that i tried to display angular velocity data but i was getting zero. I also tried the IMU data but it again showed zero. 
 Telemetry::AngularVelocityBody v;
Telemetry::IMUReadingNED acc;
When i use mavros  and run the px4.launch file, then i can subscribe to the /imu/data topic and see the orientation and stuff but not without it.

JonasVautherin, 2019-09-15:
So you managed to create a connection to the drone? Does something else work, like arming and taking off, for instance?

deepak-1530, 2019-09-16:
I connected the pixhawk to my PC over USB. I was not able to arm it. But I only wanted to access the IMU data.

JonasVautherin, 2019-09-16:
> But I only wanted to access the IMU data.

Ok, but that's interesting information. If nothing works, then it is maybe not an issue specific to the IMU data :sweat_smile:.

Can you show me the output of MAVSDK when you connect it to the pixhawk? Also, does MAVSDK work with SITL for you?

deepak-1530, 2019-09-16:
[image] 
This is the output after I run the px4.launch command in mavros. I have not tried MAVSDK with SITL yet.

deepak-1530, 2019-09-16:
[image]
This is the output of mavros stuff (px4.launch). I have not tried MAVSDK with the SITL yet.

deepak-1530, 2019-09-16:
[image]
This is the output of mavros(px4.launch). I have not tried mavsdk on the sitl yet.

JonasVautherin, 2019-09-16:
I trust you when you say that it works with MAVROS. I want to see the output of MAVSDK in order to help you understand what is going wrong :+1:

julianoes, 2019-09-17:
@deepak-1530 by the way, you can copy console output using Ctrl+Shift+C. It makes it much more readable (and searchable).

deepak-1530, 2019-09-18:
[image]
This is the MAVSDK example code output. I just displayed the acceleration data

julianoes, 2019-09-18:
@deepak-1530:

What code are you using to print the accel values?

deepak-1530, 2019-09-18:
Telemetry::IMUReadingNED IMU;
std::cout<<telemetry->health()<<std::endl; 
int i = 1;
while(i>0)
{
std::cout<<" acceleration is (down) "<<IMU.acceleration.down_m_s2<<std::endl;
sleep_for(milliseconds(100));
}

This is the piece of code that I added in the takeoff_and_land.cpp example code to see the IMU readings.

JonasVautherin, 2019-09-19:
Hmm so you create an `IMU` object, and then print its value. That object was initialized with zeros as default values, so that's what you're printing.

julianoes, 2019-09-19:
@deepak-1530 you are printing an uninitialised struct.

You should do something like this:

```
while (true)
{
    Telemetry::IMUReadingNED imu = telemetry->imu_reading_ned();
    std::cout<<" acceleration is (down) "<<imu.acceleration.down_m_s2<<std::endl;
    sleep_for(milliseconds(100));
}
```

For more examples, see:
https://github.com/mavlink/MAVSDK/blob/develop/src/integration_tests/telemetry_sync.cpp

closed 2019-09-19 by julianoes