mavlink/MAVSDK #858: Add GPS UTC time to Telemetry::GPSInfo

AlexKlimaj, 2019-09-18:
Being able to timestamp events with GPS UTC time would be very useful in an application I am working on. Can the time_usec value from GPS_RAW_INT be added to Telemetry::GPSInfo?

https://mavlink.io/en/messages/common.html#GPS_RAW_INT

hamishwillee, 2019-09-18:
@AlexKlimaj Perhaps add a PR?

JonasVautherin, 2019-09-19:
Do you need the GPS time in [`GpsInfo`](https://github.com/mavlink/MAVSDK-Proto/blob/master/protos/telemetry/telemetry.proto#L200-L203) somehow, or did you mean [`Position`](https://github.com/mavlink/MAVSDK-Proto/blob/master/protos/telemetry/telemetry.proto#L147-L152)?

julianoes, 2019-09-19:
Are you sure that GPS_RAW_INT contains the UTC time?

AlexKlimaj, 2019-09-19:
@JonasVautherin it would make sense being in [GpsInfo](https://github.com/mavlink/MAVSDK-Proto/blob/master/protos/telemetry/telemetry.proto#L200-L203) since num_satellites and fix_type are also in [GPS_RAW_INT](https://mavlink.io/en/messages/common.html#GPS_RAW_INT).

@julianoes It looks like it is getting populated in PX4. [msg.time_usec = gps.timestamp;](https://github.com/PX4/Firmware/blob/fa13f63db6dc7b0131426aae90dbb99593ff2014/src/modules/mavlink/mavlink_messages.cpp#L1496)

AlexKlimaj, 2019-09-19:
It looks like you can grab the GPS time from [789](https://github.com/mavlink/MAVSDK/pull/798)?

julianoes, 2019-09-20:
> It looks like you can grab the GPS time from 789?

Oh, exactly. This should already work.

JonasVautherin, 2019-09-20:
In C++, yes. It hasn't been ported to the other languages. @AlexKlimaj: I guess you are interested in C++?

AlexKlimaj, 2019-09-20:
Yes, I am using the C++ library. 
One question I have about [789](https://github.com/mavlink/MAVSDK/pull/798) is why message [#340](https://mavlink.io/en/messages/common.html#UTM_GLOBAL_POSITION) was chosen instead of [#24](https://mavlink.io/en/messages/common.html#GPS_RAW_INT)? It seems like an odd choice.

JonasVautherin, 2019-09-20:
> One question I have about 789 is why message #340 was chosen instead of #24? It seems like an odd choice.

Probably @julianoes knows that better. But just for my information, why is that an odd choice?

AlexKlimaj, 2019-09-20:
The GPS_RAW_INT message is already broken out with num_satellites and fix_type. Also, the epoch_time is a value that is received from the GPS. It just makes more sense to go look in the GPS telemetry message for the time.

julianoes, 2019-09-21:
I agree but I think we didn't use GPS_RAW_INT.time_usec because in the description it says:
> Timestamp (UNIX Epoch time or time since system boot). The receiving end can infer timestamp format (since 1.1.1970 or since system boot) by checking for the magnitude the number.

So it's not necessarily the UTC time but can also be time since system boot.

AlexKlimaj, 2019-09-21:
Ah okay I see. So [#340](https://mavlink.io/en/messages/common.html#UTM_GLOBAL_POSITION) time will either be UTC or nothing?

> time - Time of applicability of position (microseconds since UNIX epoch).

julianoes, 2019-09-22:
> Ah okay I see. So #340 time will either be UTC or nothing?

I think it should be UTC. Why would it be nothing?

AlexKlimaj, 2019-09-22:
I mean, the autopilot won't populate it as time since boot. So it will be UTC time or not published at all by the autopilot.

closed 2019-09-22 by AlexKlimaj