mavlink/MAVSDK #869: Velocity issue with set_velocity_body() function

shrit, 2019-10-02:
Hello,

I am using iris quadcopter in an IRIS simulation. I am asking a quadcopter to move forward for 2 seconds at 1m/s. In this case I expect the quadcopter to move 2 meters. However, when calculating the traveled distance between the starting and the end positions, I have noticed that the quadcopter moves only for around 1.3 meters instead of 2 meters. This is the case for all actions executed on x and y plane (Forward, backward, left, right). However if I ask the quadcopter to move Up or Down (z axis) it will move correctly for 2 meters. 

Do anyone have an idea about what is happening?? Did I not understood the SDK correctly or this issue is related to PX4?

Best regards,

julianoes, 2019-10-02:
This behavior is expected because a quadcopter does not have infinite acceleration. You need to get up to the required speed first.

If you want to move 2 meters forward either use [position offboard control](https://github.com/mavlink/MAVSDK/blob/7cd87db701d3ddd2cda1dfcca4b777991138125a/src/plugins/offboard/include/plugins/offboard/offboard.h#L208-L213) or write your own cascaded controller around the velocity controller which closes the loop around a position and outputs a velocity setpoint.

closed 2019-10-02 by julianoes

shrit, 2019-10-02:
@julianoes, Thank you for your answer, that is very helpful.

The second option of writing a cascaded controller seems to be more useful if no positining system is used even real-life scenarios.  I think this can be done by calculating the traveled distanced using only the accelerometer values from the SDK during the movement.

julianoes, 2019-10-07:
Integrating velocity will be more accurate though than integrating accel values because you're adding up more noise when double integrating.