mavlink/MAVSDK #871: mavsdk_server crashes when a geofence is uploaded

JonasVautherin, 2019-10-03:
`mavsdk_server` gets a segmentation fault when uploading a geofence over gRPC.

Steps to reproduce:

1. Build `mavsdk_server` from #870.
2. Checkout MAVSDK-Python into the [python geofence PR](https://github.com/mavlink/MAVSDK-Python/pull/114)
3. Run jmavsim and the `mavsdk_server` built in 1 above.
4. Run `examples/geofence.py` in MAVSDK-Python fetched in 2 above.
5. The geofence is uploaded successfully, but `mavsdk_server` crashes.

@coderkalyan: Would you mind having a look at this? :blush:.

coderkalyan, 2020-01-23:
@JonasVautherin Sorry I forgot about this. I will look into it ASAP. I was having trouble recreating the error, but now that my friend bbworld1 is having the same problem I have something to go off of.

JonasVautherin, 2020-01-24:
@coderkalyan I've been debugging that, and my observation is that the callback given to `send_geofence_async(...)` is called twice: once with result `Success`, and another time with result `Timeout` (which doesn't make sense to me if it succeeded before).

`mavsdk_server` crashes on the second call (with `Timeout`), because it expects the callback to be called exactly once.

Would you mind investigating why the timeout is called after a `Success` result is sent?

julianoes, 2020-01-24:
It could be that it is something stupid in the geofence plugin with all the callbacks and locks happening there. I'm planning to refactor the mission and geofence plugin to clean these up, hopefully next week.

coderkalyan, 2020-01-29:
@julianoes please keep me updated when you're done with that, so I can add the changes to this!

julianoes, 2020-01-29:
I'm working on it: #964.

coderkalyan, 2020-03-24:
@julianoes Does your refactoring change anything about geofence that I should know about? Still trying to fix this PR.

@JonasVautherin any suggestions are appreciated, I still can't find the source of the problem :(

julianoes, 2020-03-25:
@coderkalyan good question. I would have to look into it and try it out.

coderkalyan, 2020-05-17:
@julianoes This can be closed, I think, the autogenerated backend from the latest release has AFAIK solved the problem.

closed 2020-05-18 by JonasVautherin