mavlink/MAVSDK #885: Landed State is jumping between modes

shrit, 2019-10-22:
Hello, 

I am doing simulation for three quadcopters in SITL Gazebo, I need to make the quadcopter fly at the same time and reach high take-off altitude. 
To do this, I use three threads and a while sleep inside the take-off function until the quadcopters finish their taking off procedure.
The problem arises when the first while loops in the takeoff function blocks on unwanted states. After many observations, I have noticed that at least in one quadcopter LandedState jump from ON_GROUND to IN_AIR  directly without passing by TAKING_OFF.
These are the while loop from the takeoff function:
```

  /*  Wait until the landed state changes the mode */
  while (true) {
    std::this_thread::sleep_for(std::chrono::milliseconds(5));
    if (landed_state_ == Telemetry::LandedState::TAKING_OFF) {
      LogInfo() << "Landed State : "<< telemetry_->landed_state_str(landed_state_);
      break;
    }
    LogInfo() << "Landed State : "<< telemetry_->landed_state_str(landed_state_);
  }
  while (landed_state_ == Telemetry::LandedState::TAKING_OFF) {
    LogInfo() << "Taking off..." ;
    std::this_thread::sleep_for(std::chrono::milliseconds(300));
  }
  LogInfo() << "Taking off has finished successfully...";
```
Knowing that is landed state is recovered asynchronous from the SDK
Do you have any idea about this issue??

JonasVautherin, 2019-10-23:
That's interesting... did you check the message coming from PX4? Does PX4 say `IN_AIR`, or does it say `TAKING_OFF` and MAVSDK somehow reports the wrong state?

shrit, 2019-10-23:
Nope, I have no idea how to check the PX4 messages, In fact, the SDK report rarely `TAKING_OFF` but most of the time, I receive `IN_AIR` instead.
The only simple hack I had to do is to replace `landed_state()` function by `flight_mode()` since there is a TAKEOFF flight mode.

julianoes, 2019-10-24:
I think this state is only sent at 1Hz. Is it possible that you're just missing the update between `LANDED` and `IN_AIR`?

shrit, 2019-10-24:
@julianoes, Ok I understand, I have to test it without the 50 milliseconds sleep.

shrit, 2019-10-26:
I am not sure if I understand correctly, the above code might be able to miss out on the transition between `ON_GROUND` and `TAKEOFF` _since the state is sent only once each second_.
In this case, it might be true that I am missing it as the above code sleep for 5 milliseconds, 

To resolve this issue I have only changed the `Landed_state` to `flight_mode`, I do not know what is the rate for sending the flight_mode, but now I never miss the Taking off mode.

@julianoes any thoughts?

julianoes, 2019-10-28:
The flight mode is sent with every `heartbeat`, usually at 1Hz. The landed state is sent with `extended_sys_state` at 1 Hz for the normal link on 14550 but at 5 Hz for the onboard link on 14540.

See:
https://github.com/PX4/Firmware/blob/7f63ed8202a0f9d74181cdd8e40624565947cbc9/src/modules/mavlink/mavlink_main.cpp#L1665
https://github.com/PX4/Firmware/blob/380cae18d199d1c237517f04d181468a7cad1420/ROMFS/px4fmu_common/init.d-posix/rcS#L246

shrit, 2019-10-28:
@julianoes Thanks for your answer,
I think this is very odd since I am using onboard link for landed state and I am always missing it.
However, I am receiving flight mode only once each second and I am never missing it. 
Do you agree that there is some kind of bug here?

julianoes, 2019-10-28:
I don't know. I suggest you either debug it yourself and then create a pull request to fix it, or you create a minimal example that I can use to reproduce it. From just this code sample and information I can't see it.

shrit, 2019-10-28:
Ok, I will do a pull request in an example to reproduce the error, since I have no idea where to start debugging.

julianoes, 2019-11-13:
I think I just ran into this as well:
```
INFO  [commander] Takeoff detected
[05:14:26|Debug] MAVLink: info: Takeoff detected (system_impl.cpp:304)
landed state: On ground(0.024)
landed state: On ground(1.26)
landed state: On ground(1.838)
landed state: On ground(2.136)
landed state: On ground(2.316)
landed state: On ground(2.401)
landed state: On ground(2.462)
landed state: On ground(2.481)
landed state: In air(2.497)
```

shrit, 2019-11-20:
@julianoes This issue is resolved now, right?

julianoes, 2019-11-20:
It will be when https://github.com/PX4/Firmware/pull/13075 is merged.

closed 2019-12-26 by LorenzMeier