mavlink/MAVSDK #897: traceMode feature like DJI

dangviethieu, 2019-11-05:
Hello all. Please help me with the question.
In DJI, they have the mission mode: traceMode, that is smooth transition between waypoints.
https://developer.dji.com/onboard-api-reference/structDJI_1_1OSDK_1_1WayPointInitSettings.html#a0f2517fd1775e16675921b8b62b919ef
And this is demo: https://gyazo.com/a724d2ba73a51775cc6c37ed23dc29b1
So I wonder that MAVSDK is have the feature same that?

julianoes, 2019-11-05:
What exactly are you trying to achieve? If you need to stop at a waypoint in order to take a picture, yes we already have that. You can use `fly_through = false` and set a camera action, e.g. `TAKE_PICTURE`.

dangviethieu, 2019-11-15:
@julianoes Thank you very much about quickly reply!
[image]
I mean the DJI has the flights mode like: drone will not reach at waypoint, it will fly through the curve corner with the distance as input value. It look like the VTOL mode.

julianoes, 2019-11-15:
Aha, I think now I understand. We have several waypoint smoothing mechanisms. I don't know much about it but I'm hoping that @bresch can help here.

dangviethieu, 2019-11-21:
Hi @bresch @julianoes . I try research about **smoothing mechanisms** of several waypoint.
[image]
I found that we can use the key: **acceptance_radius_m**, it look like curve corner when drone reach at waypoints. I try changed it, but the result isn't as expected.
Please help me understand more about smoothing mechanisms if we have it? Thanks you very much !!!

bresch, 2020-03-04:
@julianoes @dangviethieu if you want a nicer curve, you need to increase the acceptance radius more than 3m, you could try larger values (10m for example).

julianoes, 2020-11-20:
Closing this as the behaviour can be achieved as @bresch writes above in https://github.com/mavlink/MAVSDK/issues/897#issuecomment-594594163.

closed 2020-11-20 by julianoes