mavlink/MAVSDK #902: Building Mavlink_Passthrough does not build the plugin

felix-west, 2019-11-14:
As the instructions say for building the passthrough plugin, I first installed MAVSDK and built/tried some of the examples. When I wanted to install the mavlink_passthrough plugin however, I used the command "make ENABLE_MAVLINK_PASSTHROUGH=1" in the ".../MAVSDK/build/default" directory and got no error. However, when looking at the output, I get things like:
"[ 44%] Built target mavsdk_action
[ 46%] Built target mavsdk_follow_me
[ 48%] Built target mavsdk_geofence"
which seems ok. However, there is no reference to it actually building or doing anything with the mavlink_passthrough plugin. I double checked the ".../build/default/src/plugins/" and the "/usr/local/include/mavsdk/plugins/" directories but again found no mention of the plugin being there. 
How do I build the passthrough plugin to be able to include it in the given example code?

julianoes, 2019-11-14:
This is the command to build with the passthrough plugin:

```
cmake -DCMAKE_BUILD_TYPE=Debug -DBUILD_SHARED_LIBS=ON -DENABLE_MAVLINK_PASSTHROUGH -Bbuild/default -H.
```

felix-west, 2019-11-15:
Thanks for the help. When remaking the build directory like in the installation guide with the command 
" cmake -DCMAKE_BUILD_TYPE=Debug -DBUILD_SHARED_LIBS=ON -DENABLE_MAVLINK_PASSTHROUGH=ON -Bbuild/default -H. " it worked and the mavlink_passthrough plugin is now in my "/usr/local/include/mavsdk/plugins/" directory. 
I'm guessing it would work the same way with the logging plugin with "-DENABLE_LOGGING=ON" instead?

julianoes, 2019-11-15:
The logging plugin is currently commented out because it's not implemented completely:
https://github.com/mavlink/MAVSDK/blob/480a93ce9b8392e29ab2a8f995258c1d58722df6/src/plugins/CMakeLists.txt#L14

julianoes, 2020-11-27:
The MavlinkPassthrough plugin is now built by default.

closed 2020-11-27 by julianoes