mavlink/MAVSDK #903: How to add "src/core" functions to example projects

felix-west, 2019-11-14:
I'm currently trying to add some functionality to my drone. To do that I first need to be able to give the drone different commands over the serial link. However, the functionality provided in the various plugins found in ".../MAVSDK/src/plugins/" is not enough for this. I found various pieces of code in the ".../MAVSDK/src/core/" directory. 
When I try to include these files directly in my example code (using the absolute path in the includes) the build fails. Also, when I try to include some of the header files found in "/usr/local/include/mavsdk/plugins/mavlink_ftp/mavlink/v2.0/common/" in an attempt to send messages manually it also fails to build(particularly as an attempt to try to get at least some functionality found in the core to work)
How do I go about adding the functions found in the ".../MAVSDK/src/core/" directory to my own code? So far I only got the example code with minimal modifications to work.

julianoes, 2019-11-14:
I don't know exactly what you're doing. I suggest you either post a diff, or share a branch in a pull request where I see the diff, or you need to at least show the whole error message.

felix-west, 2019-11-15:
Essentially what I want to do is use functions implemented in the file ".../MAVSDK/src/core/mavlink_commands.cpp" in my own test program. Right now I got that part to work by including the absolute path of the header file "/home/.../core/mavlink_commands.h" and changing the contents of the "mavlink_include.h" file in the "core" directory from "#include "mavlink/v2.0/common/mavlink.h"" to the absolute path: "#include "/usr/local/include/mavsdk/plugins/mavlink_ftp/mavlink/v2.0/common/mavlink.h""
With that change I got the test project to make as well as to be able to include the implemented functions contained in the "mavlink_commands.cpp" file. 

However, now I'm having issues actually sending a message. What I want to do is to send a "mavlink_command_long_t" with my own command. I've tried using including the passthrough plugin and packing a command_long message into the "mavlink_message_t" before sending it. This built and worked, giving me no errors. However, when using the mavlink inspector of QGC I found that none of the messages that I sent were received by the drone, even if the corresponding inputs were all set to broadcast. 
I then tried using the functions used in ".../MAVSDK/src/core/mavlink_commands.cpp" by adding it in the way I described above. In time I got it to the point where QT creator showed me no errors. However building it resulted with the following error: 

```
/data/home/felix/src/MAVSDK/example/mission_test/mission_test.cpp:191:60:   required from here
/usr/include/c++/7/ext/new_allocator.h:136:4: error: no matching function for call to ‘mavsdk::MAVLinkCommands::MAVLinkCommands(mavsdk::System&)’
  { ::new((void *)__p) _Up(std::forward<_Args>(__args)...); }
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /data/home/felix/src/MAVSDK/example/mission_test/mission_test.cpp:24:0:
/home/felix/src/MAVSDK/src/core/mavlink_commands.h:16:14: note: candidate: mavsdk::MAVLinkCommands::MAVLinkCommands(mavsdk::SystemImpl&)
     explicit MAVLinkCommands(SystemImpl& parent);
              ^~~~~~~~~~~~~~~
/home/felix/src/MAVSDK/src/core/mavlink_commands.h:16:14: note:   no known conversion for argument 1 from ‘mavsdk::System’ to ‘mavsdk::SystemImpl&’
```
I haven't seen any real discussions or guides as to how to manually send messages on MAVSDK, only on px4 and such. However, I need to be able to send them to continue. As such, I'm completely lost right now. I've included the program that I'm testing on as well as the corresponding cmakelists.  

[CMakeLists.txt](https://github.com/mavlink/MAVSDK/files/3850514/CMakeLists.txt)

[mission_test.txt](https://github.com/mavlink/MAVSDK/files/3850510/mission_test.txt)

julianoes, 2019-11-15:
You are changing all over the place here and mixing internals of the library with externals. That's really not how this should be done and I don't know how to help you.

Can we take a step back and you can explain to me high level what you're actually trying to achieve and then maybe I can guide you how to do that?

felix-west, 2019-11-15:
Ok, now I got the message to show in QGC as a "command_ack" message when I do the following: 
`mavlink_msg_command_long_pack(passthrough->get_our_sysid(), passthrough->get_our_compid(), &test_msg,1,1,31010,0,0,0,0,0,0,0,0);
passthrough->send_message( test_msg);`
However, sending a broadcast message does not seem to make anything show up in the MAVlink inspector QGC. 

What I'm trying to do is the following: I want to have several system communicating with one another over MAVlink, the companion computer connected over RC mavlink, the flight controller, as well as 1-2 or more other components. What I essentially want is that the companion computer gives messages to the flight controller to give it missions, etc. However, at some point the companion computer needs to give signals to the other components so that they complete certain tasks. One of those tasks includes the computer giving up flight control and giving it completely to the onboard components. That is why these components need to be able to communicate with the flight controller directly. Once their tasks are done, these additional components need to give the corresponding signal so that other components know what to do, either start their own tasks or to signal task complete. 

That is why I'm currently trying to send custom messages, because the normal flight control messages are not enough to designate the commands to all participants. 

I know that I'm mixing things and breaking things, but essentially all I need right now is to send and receive command messages. After some more testing I think it might work with the passthrough functions, but I'm still confused because the messages don't show up at all or they do but not as a command_long message as I would have expected in my simulation.

julianoes, 2019-11-15:
Ok still sounds messy but I can somewhat get a glimpse. But all in all it sounds like you need some custom functionality.

Therefore, my suggestion would be to write your own custom plugin and put all the functionality in there. Then you can e.g. use the existing mission or mission_raw plugin for your missions but use your own plugin for your custom commands.

To create a new custom plugin you would copy an existing plugin folder and then change the filenames and class names etc to your custom plugin. Then you also need to add the plugin to the [CMakeLists.txt](https://github.com/mavlink/MAVSDK/blob/develop/src/plugins/CMakeLists.txt) file so it gets compiled.

So what I'm saying is that you should extend the library (MAVSDK) itself from within and not try to add more functionality inside the example which is using the API from external. The way you started including random header files etc. is going to be a disaster!

One more thing, if you also need your own mavlink messages that are not in the standard but added by you e.g. in a separate mavlink dialect like mydialect.xml or added to common.xml then you need to generate the C headers for your messages and then replace the existing ones of this library.

The place to change this would be in the [mavlink_include.h](https://github.com/mavlink/MAVSDK/blob/develop/src/core/mavlink_include.h) once and for all.

I hope that helps.

And one more thing:
Before you change all of how PX4, MAVSDK, MAVLink etc. works, think if you can make your use case work within the existing ways of doing things. You're going to have a much easier time.
For instance, for control from a companion computer you could use offboard control, just as an example.

felix-west, 2019-11-16:
I guess I got mostly confused because I saw that in the action plugin in the "action_impl.cpp" the functions used were mostly the ones included in the ".../src/core/mavlink_commands.cpp" file, whereas in the example code you only have direct access to the plugins. With the mavlink_passthrough plugin however, I have direct access to send the messages with the correct commands as needed. 

So far I've been able to issue simple commands such as takeoff and loitering by manually sending the corresponding command and filling out the 7 parameters on my own. As such, I think I should have a good enough grasp of how to continue from here on out to create my own commands, subscribe to the command_long message type and make a callback function to do stuff in the corresponding peripheral controllers. 

However, I'm still having one particular issue. When I leave the target system in the function "mavlink_msg_command_long_pack(...)" included in the "/usr/local/include/mavsdk/plugins/mavlink_passthrough/mavlink/v2.0/common/mavlink_msg_command_long.h" file as 0 and the target component also at 0, it should send a message with the corresponding data to every system and every component according to the documentation. However, I'm finding that it doesn't actually send anything when I do that. Only when I specify the the target system and the target component does the command of the message actually execute. Knowing this, I can easily adjust to it. However, I feel like I'm still doing something wrong or haven't completely understood WHY it's not doing what I think it should be doing. How should I go about correctly to send a broadcasted message that will be read and (if applicable) executed on all listening systems?

Also, maybe to get back to my original question as to how to include functions used in the "MAVSDK/src/ore" directory in my code, I think your description of how to continue is the only way to not break things. I saw that the mentioned functions are used in various plugins. Like you said, it's best to write my own plugin using whatever functions I need and then calling the plugin functions in the actual code.

julianoes, 2019-11-19:
> So far I've been able to issue simple commands such as takeoff and loitering by manually sending the corresponding command and filling out the 7 parameters on my own.

Right, that's good but note that you don't get command retransmission etc. this way. The way to get that is to use the methods that are used in plugins.

> it should send a message with the corresponding data to every system and every component according to the documentation.

I think so too but you would have to debug in detail what is happening. One thing is that the message might not be sent out because MAVSDK thinks it doesn't know who 0 is and doesn't send it at all.

> Like you said, it's best to write my own plugin using whatever functions I need and then calling the plugin functions in the actual code.

Exactly, that would make most sense, I think.

julianoes, 2020-11-25:
It's been a while, I'm closing this.

closed 2020-11-25 by julianoes