mavlink/MAVSDK #904: IOS Build of MAVSDK throws error

douglaswsilva, 2019-11-14:
Using:
- MacOS Mojave 10.14.6
- CMake 3.15

Command used to build:
```
caffeinate -i /Applications/CMake.app/Contents/bin/cmake -DBUILD_BACKEND=ON -DCMAKE_BUILD_TYPE=Release -DBUILD_SHARED_LIBS=OFF -DCMAKE_TOOLCHAIN_FILE=./tools/ios.toolchain.cmake -DPLATFORM=OS -DCMAKE_INSTALL_PREFIX=build/ios -Bbuild/ios -H.
```

Error:
```
cmake -DBUILD_BACKEND=ON -DCMAKE_BUILD_TYPE=Release -DBUILD_SHARED_LIBS=OFF -DCMAKE_TOOLCHAIN_FILE=./tools/ios.toolchain.cmake -DPLATFORM=OS -DCMAKE_INSTALL_PREFIX=build/ios -Bbuild/ios -H.
-- PX4 version: v1.10.0-beta4-156-g02271a471d
-- PX4 config file: /Users/julianoes/src/Firmware/boards/px4/sitl/default.cmake
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- cmake build type: Release
CMake Error at /usr/local/Cellar/cmake/3.15.4/share/cmake/Modules/CMakeDetermineSystem.cmake:99 (message):
  Could not find toolchain file: ./tools/ios.toolchain.cmake
Call Stack (most recent call first):
  CMakeLists.txt:189 (project)

CMake Error: CMake was unable to find a build program corresponding to "Unix Makefiles".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_ASM_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!
```

JonasVautherin, 2019-11-28:
Can you try to pass an absolute toolchain path?

```
-DCMAKE_TOOLCHAIN_FILE=/path/to/tools/ios.toolchain.cmake
```

julianoes, 2019-11-28:
@JonasVautherin yes, we solved that actually, we had to do:

```
cmake -DBUILD_BACKEND=ON -DCMAKE_BUILD_TYPE=Release -DBUILD_SHARED_LIBS=OFF -DCMAKE_TOOLCHAIN_FILE=$(pwd)/tools/ios.toolchain.cmake -DPLATFORM=OS -DCMAKE_INSTALL_PREFIX=build/ios -Bbuild/ios -H.
```

closed 2019-11-28 by julianoes