mavlink/MAVSDK #907: How to  use the gimbal properly?

MicroYY, 2019-11-19:
The sdk provides an implementation of gimbal.
I tried TYPHOON480 which is equipped with a gimbal and successfully set pitch and yaw angle using the command `vmount test yaw 30`.
I use the python wrapper in my project, the gimbal seems to work when I use the `drone.action.takeoff()` and then `drone.gimbal.set_pitch_and_yaw`.
But when it comes to the OFFBOARD mode, given a velocity to takeoff and then `drone.gimbal.set_pitch_and_yaw`, the gimbal will shakely very badly.
I don't know whether the C++ SDK has this problem. Maybe there is something wrong with my configuration.
So how to use the gimbal properly? Would you offer an example?

julianoes, 2019-11-19:
Thanks for the note. That's odd. @Jaeyoung-Lim have you seen that the Typhoon H480 gimbal would behave differently in offboard?

Jaeyoung-Lim, 2019-11-19:
On my side no, @MicroYY can you also try takeoff and control the gimbal without entering offboars mode?

If you are using a older version of the firmware, the gimbal was overturned and that might explain the oscillations after takeoff, but it is not related to offboard mode

MicroYY, 2019-11-19:
@Jaeyoung-Lim 
I tried directly takeoff and control the gimbal, and it worked.
But when i entered offboard and then takeoff, it didn't work.

I have just updated to the latest firmware of PX4, but the problem remains.
And this time, i use QGC to takeoff, the image shakes even harder.

Jaeyoung-Lim, 2019-11-19:
@MicroYY Which setpoints are you sending in offboard mode?

MicroYY, 2019-11-19:
@MicroYY 
I just test the python script 

> [offboard_velocity_ned.py](https://github.com/mavlink/MAVSDK-Python/blob/master/examples/offboard_velocity_ned.py)

and add ` await drone.gimbal.set_pitch_and_yaw()` after `await drone.offboard.set_velocity_ned(VelocityNedYaw(0.0, 0.0, -2.0, 0.0))` is executed.
It seems the gimbal still doesn't work.

And I test  an script

> [takeoff_and_land.py ](https://github.com/mavlink/MAVSDK-Python/blob/master/examples/takeoff_and_land.py)

and add the same command after ` await drone.action.takeoff()`, the gimbal works.

The firmware and MAVSDK-Python are all the latest. I use `roslaunch` to start gazebo and a depth camera is mounted in the gimbal. I want depth map and publish it to a rostopic. Rviz shows the image caputred by the camera.

This is my roslaunch file.

[posix_sitl.launch.txt](https://github.com/mavlink/MAVSDK/files/3863637/posix_sitl.launch.txt)

julianoes, 2020-11-23:
Closing this as it's outdated. Please re-open if the issue is still there.

closed 2020-11-23 by julianoes