mavlink/MAVSDK #909: Yaw/heading not accessible in MissionItem class

mwshafer, 2019-11-20:
It seems that `MissionItems` don't include a vehicle heading, which would be really helpful. This could be accomplished by adding the following methods to the `MissionItem` class:

- `yaw_to()` (double degrees from N): Rotate the vehicle to the specified angle after reaching position set by `set_position()`.
- `yaw_for()` (double seconds): Rotate the vehicle for the specified time after reaching position set by `set_position()` at rate specified by `set_yaw_rate()`.
- `set_yaw_rate()` (double rotation speed in deg/s): Set the target rotation rate for vehicle yawing. 
- `set_yaw_departure_ned` (double degrees from N): Set the heading for after mission item (transition to next mission item)

I am not sure the last method would be needed if the heading specified by `yaw_to()` could be maintainted between waypoints. If it could, the last item would be redundant. If multiple headings are desired at a given position, they could simply be stacked as a series of mission items with the same position, but differing values set by `yaw_to()`. 

I have a project wherein we need to be able to specify the vehicle heading because we have a directional antenna attached. I'd like to transition to Mavsdk, but need this heading control capacity.

julianoes, 2019-11-20:
Would you prefer the vehicle to always have the heading set to a specific direction or only at waypoints? The reason that we usually fly "forward" to the next waypoint is because we generally have obstacle avoidance facing forward and can then use gimbals to look around if required.

mwshafer, 2019-11-20:
I think that maybe by default, MAVSDK could use the fly "forward" feature that you mentioned for the reasons you highlighted. The `set_yaw_departure()` would be used to override that feature only if the mission designer requires it. The documentation would just need a notice that obstacle avoidance sensors wouldn't be pointed in the direction of flight if that feature is used. What do you think?

If this gets added in phases, for me, the first three of the methods listed are most crucial. I listed them in order of necessity. The last one would be really helpful eventually, but isn't absolutely critical. I am still new to this open source development process. How long do you think adding such features might take? Thanks!

mwshafer, 2019-11-21:
I just realized that a `set _yaw_dir` would also be needed to specify the direction of rotation. The direction of rotation could alternatively, be set using a second input argument in the `yaw_to` and `yaw_for` methods.

julianoes, 2019-11-22:
> The last one would be really helpful eventually, but isn't absolutely critical. I am still new to this open source development process. How long do you think adding such features might take?

@mwshafer, that's a good question. This is an open source project so no one has to do anything as such. And whoever needs something will have to make a pull request and contribute what they need. 
This being said, us developers we are interested in making people get use out of the software and we try to fix what is broken and add what is asked for. The timeline for a feature then depends on what else is going on, how many developers ask for something, and how well it is aligned with what the project should enable and what we as maintainers need anyway for their own/internal projects.

For this specific feature I would argue that it is not unreasonable and something that will come up for others as well at some point. Therefore, it makes sense to add it in the next few weeks, at least that is my goal but I can't promise it.

mwshafer, 2020-01-30:
@julianoes Do you know if this as seen any development. I am working on a funding proposal and was recently thinking about yawing flight modes or control again. Just thought I would check in and reiterate that this would be extremely helpful. Thanks!

julianoes, 2020-01-30:
I'm currently working on refactoring the mission transfer (#964) in order to make it easier to change the mission plugin and also add things like geofence, rally point upload etc.

Once that's done we can surely revisit the yaw settings, it should then be less overwhelming to change it. And as an alternative you could also use the `mission_raw` plugin which allows any sort of MAVLink mission items.

mwshafer, 2020-01-30:
Great to hear! I'll check in on the `mission_raw ` plugin. Thanks.

julianoes, 2020-01-31:
The `mission_raw` plugin is currently a bit crippled because it only implements download but not upload but that will change after the refactor:

https://github.com/mavlink/MAVSDK/blob/c5404eb7ca88b6a404b82154d7ad49b3cc6b7d17/src/plugins/mission_raw/include/plugins/mission_raw/mission_raw.h#L100-L108

overdrum, 2020-03-26:
I don't know how to handle that...i.e., MissionRaw::MavlinkMissionItemInt::param4. 
Would somebody explain that?

nightsparc, 2020-04-09:
> I'm currently working on refactoring the mission transfer (#964) in order to make it easier to change the mission plugin and also add things like geofence, rally point upload etc.
>
> Once that's done [...]

Hi! Hope you all are alright - ah, #964 seems to be merged into dev now :) great! Do you already have any plan or schedule to extend the `MissionItem` to include yaw and the other stuff when the COVID19-shutdowns are coming to an end? I'd rather not using `MissionRaw` or going back to native MAVlink handling if I could avoid it :)

mwshafer, 2020-04-27:
@julianoes I see you removed the 'feature request' tag and added the 'feature' tag to this issue. What is the difference? Is this issue getting developed now? Thanks!

julianoes, 2020-04-28:
> I see you removed the 'feature request' tag and added the 'feature' tag to this issue.

I was just consolidating the tags. Once we are done with the auto-generation we can consider handling yaw better in the mission plugin. Until then we can only recommend to use mission_raw which should have most features as of the next release.

mwshafer, 2020-12-02:
@julianoes Any updates on where this might be in the queue? Thanks.

julianoes, 2024-05-27:
Looks like there is yaw:
https://github.com/mavlink/MAVSDK-Proto/blob/34bce6d61dba5f72b602bdbd31f04fceff093881/protos/mission/mission.proto#L210

closed 2024-05-27 by julianoes