mavlink/MAVSDK #912: FlightMode options for MAVSDK limited to Auto mode and Offboard

wsolichin-bto, 2019-11-22:
So currently, with MAVSDK, if the PX4 is either in Position / Altitude / Stabilized / Manual Mode, it will be displayed as Unknown in MAVSDK using Telemetry Flight Mode from API Reference here: https://mavsdk.mavlink.io/develop/en/api_reference/classmavsdk_1_1_telemetry.html#classmavsdk_1_1_telemetry_1aae86fcb399b0368209fcbd35ecdd5fad. it will be very useful for monitoring the state of the drone if it can for example go to Altitude Mode or Position Mode to test if there's either height source sensors or position source sensors. Is there any other way to get the rest of the flight modes mentioned?

julianoes, 2019-11-25:
I agree that it can be useful to be able to know the flight mode in general, however, I don't think it should be used to check the state of the position estimation (e.g. if position control is possible or only altitude control. This information should be provided by status flags. That status information is a known problem and something that @MaEtUgR is working on.

bozkurthan, 2019-12-24:
@wsolichin-bto can you check now?
Other modes can be received by using telemetry plugin.

julianoes, 2020-01-17:
The modes have been added in #945.

closed 2020-01-17 by julianoes