mavlink/MAVSDK #914: fly_qgc_mission issue: Mission int messages not supported

petergerten, 2019-11-26:
When trying to upload a QGC plan file via fly_qgc_mission I get the below error:

```
[07:50:24|Info ] MAVSDK version: 0.21.4 (mavsdk_impl.cpp:25)
Connection URL: udp://:14550
Importing mission from mission plan: testmission01.plan
Waiting to discover system...
[07:50:25|Info ] New system on: 192.168.42.1:14551 (udp_connection.cpp:263)
[07:50:25|Debug] New: System ID: 1 Comp ID: 100 (mavsdk_impl.cpp:390)
[07:50:25|Debug] Component Camera 1 (100) added. (system_impl.cpp:401)
[07:50:25|Debug] Discovered 1 component(s) (UUID: 1) (system_impl.cpp:561)
Discovered system with UUID: 1
System ready
[07:50:25|Warn ] UNSUPPORTED mission item command (20) (mission_impl.cpp:1541)
Found 2 mission items in the given QGC plan.
Uploading mission...
[07:50:25|Warn ] Mission int messages not supported (mission_impl.cpp:366)
Mission upload failed: Error
```
What does "Mission int messages not supported" mean? (My mission is a trvial mission to fly to one waypoint created in QGC).

Is the "Mission upload failed: Error" due to the above warning or might be unrelated (as it is a warning and not an error)?

julianoes, 2019-11-26:
It looks like you are connecting to a camera only but no autopilot and given the error it then finds that the camera doesn't support mission upload. Could that be the case?

petergerten, 2019-11-26:
I just noticed that I get the same error message even when only using the` fly_mission` example:
```
[07:55:28|Info ] MAVSDK version: 0.21.4 (mavsdk_impl.cpp:25)
Waiting to discover system...
[07:55:28|Info ] New system on: 192.168.42.1:14551 (udp_connection.cpp:263)
[07:55:28|Debug] New: System ID: 1 Comp ID: 100 (mavsdk_impl.cpp:390)
[07:55:28|Debug] Component Camera 1 (100) added. (system_impl.cpp:401)
[07:55:28|Debug] Discovered 1 component(s) (UUID: 1) (system_impl.cpp:561)
Discovered system with UUID: 1
System ready
Creating and uploading mission
Uploading mission...
[07:55:28|Warn ] Mission int messages not supported (mission_impl.cpp:366)
Mission upload failed (Error), exiting.
```

petergerten, 2019-11-26:
@julianoes It is a Yuneec H520 where the main connection is made to the camera, but the flightcontroller is connected on the same mavlink link as well.

petergerten, 2019-11-26:
```
[08:00:32|Info ] MAVSDK version: 0.21.4 (mavsdk_impl.cpp:25)
Connection URL: udp://:14550
Importing mission from mission plan: testmission01.plan
Waiting to discover system...
[08:00:33|Info ] New system on: 192.168.42.1:14551 (udp_connection.cpp:263)
[08:00:33|Debug] New: System ID: 1 Comp ID: 100 (mavsdk_impl.cpp:390)
[08:00:33|Debug] Component Camera 1 (100) added. (system_impl.cpp:401)
[08:00:33|Debug] Discovered 1 component(s) (UUID: 1) (system_impl.cpp:561)
Discovered system with UUID: 1
[08:00:33|Debug] Component Autopilot (1) added. (system_impl.cpp:401)
[08:00:33|Debug] MAVLink: info: data link #0 lost (system_impl.cpp:308)
[08:00:33|Debug] MAVLink: info: data link #0 regained (system_impl.cpp:308)
[08:00:33|Debug] Component Gimbal (154) added. (system_impl.cpp:401)
System ready
[08:00:38|Warn ] UNSUPPORTED mission item command (20) (mission_impl.cpp:1541)
Found 2 mission items in the given QGC plan.
Uploading mission...
[08:00:38|Warn ] Mission int messages not supported (mission_impl.cpp:366)
Mission upload failed: Error
```
Here it detected the autopilot and I got an additional warning regarding (UNSUPPORTED mission item command (20) ) but still the same error message

petergerten, 2019-11-26:
ok, I found it in the code in mission_impl.cpp:

```
    if (!_parent->does_support_mission_int()) {
        LogWarn() << "Mission int messages not supported";
        report_mission_result(callback, Mission::Result::ERROR);
        return;
    }
```
So the warning does result in the mission upload error.
I could not figure out what `does_support_mission_int()` is or what is does though.

julianoes, 2019-11-26:
Oh you're right, the autopilot is also found:
```
[08:00:33|Debug] Component Autopilot (1) added. (system_impl.cpp:401)
```

That's odd then.

julianoes, 2019-11-27:
I can reproduce the problem. I'm looking into it.

julianoes, 2019-11-27:
Ok, I believe this is a problem of the H520. It seems like the latest version does not send a heartbeat from the autopilot anymore.

@darioxz @Stifael could you check this?

petergerten, 2019-11-27:
@julianoes thanks for looking into it!

I actually can reproduce the issue with px4 SITL as well.
Can you provide some background on what does_support_mission_int() means?

julianoes, 2019-11-27:
> @darioxz @Stifael could you check this?

Nevermind! It's fine in QGC, so it's probably a regression in MAVSDK.

julianoes, 2019-11-27:
>  Can you provide some background on what does_support_mission_int() means?

Don't worry about this in specific. It's just a side effect given that the heartbeat and autopilot version don't arrive.

julianoes, 2019-11-27:
Ok so here is what is happening:

1. The H520 sends heartbeats from the camera only at first.
2. MAVSDK does not send heartbeats back unless it manages to initialize an autopilot component which it doesn't see.
3. Once H520 camera receives a heartbeat it starts forwarding packets from the autopilot.

@petergerten a quick hack is to comment out this line:
https://github.com/mavlink/MAVSDK/blob/d9c09d238ed496a44cfe9b9213c240ab1d4b2cef/src/core/system_impl.cpp#L323

For the longer term, I want to refactor some of this discovery/connection logic and clean it up. It then should send out heartbeats even if only a camera and no autopilot has communicated yet.

petergerten, 2019-11-28:
@julianoes the hack you described did not work for me.
I eventually got it working my removing the `!_parent->does_support_mission_int())` check in `mission_impl.cpp`

julianoes, 2019-11-28:
@petergerten I doubt you correctly did the hack then. Did you make sure to rebuild the library and install it again?

petergerten, 2019-12-04:
@julianoes 
ok, will check again.

Anyway, I would still appreciate if someone could explain what `does_support_mission_int()` means.
I sometimes get it with px4 SITL as well - not only with the H520.

Searching for it (https://github.com/mavlink/MAVSDK/search?q=does_support_mission_int&unscoped_q=does_support_mission_int) does not give many clues and honestly makes the codebase here a bit difficult to approach.

julianoes, 2019-12-04:
Ok MAVSDK needs to receive the message [AUTOPILOT_VERSION](https://mavlink.io/en/messages/common.html#AUTOPILOT_VERSION) once because it needs to know about the `capabilities` flags.
One of those flags is [MAV_PROTOCOL_CAPABILITY_MISSION_INT](https://mavlink.io/en/messages/common.html#MAV_PROTOCOL_CAPABILITY_MISSION_INT).

This flag means that the vehicle supports the `_INT` version of the [mission protocol](https://mavlink.io/en/services/mission.html) which has better accuracy than the legacy one.

The fact that you get the `does_support_mission_int() = false` is because you have not received the `AUTOPILOT_VERION` message indicating the capabilities but does not actually mean that `_INT` messages are not allowed. Therefore, I've been trying to fix the problem of not receiving `AUTOPILOT_VERSION` instead of removing the check for mission int which is just a symptom but not the root cause!

petergerten, 2019-12-04:
@julianoes thanks for the explanation!

julianoes, 2020-11-25:
I believe this can be closed now.

closed 2020-11-25 by julianoes