mavlink/MAVSDK #917: MAVSDK - setup_udp_remote - no MAVLink being received using Passthrough

VarkAven, 2019-12-03:
**Description:**

I have tried to connect to QGC to send/receive MAVLink messages using _MAVSDK 0.21.3_ setup_udp_remote, utilizing MavlinkPassthrough plugin. It seems like connection created this way is not able to receive MAVLink messages.

I've prepared a demonstration application to show just that:

**Demonstration:**

1. Clone https://github.com/VarkAven/mavsdk_udp_issue 
2. Build it using standard cmake build (make build directory, source CMakeLists.txt etc...) (Make sure you have _MAVSDK 0.21.3_ installed on your system)
3 Launch QGC (3.5.5), _set it up to receive UDP connections_ (QGC -> General -> check UDP field in "AutoConnect to the following devices")
4. Launch the demo application, **no arguments required**.
5. Observe that **"Received PING response!"** **does not appear.** Neither **"Received MAVLink message. It's msgid:"** despite being subscribed and having a callback to receive every MAVLink message.

**Some more information:**

Connecting to QGC via bare Linux API UDP socket and sending MAVLink Ping request yields QGC responding with Ping response MAVLink message. So QGC has Ping micro service implemented.

JonasVautherin, 2019-12-03:
I recently wrote a small test module connecting to QGC from MAVSDK, pretending to be an autopilot. I found a few issues in MAVSDK, that I patched [here](https://github.com/Auterion/fake_autopilot/blob/master/mavsdk.patch).

More specifically, MAVSDK ignores messages coming from QGC...

1. ... in [receive_message()](https://github.com/mavlink/MAVSDK/blob/develop/src/core/mavsdk_impl.cpp#L78-L84):

```cpp
    // FIXME: Ignore messages from QGroundControl for now. Usually QGC identifies
    //        itself with sysid 255.
    //        A better way would probably be to parse the heartbeat message and
    //        look at type and check if it is MAV_TYPE_GCS.
    if (message.sysid == 255) {
        return;
    }
```

2. ... in [receive()](https://github.com/mavlink/MAVSDK/blob/develop/src/core/udp_connection.cpp#L245-L246):

```cpp
    // FIXME: We ignore messages from QGC (255) for now.
    if (!saved_remote && sysid != 0 && sysid != 255) {
```

Also, my patch adds an `Autopilot` configuration, but I'm not sure if you need that.

Could you try removing those filters in MAVSDK and let us know if that helps? Then I am not completely sure how to fix that properly: should a GCS always ignore other QGC? What about a companion? Or more specifically, when should a MAVLink system ignore another one?

julianoes, 2020-11-20:
There was no response in a long time, closing.

closed 2020-11-20 by julianoes