mavlink/MAVSDK #925: Adding DISTANCE_SENSOR and possibly other raw sensor messages.

physicsman, 2019-12-09:
For offboard work I feel it would be useful to have access to the the raw sensor data. For example I am using Attitude+Lidar Rangefinder+Thermal Sensor  to create a heat map during a survey. 

**Suggestions/Ideas:**

A telemetry subclass ie `telemetry.raw()` that gathers sensor information and current readings available through mavlink. Also, IMO the sensor reading timestamp "time_boot_ms" (ex: https://mavlink.io/en/messages/common.html#DISTANCE_SENSOR) is especially useful from my perspective for sensor fusion and filtering to approximate vehicle state.

irsdkv, 2019-12-10:
@JonasVautherin I think I can participate in part of implement receiving and handling `DISTANCE_SENSOR` message.
In which plugin this should implemented on your opinion?
In `Telemetry` (whenever it already large)?

JonasVautherin, 2019-12-10:
Maybe in telemetry... I'm just wondering what happens when you don't have a distance sensor at all. What if you subscribe to that stream? You just get no event?

@julianoes: do you have an opinion?

irsdkv, 2019-12-10:
> You just get no event?

It seems so:
https://github.com/irsdkv/Firmware/blob/master/src/modules/mavlink/mavlink_messages.cpp#L4277

JonasVautherin, 2019-12-10:
Ok! Let's put that in the telemetry plugin then! Thanks @irsdkv!

physicsman, 2019-12-11:
I don't know the implementation however I think multiple distance sensor support will be coming for object avoidance (i.e. a face-forward and face-downward sensor).

julianoes, 2020-11-20:
DISTANCE_SENSOR has been implemented in #1222.

closed 2020-11-20 by julianoes