mavlink/MAVSDK #929: C:/MAVSDK/build/default/third_party/jsoncpp failed to build!

maerad7, 2019-12-11:
[image]
[image]

 C:/MAVSDK/build/default/third_party/jsoncpp failed to build! 

I get an error Why? The installation process is as follows.

cmdline => visual studio 2019 x86_x64 cross tools command prompt and admin permision
cd MAVSDK
git checkout master
git submodule update --init --recursive
cmake -G "Visual Studio 16 2019" -DBUILD_SHARED_LIBS=ON -DENABLE_MAVLINK_PASSTHROUGH=1 -Bbuild/default -H.
cmake --build build/default --config Debug
ERROR

JonasVautherin, 2019-12-11:
hmm, "patch step" is failing. Is that a clean build? I see messages like "CMake doesn't need to be rerun...".

Can you try a clean build? On Windows it may be that the patch step sometimes fails if it is not a clean build... :thinking:

lizn520, 2020-04-12:
same problem in Visual Studio 2017, 
Output: 
2>Creating directories for 'jsoncpp'
2>Building Custom Rule D:/SoftWareDevelopement/Ruigan/MAVSDK/third_party/jsoncpp/CMakeLists.txt
2>Performing download step (git clone) for 'jsoncpp'
2>-- Avoiding repeated git clone, stamp file is up to date: 'D:/SoftWareDevelopement/Ruigan/MAVSDK/build/release/third_party/jsoncpp/jsoncpp/src/jsoncpp-stamp/jsoncpp-gitclone-lastrun.txt'
2>Performing update step for 'jsoncpp'
2>Performing patch step for 'jsoncpp'
2>CUSTOMBUILD : error : patch failed: CMakeLists.txt:148
2>CUSTOMBUILD : error : CMakeLists.txt: patch does not apply
2>CUSTOMBUILD : error : patch failed: src/lib_json/CMakeLists.txt:69
2>CUSTOMBUILD : error : src/lib_json/CMakeLists.txt: patch does not apply

JonasVautherin, 2020-04-16:
Same question as above :sweat_smile:. Is that a clean build? Doesn't seem like a clean build:

> Avoiding repeated git clone, stamp file is up to date

julianoes, 2020-11-20:
If this is still an issue, please re-open.

closed 2020-11-20 by julianoes