mavlink/MAVSDK #933: timesync warnings

julianoes, 2019-12-17:
I believe since the timesync functionality was added in #890 we now always get warnings on the PX4 side:

```
ERROR [mavlink] [timesync] Time jump detected. Resetting time synchroniser.
```

@irsdkv should we disable the timesync plugin by default and only enable it for things like mocap? Or is the PX4 "error" just wrong and this is fine?

irsdkv, 2019-12-17:
I think that it will be ok to disable it by default if this annoying message occurs always.
Please let me some time to fix tymesync algorithm to prevent it.

irsdkv, 2019-12-17:
That's what I found:
1. On my hardware (Pixhawk 4) timesync works ok
1.1. What I do:
1.1.0. Modify master Firmware to log hitting in [this line](https://github.com/PX4/Firmware/blob/07d656e971a72d1202651dfd3b4642736fb078d7/src/modules/mavlink/mavlink_timesync.cpp#L101):
1.1.1. Upload Firmware on Pixhawk 4
1.1.2. Connect with FMU Debug port to system console
1.1.3. Start MAVSDK server
1.1.4. Start `TIMESYNC` stream (in my case on ttyS2) on Pixhawk: `mavlink stream -d /dev/ttyS2 -s TIMESYNC -r 2`
1.2. What I received on System console:
1.2.1. Sometimes `High RTTY` [this line](https://github.com/PX4/Firmware/blob/07d656e971a72d1202651dfd3b4642736fb078d7/src/modules/mavlink/mavlink_timesync.cpp#L133) (that's ok).
1.2.2. My log msg (so adjusting go normally).

So seems that in my case all worked ok. But on some systems `system_time` subsystem may work different.

I think that this can be fixed with [this PR](#934).
Please try to build Server for it and let me know what happened.

julianoes, 2019-12-17:
Hm, maybe the problem is because I'm using `PX4_SIM_SPEED_FACTOR=20 make px4_sitl gazebo` saw the time on the PX4 side runs 20 times faster.

fnoop, 2019-12-17:
If you've got a system timesync running (ntpd, chronyd, systemd-timesyncd), whenever they adjust the system clock you will probably get this error.

julianoes, 2019-12-19:
@fnoop that makes sense.
I think for me case the issue is that I run the simulation faster than realtime, so for that case the sync does not work, and just shouldn't be used.