mavlink/MAVSDK #936: cmake --target install doesn't install thirdparty files

fnoop, 2019-12-17:
Hi, if SUPERBUILD=ON (default), all the thirdparty stuff gets built in build/default/third_party/install, but none of it gets installed, so the installation doesn't work.  eg.:

```
[dev] [mav@maverick-ubuntu ~/software/mavsdk.notworking/lib]$ LD_LIBRARY_PATH=. ldd -r libmavsdk_server.so
	linux-vdso.so.1 (0x00007ffec2bd5000)
	libmavsdk_action.so => ./libmavsdk_action.so (0x00007f9cef8d6000)
	libmavsdk_calibration.so => ./libmavsdk_calibration.so (0x00007f9cef6c6000)
	libmavsdk_geofence.so => ./libmavsdk_geofence.so (0x00007f9cef4a9000)
	libmavsdk_gimbal.so => ./libmavsdk_gimbal.so (0x00007f9cef2a4000)
	libmavsdk_camera.so => ./libmavsdk_camera.so (0x00007f9cef029000)
	libmavsdk_mission.so => ./libmavsdk_mission.so (0x00007f9ceedd3000)
	libmavsdk_offboard.so => ./libmavsdk_offboard.so (0x00007f9ceebb8000)
	libmavsdk_telemetry.so => ./libmavsdk_telemetry.so (0x00007f9cee992000)
	libmavsdk_info.so => ./libmavsdk_info.so (0x00007f9cee78c000)
	libmavsdk_param.so => ./libmavsdk_param.so (0x00007f9cee584000)
	libmavsdk_shell.so => ./libmavsdk_shell.so (0x00007f9cee373000)
	libmavsdk_mocap.so => ./libmavsdk_mocap.so (0x00007f9cee164000)
	libmavsdk.so => ./libmavsdk.so (0x00007f9cede88000)
	libgrpc++.so => not found
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f9cedc69000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f9ced8e0000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f9ced6c8000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f9ced2d7000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f9cf05a6000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f9cecf39000)
undefined symbol: _ZN4grpc6Status2OKE	(./libmavsdk_server.so)
undefined symbol: _ZN4grpc24g_core_codegen_interfaceE	(./libmavsdk_server.so)
undefined symbol: _ZN4grpc6g_glipE	(./libmavsdk_server.so)
undefined symbol: _ZN9grpc_impl13ServerBuilderD1Ev	(./libmavsdk_server.so)
undefined symbol: _ZN9grpc_impl13ServerBuilderC1Ev	(./libmavsdk_server.so)
undefined symbol: _ZN9grpc_impl13ServerBuilder13BuildAndStartEv	(./libmavsdk_server.so)
undefined symbol: _ZN9grpc_impl13ServerBuilder16AddListeningPortERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESt10shared_ptrINS_17ServerCredentialsEEPi	(./libmavsdk_server.so)
undefined symbol: _ZN9grpc_impl13ServerBuilder15RegisterServiceEPN4grpc7ServiceE	(./libmavsdk_server.so)
undefined symbol: _ZN9grpc_impl25InsecureServerCredentialsEv	(./libmavsdk_server.so)
```

Is there a cmake switch to install the dep libraries?

JonasVautherin, 2019-12-17:
They will get installed in `build/default/third_party/install` in your case. You can change the output by setting [-DDEPS_INSTALL_PATH](https://github.com/mavlink/MAVSDK/blob/develop/CMakeLists.txt#L20) :blush:.

fnoop, 2019-12-17:
Works, thanks :)

closed 2019-12-17 by fnoop