mavlink/MAVSDK #937: Airspeed data missing on telemetry

itrigitech, 2019-12-17:
I've been looking for the airspeed data in telemetry and doesn't seem to be implemented. For some vehicles this information is important depending the fly mode so I was thinking to modify the telemetry plugin to add this information because I need it. The mavlink message is VFR_HUD. 
If I want to contribute how should I start?

JonasVautherin, 2019-12-18:
I see that [VFR_HUD](https://mavlink.io/en/messages/common.html#VFR_HUD) is defined as follows:

```
Metrics typically displayed on a HUD for fixed wing aircraft.

airspeed     | float    | m/s | Current indicated airspeed (IAS).
groundspeed  | float    | m/s | Current ground speed.
heading      | int16_t  | deg | Current heading in compass units (0-360, 0=north).
throttle     | uint16_t | %   | Current throttle setting (0 to 100).
alt          | float    | m   | Current altitude (MSL).
climb        | float    | m/s | Current climb rate.
```

For information, we have a [`GroundSpeedNed`](https://github.com/mavlink/MAVSDK-Proto/blob/master/protos/telemetry/telemetry.proto#L34) that emits [`SpeedNed`](https://github.com/mavlink/MAVSDK-Proto/blob/master/protos/telemetry/telemetry.proto#L201-L206) events.

I guess VFR_HUD is only emitted for fixed wing or VTOL (in fixed wind mode) aircrafts? Or is it also emitted for multicopters (I guess multicopers typically don't compute their airspeed, do they?)?

julianoes, 2019-12-19:
> I guess VFR_HUD is only emitted for fixed wing or VTOL (in fixed wind mode) aircrafts? Or is it also emitted for multicopters (I guess multicopers typically don't compute their airspeed, do they?)?

Good question. We'd have to try it out.

Anyway, you can add it to the telemetry plugin, and I think the `VFR_HUD` message is the one to use.

itrigitech, 2019-12-19:
I don't know how it works but i could imagine the message sent if the airspeed sensor is present, anyway in Mavlink documentation says "Metrics typically displayed on a HUD for fixed wing aircraft." 
Anyway, testing with gazebo multicopter i see the VFR_HUD message sent so i imagine it depends more in the fact of the sensor being present or not.

I think it's a good idea to add it to the telemetry plugin then. For some drones this information is very important so i will try to add it when i have time.

itrigitech, 2019-12-24:
I'm starting to add the message to the plugin. I'm thinking how to structure the information. The important information in this message is the airspeed for me but the rest of the information we have it already from other messages. I really don't know if there is any difference or not in this information received but i guess not.

What would you do with that? add all the information as it comes ? o just take the information needed? just the airspeed

JonasVautherin, 2019-12-24:
> I really don't know if there is any difference or not in this information received but i guess not.

In that case I would not repeat it.

> the rest of the information we have it already from other messages

I believe we get `groundspeed`, `heading` and `alt`, but not `airspeed`, `throttle` and `climb`. Is that correct? In that case I guess we could add the new ones only :+1:.

PR #957 (2020-01-24): Adding airspeed to telemetry
As we have talked i've added the vfr_hud message to the telemetry plugin. As i really didn't know how to name this i leave it as vfr_hud but if you think that we should change names tell me your proposal.

I've created all the integration tests and so but as is my first time contributing here i don't know if i've missed something. Let me know if is the case.

Resolves #937.
closed 2020-01-24 by julianoes