mavlink/MAVSDK #939: Support for global position setpoints with SET_POSITION_TARGET_GLOBAL_INT

cjffly, 2019-12-18:
As mentioned in here in [#12819](https://github.com/PX4/Firmware/pull/12819), MAVSDK does not support global position setpoints yet. According to @julianoes [comments](https://github.com/PX4/Firmware/pull/12819#issuecomment-526110965),  it seems not too hard to add a block of codes on the MAVSDK side. Is there any plan to add that?

JonasVautherin, 2019-12-20:
> Is there any plan to add that?

Not on our side, I believe. But happy to review contributions :blush:.

cjffly, 2019-12-23:
@JonasVautherin Thanks for the info. I am not a C++ expert but tried adding the SET_POSITION_TARGET_GLOBAL_INT on the MAVSDK side, similar to the LOCAL_NED codes. Then I tested with jmavsim, the multi-copter could yaw accordingly but not the global position coordinates. 

My goal is simply using MAVSDK to set the drone a GPS location and fly there, similar to what the QGC does with the mouse click go_to_location. Could some one tell me can SET_POSITION_TARGET_GLOBAL_INT achieve this?

JonasVautherin, 2019-12-24:
> My goal is simply using MAVSDK to set the drone a GPS location and fly there, similar to what the QGC does with the mouse click go_to_location.

Doesn't [goto_location](https://github.com/mavlink/MAVSDK/blob/develop/src/plugins/action/include/plugins/action/action.h#L154) do what you want already?

cjffly, 2019-12-24:
> > My goal is simply using MAVSDK to set the drone a GPS location and fly there, similar to what the QGC does with the mouse click go_to_location.
> 
> Maybe I misunderstood the question in the beginning confused. Doesn't [goto_location](https://github.com/mavlink/MAVSDK/blob/develop/src/plugins/action/include/plugins/action/action.h#L154) do what you want already?

You are right! I overlooked and didn't notice this exist.

I read all the Examples and thinking maybe there are some functions in follow_me or off_board could achieve my goal, and ignore the Action class. Maybe we can add this goto_location into takeoff_land example, so other people won't miss this?

JonasVautherin, 2019-12-25:
> Maybe we can add this goto_location into takeoff_land example, so other people won't miss this?

Sure! Good idea! :blush:

jbeyerstedt, 2020-11-03:
I would just like to add a +1 requesting this feature. Especially for UTM applications, which are taking positions in WGS84 coordinates, it'll be very useful to set the own vehicles's position using the same reference frame. Incoming coordinates are not only other vehicle's positions (see [UTM_GLOBAL_POSITION](https://mavlink.io/en/messages/common.html#UTM_GLOBAL_POSITION)), but might also be temporary and stationary geofences (for obstacles, but also dynamically closed sections of the airspace).

Using the follow-me mode or goto location might be a short-term workaround, but not a good solution, because follow-me is intended for something else (and might interfere with the mobile QGC app setting the target position) and goto has no connection loss failsafe like offboard control. Also the pilot should always have the opportunity to disable control of the companion, which would not be possible using goto. (If I understood the message's documentation correctly, receiving the message once will switch to hold mode with a custom position. Such a mode change might not always be a reasonable choice.)

romain-chiap, 2021-11-17:
Hi, I added it in this [PR](https://github.com/mavlink/MAVSDK/pull/1620)

romain-chiap, 2021-11-17:
Alternatively, you can use action.[goto_location](https://mavsdk.mavlink.io/main/en/cpp/api_reference/classmavsdk_1_1_action.html#classmavsdk_1_1_action_1afb3546fa994357e491816f2032716818)() which is mostly similar

PR #1620 (2021-11-17): Offboard pos global
Offboard global position control using [SET_POSITION_TARGET_GLOBAL_INT](https://mavlink.io/en/messages/common.html#SET_POSITION_TARGET_GLOBAL_INT), (latitude, longitude, altitude relative from home, and yaw).

Tested successfully in [sih](https://docs.px4.io/v1.12/en/simulation/simulation-in-hardware.html) simulator with a quad and a fw.

Proto is in [PR-264](https://github.com/mavlink/MAVSDK-Proto/pull/264)

Resolves #939