mavlink/MAVSDK #947: MAVSDK shows as "Vehicle 190"

petergerten, 2019-12-31:
My Setup:
```
PX4 SITL <-> Mavlink Router <-> MAVSDK
                            <-> QGC
```

When using the MAVLink Inspector in QGC, I can see:
Vehicle 1 -> all expected PX4 SITL messages
Vehicle 190 -> HEARTBEAT, TIMESYNC

Why is MAVSDK showing as "Vehicle 190"?

Also I saw that UUIDs are used now - is there a good overview somewhere on how system ID vs. UUID come into play?

JonasVautherin, 2020-01-06:
This is because MAVSDK currently identifies itself as [MISSION_PLANNER](https://github.com/mavlink/MAVSDK/blob/develop/src/core/mavsdk_impl.cpp#L317-L318):

```cpp
uint8_t MavsdkImpl::get_own_system_id() const
{
    switch (_configuration.load()) {
        case Mavsdk::Configuration::GroundStation:
            return MAV_COMP_ID_MISSIONPLANNER;
```

I believe that one should be able to set the system id through the [configuration](https://github.com/mavlink/MAVSDK/blob/develop/src/core/mavsdk_impl.cpp#L246-L249):

```cpp
void MavsdkImpl::set_configuration(Mavsdk::Configuration configuration)
```

[`Configuration`](https://github.com/mavlink/MAVSDK/blob/develop/src/core/mavsdk.h#L137-L142) is currently an enum class:

```cpp
    enum class Configuration {
        Autopilot, /**< @brief SDK is used as an autopilot. */
        GroundStation, /**< @brief SDK is used as a ground station. */
        CompanionComputer /**< @brief SDK is used on a companion computer onboard the system (e.g.
                             drone). */
    };
```

But maybe it should contain the sysid, too.

@julianoes: Do you have an opinion on that?

petergerten, 2020-01-07:
@JonasVautherin My confusion is mainly because I though "Vehicle" would show the system ID, not the component ID. If not, what is a "Vehicle" in MAVLink terms?

JonasVautherin, 2020-01-07:
> "Vehicle" would show the system ID

It does. We just return [`MAV_COMP_ID_MISSIONPLANNER`](https://mavlink.io/en/messages/common.html#MAV_COMP_ID_MISSIONPLANNER) as the system id, which is "190". So I would say that the peculiarity is on MAVSDK's side :sweat_smile:.

coderkalyan, 2020-01-28:
@JonasVautherin I for one would prefer it greatly if we could expose both the system ID as well as the component ID in the Configuration, and furthermore make this available in the gRPC bindings. Also, QGC's sysid is not hardcoded - it's set to 255 by default, but this can be changed under Application Settings -> MAVLink. I'm not sure what it does for component ID either.

coderkalyan, 2020-05-17:
@JonasVautherin I think this can be closed ever since the new Configuration class including system and component ID was merged?

JonasVautherin, 2020-05-18:
Yes, thanks @coderkalyan :blush:

closed 2020-05-18 by JonasVautherin